esphome/issues #181: Ultrasonic Distance Sensor issues in 1.12.1

astromgren, 2019-03-30:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io Addon v1.12.1

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 NodeMcu CP2102
**Affected component:**

https://esphome.io/components/sensor/ultrasonic.html

**Description of problem:**
I'm experiencing the same issues found in closed [Issue 147](https://github.com/esphome/issues/issues/147). Returns distance measurement of 0.1 m
Does not change values return. Tested on two different boards but I don't have another sensor to test.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: salt_sensor
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: "myssid"
  password: "mypw"

sensor:
  - platform: ultrasonic
    trigger_pin: D1
    echo_pin: D2
    name: "Ultrasonic Sensor"
    accuracy_decimals: 2

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

```
Log
```
INFO Reading configuration...
INFO Starting log output from salt_sensor.local using esphome API
INFO Connecting to salt_sensor.local:6053 (192.168.86.26)
INFO Successfully connected to salt_sensor.local
[12:05:55][I][application:097]: esphome-core version 1.12.1 compiled on Mar 30 2019, 11:45:20
[12:05:55][C][wifi:371]: WiFi:
[12:05:55][C][wifi:253]:   SSID: [redacted]
[12:05:55][C][wifi:254]:   IP Address: 192.168.86.26
[12:05:55][C][wifi:256]:   BSSID: [redacted]
[12:05:55][C][wifi:257]:   Hostname: 'salt_sensor'
[12:05:55][C][wifi:261]:   Signal strength: -37 dB ▂▄▆█
[12:05:55][C][wifi:262]:   Channel: 6
[12:05:55][C][wifi:263]:   Subnet: 255.255.255.0
[12:05:55][C][wifi:264]:   Gateway: 192.168.86.1
[12:05:55][C][wifi:265]:   DNS1: 192.168.86.1
[12:05:55][C][wifi:266]:   DNS2: 0.0.0.0
[12:05:55][C][logger:142]: Logger:
[12:05:55][C][logger:143]:   Level: DEBUG
[12:05:55][C][logger:144]:   Log Baud Rate: 115200
[12:05:55][C][logger:145]:   Hardware UART: UART0
[12:05:55][C][sensor.ultrasonic:061]: Ultrasonic Sensor 'Ultrasonic Sensor'
[12:05:55](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[12:05:55](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[12:05:55](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[12:05:55][C][sensor.ultrasonic:062]:   Echo Pin: GPIO4 (Mode: INPUT)
[12:05:55][C][sensor.ultrasonic:063]:   Trigger Pin: GPIO5 (Mode: OUTPUT)
[12:05:55][C][sensor.ultrasonic:064]:   Pulse time: 10 µs
[12:05:55][C][sensor.ultrasonic:065]:   Timeout: 58309 µs
[12:05:55][C][sensor.ultrasonic:066]:   Update Interval: 5000 ms
[12:05:55][C][api:101]: API Server:
[12:05:55][C][api:102]:   Address: salt_sensor.local:6053
[12:05:55][C][ota:127]: Over-The-Air Updates:
[12:05:55][C][ota:128]:   Address: salt_sensor.local:8266
[12:05:56][D][sensor.ultrasonic:049]: Got distance: 0.1 m
[12:06:01][D][sensor.ultrasonic:049]: Got distance: 0.1 m
[12:06:06][D][sensor.ultrasonic:049]: Got distance: 0.1 m
[12:06:11][D][sensor.ultrasonic:049]: Got distance: 0.1 m
[12:06:16][D][sensor.ultrasonic:049]: Got distance: 0.1 m
[12:06:21][D][sensor.ultrasonic:049]: Got distance: 0.1 m
[12:06:26][D][sensor.ultrasonic:049]: Got distance: 0.1 m
[12:06:31][D][sensor.ultrasonic:049]: Got distance: 0.1 m
[12:06:36][D][sensor.ultrasonic:049]: Got distance: 0.1 m

```
**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

RomanWojnar, 2019-03-30:
Why don't use the Otto Winter advice from Issuse 147, adding two lines:
```
esphome:
  # ...
  esphome_core_version:
    branch: fix-ultrasonic
```
It works.

astromgren, 2019-03-30:
Apologies as I wasn't clear.  I should have included that I have also tried the fix-ultrasonic brach without success
```
esphome:
  name: salt_sensor
  platform: ESP8266
  board: nodemcuv2
  esphome_core_version:
    branch: fix-ultrasonic

wifi:
  ssid: myssid
  password: mypasswd

sensor:
  - platform: ultrasonic
    trigger_pin: D1
    echo_pin: D2
    name: "Ultrasonic Sensor"
    accuracy_decimals: 2

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
```
Logs
```
INFO Reading configuration...
INFO Starting log output from salt_sensor.local using esphome API
INFO Connecting to salt_sensor.local:6053 (192.168.86.26)
INFO Successfully connected to salt_sensor.local
[14:26:05][I][application:097]: esphome-core version 1.13.0-dev compiled on Mar 30 2019, 14:22:43
[14:26:05][C][wifi:371]: WiFi:
[14:26:05][C][wifi:253]:   SSID: [redacted]
[14:26:05][C][wifi:254]:   IP Address: 192.168.86.26
[14:26:05][C][wifi:256]:   BSSID: [redacted]
[14:26:05][C][wifi:257]:   Hostname: 'salt_sensor'
[14:26:05][C][wifi:261]:   Signal strength: -55 dB ▂▄▆█
[14:26:05][C][wifi:262]:   Channel: 6
[14:26:05][C][wifi:263]:   Subnet: 255.255.255.0
[14:26:05][C][wifi:264]:   Gateway: 192.168.86.1
[14:26:05][C][wifi:265]:   DNS1: 192.168.86.1
[14:26:05][C][wifi:266]:   DNS2: 0.0.0.0
[14:26:05][C][logger:142]: Logger:
[14:26:05][C][logger:143]:   Level: DEBUG
[14:26:05][C][logger:144]:   Log Baud Rate: 115200
[14:26:05][C][logger:145]:   Hardware UART: UART0
[14:26:05][C][sensor.ultrasonic:045]: Ultrasonic Sensor 'Ultrasonic Sensor'
[14:26:05](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[14:26:05](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[14:26:05][C][sensor.ultrasonic:045]:   Icon: 'mdi:arrow-expand-vertical'
[14:26:05][C][sensor.ultrasonic:046]:   Echo Pin: GPIO4 (Mode: INPUT)
[14:26:05][C][sensor.ultrasonic:047]:   Trigger Pin: GPIO5 (Mode: OUTPUT)
[14:26:05][C][sensor.ultrasonic:048]:   Pulse time: 10 µs
[14:26:05][C][sensor.ultrasonic:049]:   Timeout: 11662 µs
[14:26:05][C][sensor.ultrasonic:050]:   Update Interval: 60000 ms
[14:26:05][C][api:101]: API Server:
[14:26:05][C][api:102]:   Address: salt_sensor.local:6053
[14:26:05][C][ota:127]: Over-The-Air Updates:
[14:26:05][C][ota:128]:   Address: salt_sensor.local:8266
[14:26:59][D][sensor.ultrasonic:040]: 'Ultrasonic Sensor' - Got distance: 0.00 m
[14:27:05][I][ota:152]: Boot seems successful, resetting boot loop counter.
[14:27:59][D][sensor.ultrasonic:040]: 'Ultrasonic Sensor' - Got distance: 0.00 m
[14:28:59][D][sensor.ultrasonic:040]: 'Ultrasonic Sensor' - Got distance: 0.00 m
[14:29:08][D][api:073]: Disconnecting aioesphomeapi (192.168.86.63)
[14:29:59][D][sensor.ultrasonic:040]: 'Ultrasonic Sensor' - Got distance: 0.00 m
© 2019 Copyright Otto Winter, Made with Materialize
```

leoMehlig, 2019-03-30:
I have the same issue with two sensors and a sonoff basic

astromgren, 2019-03-31:
I removed the branch and updated to 1.12.2 today and am still seeing the same result.  I've ordered a couple more sensors to rule that out, but I'm not sure what else I can attempt at this time.

```
[11:19:25][I][application:097]: esphome-core version 1.12.2 compiled on Mar 31 2019, 11:13:45
```
...
```
[11:19:43][D][sensor.ultrasonic:040]: 'Ultrasonic Sensor' - Got distance: 0.00 m
[11:20:43][D][sensor.ultrasonic:040]: 'Ultrasonic Sensor' - Got distance: 0.00 m
[11:21:43][D][sensor.ultrasonic:040]: 'Ultrasonic Sensor' - Got distance: 0.00 m
```

RomanWojnar, 2019-03-31:
It works for me.
Wemos D1 mini and Wemos D1 mini PRO.

OttoWinter, 2019-04-01:
This has been resolved in 1.12.2

closed 2019-04-01 by OttoWinter

leoMehlig, 2019-04-04:
The new sensors just arrived and I still have the same issue with the sonoff basic...

pataha, 2019-05-07:
@leoMehlig can you show the wiring of sensor with sonoff basic?, I used 5v pad on sonoff with srf-05 + voltage divider on echo pin but it seems does not work to me

leoMehlig, 2019-05-20:
@pataha  I did't use any voltage divider. Just 5v to sensor.