esphome/issues #185: SP22 toggle button error

hubgitbb, 2019-03-31:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

**Affected component:**

**Description of problem:**
i have two teckin sp22 power plug (named p1 and p2). they also have a push button. now if i push the button on plug p1 i want that on plug p2 toggle power on or off.

this doesent work:
on_press:
- then:
- switch.toggle: p2_Relay

Couldn't find ID 'p2_Relay'

if i try the same with the local button i get the same error

on_press:
- then:
- switch.toggle: "${plug_name}_button"

Couldn't find ID 'p1_button'

**Problem-relevant YAML-configuration entries:**
```yaml
substitutions:
plug_name: p1
Higher value gives lower watt readout

current_res: "0.00182"
Lower value gives lower voltage readout

voltage_div: "788"

esphome:
name: ${plug_name}
platform: ESP8266
board: esp8285

wifi:
ssid: 'myssid'
password: '********'
Enable logging

logger:
Enable Web server

web_server:
port: 80
Enable Home Assistant API

api:
password: '*********'

ota:
password: '********'

time:

    platform: homeassistant
    id: homeassistant_time

binary_sensor:

    platform: gpio
    pin:
    number: GPIO1
    inverted: True
    name: "${plug_name}_button"
    on_press:
        then:
            switch.toggle: p2_Relay

switch:

    platform: gpio
    name: "${plug_name}_Relay"
    pin: GPIO14
    restore_mode: ALWAYS_ON
    platform: gpio
    name: "${plug_name}_LED_Blue"
    pin: GPIO13
    inverted: True
    restore_mode: ALWAYS_OFF
    platform: gpio
    name: "${plug_name}_LED_Red"
    pin: GPIO3
    inverted: True
    restore_mode: ALWAYS_OFF
    sensor:
        platform: hlw8012
        sel_pin:
        number: GPIO12
        inverted: True
        cf_pin: GPIO04
        cf1_pin: GPIO05
        current_resistor: ${current_res}
        voltage_divider: ${voltage_div}
        current:
        name: "${plug_name}_Amperage"
        unit_of_measurement: A
        voltage:
        name: "${plug_name}_Voltage"
        unit_of_measurement: V
        power:
        name: "${plug_name}_Wattage"
        unit_of_measurement: W
        id: "${plug_name}_Wattage"
        change_mode_every: 8
        update_interval: 10s
        platform: total_daily_energy
        name: "${plug_name}_Total Daily Energy"
        power_id: "${plug_name}_Wattage"
        filters:
        # Multiplication factor from W to kW is 0.001
        - multiply: 0.001
        unit_of_measurement: kWh

Extra sensor to keep track of plug uptime

    platform: uptime
    name: ${plug_name}_Uptime Sensor

```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

OttoWinter, 2019-03-31:
that's because the relays doesn't have an id set. please read the configuration types page

closed 2019-03-31 by OttoWinter

hubgitbb, 2019-04-02:
i have now added an id o each plug an the push button works on the same device but not o the remote device.