esphome/issues #186: Make yaml error reports actually mean something

ASMfreaK, 2019-03-31:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Does not matter, I **think**
**ESP (ESP32/ESP8266, Board/Sonoff):**

Does not matter, I **think**
**Affected component:**

[`gpio switch`](https://esphome.io/components/switch/gpio.html) has this problem

**Description of problem:**
When having a errneous yaml config like provided next, ESPhome in wizard provides no information whatsoever on the actual information, just nothing.

**Problem-relevant YAML-configuration entries:**
```yaml
switch:
  - platform: gpio
    pin: D1
    name: "Some name"
  - platfrom: gpio
    pin: D2
    name: "Some another name"
```

**Traceback (if applicable):**

```
INFO Reading configuration...
Failed config
```

stderr log from esphome server
```
INFO 101 GET /validate (<REDACTED>) 4.15ms
INFO Running command 'esphome --dashboard ./<REDACTED>.yaml config'
```

**Additional information and things you've tried:**

gitolicious, 2019-04-20:
Will be fixed with the changes in https://github.com/esphome/esphome/tree/cleanup-dashboard

OttoWinter, 2019-04-25:
Fixed in dev

closed 2019-04-25 by OttoWinter