esphome/issues #187: No network found! (1.12.2)

tweaker, 2019-04-01:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Docker

**ESP (ESP32/ESP8266, Board/Sonoff):**
Sonoff basic

**Affected component:**
https://esphome.io/devices/sonoff_basic.html

**Description of problem:**
I updated from 1.12.1 to 1.12.2 and sonoff stopped connecting to wifi

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: bathroom_fan
  platform: ESP8266
  board: esp01_1m

wifi:
  domain: .local
  networks:
  - ssid: ASUS
    password: [Redacted]
  use_address: bathroom_fan.local

logger:

api:
  password: [Redacted]

ota:
  password: [Redacted]

web_server:
  port: 80
  
binary_sensor:
  - platform: status
    name: Bathroom Fan Status
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    name: Bathroom Fan Button
    on_press:
      - switch.toggle: relay
    
switch:
  - platform: gpio
    name: Bathroom Fan Relay
    pin: GPIO12
    id: relay
    
sensor:
  - platform: wifi_signal
    name: Bathroom Fan WiFi Signal
    update_interval: 60s
  - platform: uptime
    name: Bathroom Fan Uptime

status_led:
  pin:
    number: GPIO13
    inverted: True
```

**Traceback (if applicable):**

```
# sudo cu -s 115200 -l /dev/cu.SLAB_USBtoUART                                                                              
Connected.
[D][wifi:286]: Found networks:
[D][wifi:288]:   No network found!
[D][wifi:271]: Starting scan...
[D][wifi:286]: Found networks:
[D][wifi:288]:   No network found!
[D][wifi:271]: Starting scan...
[D][wifi:286]: Found networks:
[D][wifi:288]:   No network found!
[D][wifi:271]: Starting scan...
[D][wifi:286]: Found networks:
[D][wifi:288]:   No network found!
[D][wifi:271]: Starting scan...
[D][wifi:286]: Found networks:
[D][wifi:288]:   No network found!
[D][wifi:271]: Starting scan...
[D][wifi:286]: Found networks:
[D][wifi:288]:   No network found!
[D][wifi:271]: Starting scan...
[D][wifi:286]: Found networks:
[D][wifi:288]:   No network found!
[W][wifi:429]: Restarting WiFi adapter...
[D][wifi:271]: Starting scan...
[D][wifi:286]: Found networks:
[D][wifi:288]:   No network found!
```

**Additional information and things you've tried:**

JakobM999, 2019-04-01:
Same problem here "No Wifi" with the sonoff basic after upgrading from 1.12.1 -> 1.12.2
i have static ip

OttoWinter, 2019-04-01:
Hmm, that is bad. No wifi also means no ota safe mode. But I cannot replicate it here with a sonoff basic and the same config (with only passwords and ssid changed).

1.12.2 has had two changes related to wifi:
- https://github.com/esphome/esphome-core/pull/566
- https://github.com/esphome/esphome-core/pull/572

neither of which should cause the issue you mentioned. 566 was just a small bugfix and the interface is restarted anyway after a few failed scan attempts, plus this would only show up on the second upload after 1.12.2, not when switching.

And 572 was basically just a some files being moved around - with no changes to the contents.

Anything show up with verbose logs? If the scan fails for some reason it would show up in those logs with more infos.

tatunts, 2019-04-01:
Hello @tweaker, have you seen the issue I've opened a couple of weeks ago? https://github.com/esphome/issues/issues/162

Your problem seems to be the same as mine. I've updated to the latest version (1.12.2), flashed a couple of problematic devices and now it seems that the problem is solved.

The behavior is exactly the same: If sensor or a switch automation is taken out, wifi connects successfully.

Thanks.

tweaker, 2019-04-02:
I had a similar problem on earlier versions if, for example, in the configuration of the sensor to be placed above the binary sensor. Maybe my device is bugs. On other devices, everything is normally stitched and works.
I myself can not understand the logic of this bug. The device is placed in the wall for exhaust and every time you have to disassemble everything in case of a bug.

redskinhu, 2019-04-02:
Hello

I have a same problem but I have NodeMCU clone not ESP01.

[image]

I used ESPEasy with this module before and I had a weird wifi issues like this.

If the binary sensor on the GPIO0 is disabled everything works fine. Tested with normal/dev and `fast_connect: on/off` settings.

My config:
```
esphome:
  name: ${devicename}
  platform: ESP8266
  board: nodemcuv2
  #arduino_version: dev

substitutions:
  devicename: "test40_nodemcuv2"
  upper_devicename: "Test40"

web_server:
  port: 80

logger:
  level: DEBUG

api:
  password: !secret esphome_api

ota:
  safe_mode: True

wifi:
  domain: .station
  ssid: !secret wifissid
  password: !secret wifikey
  reboot_timeout: 20min
  #fast_connect: on

switch:
  - platform: restart
    name: ${upper_devicename} restart

binary_sensor:
  - platform: status
    name: ${upper_devicename} Status

  # - platform: gpio
    # name: ${upper_devicename} Flash Button
    # id: ${devicename}_flash_button
    # pin:
      # number: GPIO0
      # inverted: True
      # mode: INPUT_PULLUP
    # filters:
      # - delayed_on: 80ms

status_led:
  pin:
    number: 16
    inverted: True
```
Thx
T

tatunts, 2019-04-04:
Hello @redskinhu, have you tried the latest version? Thanks.

Daandamhuis, 2019-04-04:
With this config it works;

```
substitutions:
  devicename: tv_schuur
  upper_devicename: TV Schuur
  device_type: Sonoff Basic
  firmware_version: v1.0

esphome:
  name: $devicename
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: !secret wifi_general
  password: !secret wifi_password
  
# Enable logging
logger:
  baud_rate: 0

uart:
  rx_pin: RX
  baud_rate: 115200

# Enable Home Assistant API
api:
  password: !secret api_password

ota:
  password: !secret ota_password

text_sensor:
  - platform: version
    name: ${device_type} ${firmware_version}     

sensor:
  - platform: wifi_signal
    name: ${upper_devicename} WiFi Signal
    update_interval: 60s
  - platform: uptime
    name: ${upper_devicename} Uptime

binary_sensor:
  - platform: status
    name: ${upper_devicename} Status
  - platform: gpio
    id: button
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    on_press:
      - light.toggle: light

output:
  - platform: gpio
    id: relay
    pin: GPIO12

light:
  - platform: binary
    id: light
    name: $upper_devicename
    output: relay
```
And with this one it doesn't. Here I changed from Light to Switch. It is a Sonoff basic.
```
substitutions:
  devicename: tv_schuur
  upper_devicename: TV Schuur
  device_type: Sonoff Basic
  firmware_version: v1.0

esphome:
  name: $devicename
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: !secret wifi_general
  password: !secret wifi_password
  
# Enable logging
logger:
  baud_rate: 0

uart:
  rx_pin: RX
  baud_rate: 115200

# Enable Home Assistant API
api:
  password: !secret api_password

ota:
  password: !secret ota_password

text_sensor:
  - platform: version
    name: ${device_type} ${firmware_version}     

sensor:
  - platform: wifi_signal
    name: ${upper_devicename} WiFi Signal
    update_interval: 60s
  - platform: uptime
    name: ${upper_devicename} Uptime

status_led:
  pin:
    number: GPIO13
    inverted: yes

binary_sensor:
  - platform: status
    name: ${upper_devicename} Status
  - platform: gpio
    id: button
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    on_press:
      - switch.toggle: relay

switch:
  - platform: gpio
    name: $upper_devicename
    pin: GPIO12
    id: relay
```

wmnl25, 2019-04-13:
> Hello
> 
> I have a same problem but I have NodeMCU clone not ESP01 (1.12.2 installed).
> 
> [image]
> 
(....)
> 
> Thx
> T

Same (weird) problems with this specific board overhere. Tried with three of these boards. 

This works:
```yaml
output:
  - platform: gpio
    id: relay
    pin: D2

light:
  - platform: binary
    id: light
    name: bar ledstrip
    output: relay
```

But this doesn't:
```yaml
output:
  - platform: esp8266_pwm
    id: dimchannel
    pin: D2
light:
  - platform: monochromatic
    name: bar ledstrip
    output: dimchannel
```

animavitis, 2019-04-29:
Have similar issue with nodeMCU and noticed interesting thing:

In my conifguration i have custom sensor component for emonlib: https://github.com/esphome/feature-requests/issues/96

If in customEmonlib.h file have line ` CustomEmonLib() : PollingComponent(15000) {}` then WiFi is working fine, but when i set it to ` CustomEmonLib() : PollingComponent(100) {}` then WiFi is not working at all.

Hope this info help

olealm, 2019-05-04:
I have the same problem with SonoffBasic's, but not consistant:
The first batch of Sonoffs I bought (without cables inside for the load current) has the same problem with 1.12.2, and I have to compile with the 'dev' version to get them to connect to WiFi.
The next batch I bought (newer edition I presume, with internal cables for the load current) works fine with 1.12.2
I have 6 Sonoff Sockets and they all work fine with 1.12.2.

RoadkillUK, 2019-05-12:
Just thought I'd put in my experience with this issue (I don't have a solution).

I use the same boards as the one pictured above and I'm getting similar issues.  I have a simple script that receives IR command, sets a timer and sends an IR command.

I have several of these boards to play with and get the same results from each one.

Here is the working script, remove the hashes for the non working version.

Sometimes I get 'No Network Found' other times I'll get a list of available WIFI networks then can't connect, the restart WIFI (rinse and repeat), occasionally it will connect, but not for long.

```
esphome:
  name: bedroomtv
  platform: ESP8266
  board: nodemcuv2
  on_boot:
    priority: 50
    then:
      - switch.turn_off: ledred

wifi:
  ssid: !secret wifi
  password: !secret wifi_password
  manual_ip:
    static_ip: 192.168.0.152
    gateway: 192.168.0.1
    subnet: 255.255.255.0
    dns1: 192.168.0.1
    dns2: 8.8.8.8

#remote_receiver:
#  pin:
#    number: D6
#    inverted: True
#  dump: nec
#  tolerance: 50
  
remote_transmitter:
  pin: D5
  carrier_duty_percent: 50%

#binary_sensor:
#  - platform: remote_receiver
#    name: "TV OK"
#    filters:
#      - delayed_off: 200ms  
#    nec:
#      address: 0x01FE
#      command: 0x59A6
#    on_press:
#      then:
#        - if:
#            condition:
#              switch.is_off: ledred
#            then:
#              - script.execute: mainscript   
script:
  - id: mainscript
    then:
      - switch.turn_on: ledred
      - delay: 2s
      - switch.turn_on: tvtoggle
      - delay: 40min
      - switch.turn_on: tvtoggle
      - switch.turn_off: ledred

switch:
  - platform: remote_transmitter
    name: "TV Power Toggle"
    id: tvtoggle
    nec:
      address: 0x01FE
      command: 0x43BC

  - platform: gpio
    name: "Bedroom TV LED Red"
    pin:
      number: D2
    id: ledred
#    internal: True

  - platform: gpio
    name: "Bedroom TV Remote"
    pin:
      number: D1
    id: bedroomtv
    on_turn_on:
      then:
        - script.execute: mainscript
        - delay: 1s
        - switch.turn_off: bedroomtv

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: 'bondie'

ota:
  password: 'bondie'
  
```

k1ng440, 2019-05-14:
defining network like below seems to help

```
wifi:
  networks:
    - ssid: xxx
      bssid: "xxxx"
      password: "xxx"
    - ssid: xxx
      bssid: "xxxx"
      password: "xxx"
    - ssid: xxx
      bssid: "xxxx"
      password: "xxx"

```

5310H, 2019-05-14:
Same problem on Sonoff SV.  Always works after the initial flash but always fails after powered off and restarted.  Always after reboot.  Acts like it can find wifi.  Have added fast_connect: true with same results.  Can replicate with same results.   Red status blinks about every 10 secs.  Logs show repeatedly scanning for Wifi.  Remember seeing (wifi 027) message.  Let me know what logs or tests you'd like me to forward.
Using 1.12.2  

Tested device with Tasmota with no issues.

5310H, 2019-05-16:
Installed 1.12.1, recompile code and can't replicate the issue anymore.   Will continue testing tomorrow.

OttoWinter, 2019-05-16:
@k1ng440 That's almost certainly not the issue here and I think it's just by chance that works for you - unless you also specified bssid with a single network before (in which case it would only connect to one network and if that was out of range to none). The warning message here happens before the network matching so impossible that's the issue here.

> Installed 1.12.1

There has been no major change to wifi between 1.12.2 and 1.12.1 - once again I believe this is just random.

The fact that this issue is so random (and I can't replicate it) makes it really hard to diagnose. With 90% of the solutions posted here (and in other issues) I'm certain it's just random chance. The core issue is somewhere in the ESP-SDK, but I don't have access to that.

k1ng440, 2019-05-16:
@OttoWinter i agree. 

> The core issue is somewhere in the ESP-SDK,
Tasmota and my custom code works perfectly fine

OttoWinter, 2019-06-18:
Moving the discussion to #455

closed 2019-06-18 by OttoWinter