esphome/issues #191: Compiling yaml is causing RPi to freeze

jdenver13, 2019-04-02:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

RPi3+ in wifi mode
**ESP (ESP32/ESP8266, Board/Sonoff):**
1.12.1 then upgraded to 1.12.2, still not working

**Affected component:**

**Description of problem:**
Been using ESPhome for a while now and all seems ok until I tried to compile a new .bin file and it just freezes, the RPi attempts to restart but then freezes, I loose access to SSH and access to the unifi add-on so I know its failed, the only way now is to recycle the power and delete the home-assistant_v2.db file as its now corrupt.

This was happening before upgrading to v1.12.2 on version 1.12.1

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```
substitutions:
  plug_name: smartplug_4
esphome:
  name: ${plug_name}
  platform: ESP8266
  board: esp01_1m
wifi:
  ssid: 'password'
  password: 'password111'
logger:
  baud_rate: 0
uart:
  tx_pin: TX
  rx_pin: RX
  baud_rate: 4800
web_server:
  port: 80
api:
  password: 'password222'
ota:
  password: 'password333'
time:
- platform: homeassistant
  id: homeassistant_time
binary_sensor:
  - platform: gpio
    id: button
    pin:
      number: GPIO12
      mode: INPUT_PULLUP
      inverted: True
    on_press:
      - switch.toggle: relay
switch:
- platform: gpio
  name: "${plug_name}_Relay"
  id: relay
  pin: GPIO14
  restore_mode: ALWAYS_ON
sensor:
- platform: wifi_signal
  name: "${plug_name} WiFi"
- platform: adc
  pin: VCC
  name: "VCC Voltage"
- platform: cse7766
  voltage:
    name: "${plug_name}_Voltage"
  current:
    name: "${plug_name}_Amperage"
  power:
    name: "${plug_name}_Wattage"
    id: "${plug_name}_Wattage"
- platform: total_daily_energy
  name: "${plug_name}_Total Daily Energy"
  power_id: "${plug_name}_Wattage"
  filters:
    - multiply: 0.001
  accuracy_decimals: 2
  unit_of_measurement: kWh
- platform: uptime
  name: ${plug_name}_Uptime Sensor
  filters:
    - lambda: return (x/60.0)/60;
  unit_of_measurement: "H"
**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

Rebooted RPi and restarted ESPHome, also tried a clean build. I have 3 other smart plugs using the same .yaml (different names) and where they had all compiled and flashed over the air, they now no longer get through the compiling stage.

I think this all seemed to happen after following this issue: https://github.com/esphome/issues/issues/143 but I can't be sure as I was having issues with SSL logging in locally and infulxdb add-on at the time

[image]
[image]

Stretcherfetcher, 2019-04-02:
+1 for me also, freezing at the same spot. Using esphome 1.12.2 and hassio 90.2.

brandond, 2019-04-02:
Sounds like either your power supply can't stand up to the load of the CPU running an intensive task like compiling code, or you're running out of memory.

jdenver13, 2019-04-02:
The PSU is the RPi 2.5A one, the cpu did run hot so the last few compiles have been done with a fun running and the cpu stays cool.
How would I check on memory usage?

brandond, 2019-04-02:
Log in via shell and run 'top' while compiling would be an easy way.

jdenver13, 2019-04-02:
I follow what you're saying regarding volt drop under load and that could be the case, but this has only just started and Ive complied and flashed many times without a problem, but now every time I compile this problem occurs. The PSU is only 6 months old and doesn't run warm at all.

jdenver13, 2019-04-02:
Just run 'top' and this is the results:
```
Mem: 938308K used, 10060K free, 632K shrd, 30456K buff, 101556K cached
CPU:   2% usr   0% sys   0% nic  96% idle   0% io   0% irq   0% sirq
Load average: 0.41 0.36 0.35 2/741 27
  PID  PPID USER     STAT   VSZ %VSZ CPU %CPU COMMAND
   23     7 root     S     2932   0%   1   0% sshd: root@pts/0
    7     1 root     S     2932   0%   2   0% /usr/sbin/sshd -D -e
   25    23 root     S     2204   0%   1   0% -bash
   27    25 root     R     1240   0%   2   0% top
    1     0 root     S      708   0%   0   0% /dev/init -- /run.sh
```

brandond, 2019-04-02:
You need to run top to monitor it while compiling, as the output refreshes every couple seconds. You should see a few compiler processes up at the top of the list while it's running. What's the last thing you see before it locks up?

jdenver13, 2019-04-03:
Ive recorded a screen recording and split it into 1 ½ minute files to show what's happening:
[screen_recording_1_sd.mov.zip](https://github.com/esphome/issues/files/3037519/screen_recording_1_sd.mov.zip)

[screen_recording_2_sd.mov.zip](https://github.com/esphome/issues/files/3037524/screen_recording_2_sd.mov.zip)

Hope it comes out alright.

Stretcherfetcher, 2019-04-03:
[image]

Thats it. Raspberry Pi 3 Model B+ max'ed out. Looks like I'm asking too much from it. Here is my container list......

[image]

Is a NUC the way to go? 

Also I've tried to install Esphome on my MAC but was unsuccessful, is it possible. Apologies for my Noobness and thanks for the help, great to have such a wealth of talent for us Noobs.

jdenver13, 2019-04-09:
Just purchased a NUC now so will see how it goes

Stretcherfetcher, 2019-04-09:
Yeah bought a secondhand NUC with an i3 and 80gb ssd with 4gigs ram and used this guide to set up 

https://community.home-assistant.io/t/absolute-beginners-guide-to-installing-ha-on-intel-nuc-using-docker/98412
Works great no problem compiling and HA's responses are much quicker. Best of luck.

OttoWinter, 2019-04-16:
Ok I will close this issue then. ESPHome just needs a lot of resources *because* it compiles custom binaries - a byproduct of esphome's base design and thus won't change.

closed 2019-04-16 by OttoWinter