esphome/issues #192: AM2320 sensor not working (but found at 0x5c address)

sergeolkhovik, 2019-04-02:
First, gentlemen, I want to say that I'm really impressed about this platform. Amazing!
I had some self-developed code but now migrating to HA + ESPHome. And got some issues with AM2320 sensor.

**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Latest 1.12.2, PIP installation.

**ESP (ESP32/ESP8266, Board/Sonoff):**

Have 2 the same Chinees WeMos D1 devices, one runs NodeMCU (custom build with am2320 support, https://nodemcu.readthedocs.io/en/master/en/modules/am2320/), another one runs ESPHome, both have (again, Chineese) AM2320 sensors (the same order, https://www.aliexpress.com/item/1pcs-New-AM2320-Digital-Temperature-and-Humidity-Sensor-Original-authentic-Can-replace-SHT20-SHT10/32593087118.html?spm=a2g0s.9042311.0.0.27424c4dKmnhSx), both chips have sensor connected via pins GPIO4/5 (D2/D1), and both have OLED 128x64 displays. Currently don't have any pull-up resistor on i2c bus (but I might have already tried before with no luck, cannot recall).

**Affected component:**

DHT12 T+H sensor.

**Description of problem:**
ESPHome marks am2320 sensor failed (but can find this device at 0x5c), nodemcu installation works perfectly.

**Problem-relevant YAML-configuration entries:**
```yaml
i2c:
    sda: 4
    scl: 5
    scan: True

sensor:
    - platform: dht12
      temperature:
          id: air_t
          name: D3 Air T
          icon: "mdi:temperature-celsius"
          filters:
              - offset: -1.2
      humidity:
          id: air_h
          name: D3 Air H
          filters:
              - offset: -7
      update_interval: 60s

```

**Traceback (if applicable):**

```
[19:42:55][C][i2c:035]: I2C Bus:
[19:42:55][C][i2c:036]:   SDA Pin: GPIO4
[19:42:55][C][i2c:037]:   SCL Pin: GPIO5
[19:42:55][C][i2c:038]:   Frequency: 50000 Hz
[19:42:55][I][i2c:040]: Scanning i2c bus for active devices...
[19:42:55][I][i2c:047]: Found i2c device at address 0x3C
[19:42:55][I][i2c:047]: Found i2c device at address 0x5C
[19:42:55][C][output.gpio:025]: GPIO Binary Output:
[19:42:55][D][sensor.dht12:057]: DHT12:
[19:42:55][C][sensor.dht12:058]:   Address: 0x5C
[19:42:55][E][sensor.dht12:060]: Communication with DHT12 failed!
[19:42:55][C][sensor.dht12:062]:   Temperature 'D3 Air T'
[19:42:55][C][sensor.dht12:062]:     Unit of Measurement: '°C'
[19:42:55][C][sensor.dht12:062]:     Accuracy Decimals: 1
[19:42:55][C][sensor.dht12:062]:     Icon: 'mdi:temperature-celsius'
[19:42:55][C][sensor.dht12:063]:   Humidity 'D3 Air H'
[19:42:55][C][sensor.dht12:063]:     Unit of Measurement: '%'
[19:42:55][C][sensor.dht12:063]:     Accuracy Decimals: 0
[19:42:55][C][sensor.dht12:063]:     Icon: 'mdi:water-percent'
```

**Additional information and things you've tried:**
But when I connect esp via USB I see extra info at the very beginning:

```
[D][light.state:418]: 'D3 Built-in LED' Turning OFF
[C][display.ssd1306:299]: Setting up I2C SSD1306...
[C][sensor.dht12:049]: Setting up DHT12...
[W][i2c:077]: Received NACK on transmit of address 0x5C
[W][sensor.dht12:068]: Updating DHT12 failed!
[E][component:151]: Component was marked as failed.
```

The most important thing here is that alternative configuration (NodeMCU) works without any problem, I can get some values and display them on display, code is pretty simple like:

```
    i2c.setup(0, PIN_SDA, PIN_SCL, i2c.SLOW)
    tmr.delay(2000000) -- 2 seconds
    am2320.setup()
...

        h, t = am2320.read()
        h = h / 10
        t = t / 10
        msg = string.format("appSensors: t = %0.3f, h = %0.3f", t, h);
        print(msg);

```

Oh, wait... I just found (and I really forgot about that!) that I have some delay set between i2c initialization and am2302 sensor set up. Probably this is because Chineese origin of these sensors, is there a way to implement something like this is ESPHome configuration?

Please ask me if you need any extra info/details or debugging from me to figure out the problem.
Highly appreciate any help on this.
Thanks!

OttoWinter, 2019-04-09:
Yes the initial delay could be the culprit here (according to the datasheet esphome's timing is ok though). You can add a delay by setting
```yaml
sensor:
- platform: dht12
  setup_priority: -100
  # ...
```

sergeolkhovik, 2019-04-09:
Unfortunately that didn't work :(
```
[21:16:52][D][sensor.dht12:057]: DHT12:
[21:16:52][C][sensor.dht12:058]:   Address: 0x5C
[21:16:52][E][sensor.dht12:060]: Communication with DHT12 failed!
[21:16:52][C][sensor.dht12:062]:   Temperature 'D3 Air T'
[21:16:52][C][sensor.dht12:062]:     Unit of Measurement: '°C'
[21:16:52][C][sensor.dht12:062]:     Accuracy Decimals: 1
[21:16:52][C][sensor.dht12:062]:     Icon: 'mdi:temperature-celsius'
[21:16:52][C][sensor.dht12:063]:   Humidity 'D3 Air H'
[21:16:52][C][sensor.dht12:063]:     Unit of Measurement: '%'
[21:16:52][C][sensor.dht12:063]:     Accuracy Decimals: 0
[21:16:52][C][sensor.dht12:063]:     Icon: 'mdi:water-percent'
```
I even tried -1000.
Here's my full setup if that might help:
```
esphome:
  name: d3
  platform: ESP8266
  board: d1_mini
  on_boot:
      then:
        light.turn_off: builtin_led

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_ssid
  domain: !secret wifi_domain

# Enable logging
logger:

# Enable Home Assistant API
api:
    password: !secret api_password

ota:
    password: !secret ota_password

i2c:
    sda: 4
    scl: 5
    #scan: false
    frequency: 200000

time:
    - platform: homeassistant
      id: time

sensor:

    - platform: dht12
      setup_priority: -1000
      temperature:
          id: air_t
          name: D3 Air T
          icon: "mdi:temperature-celsius"
          filters:
              - offset: -1.2
      humidity:
          id: air_h
          name: D3 Air H
          filters:
              - offset: -7
      update_interval: 60s
    - platform: homeassistant
      id: out_t
      name: "HA Outdoor T"
      entity_id: sensor.dark_sky_temperature
    - platform: homeassistant
      id: out_h
      name: "HA Outdoor H"
      entity_id: sensor.dark_sky_humidity
    - platform: homeassistant
      id: wind_speed
      entity_id: sensor.dark_sky_wind_speed

text_sensor:
    - platform: homeassistant
      id: wind_dir
      entity_id: sensor.dark_sky_wind_dir

output:
    - platform: gpio
      pin: D4
      id: builtin_led_output
      inverted: true

light:
    - platform: binary
      id: builtin_led
      name: D3 Built-in LED
      output: builtin_led_output

interval:
    - interval: 5s
      then:
          - display.page.show_next: display
          - component.update: display

font:
  - file: '../fonts/silkscreen/slkscr.ttf'
    id: fontS
    size: 8

  - file: '../fonts/corefonts/arialbd.ttf'
    id: arialb12
    size: 12

  - file: '../fonts/corefonts/verdanab.ttf'
    id: verdanb12
    size: 12

  - file: '../fonts/Bebas_Neue/BebasNeue-Regular.ttf'
    id: fontBN36
    size: 36

  - file: '../fonts/Bebas_Neue/BebasNeue-Regular.ttf'
    id: fontBN64
    size: 64

display:
  - platform: ssd1306_i2c
    id: display
    model: "SH1106 128x64"
    #reset_pin: D0
    address: 0x3C
    pages:
        - id: page_indoor
          lambda: |-
              it.printf(0, 0, id(fontS), TextAlign::TOP_LEFT, "Indoor:");
              if (id(air_t).has_state()) {
                  it.printf(32, 16, id(fontBN36), TextAlign::TOP_CENTER , "%.1f°", id(air_t).state);
              }
              if (id(air_h).has_state()) {
                  it.printf(96, 16, id(fontBN36), TextAlign::TOP_CENTER , "%02.0f%%", id(air_h).state);
              }
        - id: page_outdoor
          lambda: |-
              it.printf(0, 0, id(fontS), TextAlign::TOP_LEFT, "Outdoor:");
              if (id(out_t).has_state()) {
                  it.printf(32, 16, id(fontBN36), TextAlign::TOP_CENTER , "%.1f°", id(out_t).state);
              }
              if (id(out_h).has_state()) {
                  it.printf(96, 16, id(fontBN36), TextAlign::TOP_CENTER , "%02.0f%%", id(out_h).state);
              }
              if (id(wind_speed).has_state()) {
                  // it.printf(127, 0, id(arialb12), TextAlign::TOP_RIGHT , "%.0f%s %s", id(wind_speed).state, id(w
ind_speed).unit_of_measurement.c_str(), id(wind_dir).state.c_str());
                  it.printf(127, 0, id(arialb12), TextAlign::TOP_RIGHT , "%.1f %s", id(wind_speed).state, id(wind_d
ir).state.c_str());
              }
        - id: page_time
          lambda: |-
              // Print time in HH:MM format
              it.strftime(0, 60, id(fontBN64), TextAlign::BASELINE_LEFT, "%H:%M", id(time).now());
```

Played with i2c scan and frequency options, but with no luck.
What can I do else?

AlexDanault, 2019-04-30:
I have the same issue in #278 which you can close and we'll continue here. If you have a branch with the proposed delay I'll be more than happy to test and report back.

nettosama, 2019-05-10:
Did anyone test the dev branch to see if it's fixed or not?

AlexDanault, 2019-05-10:
> 
> 
> Did anyone test the dev branch to see if it's fixed or not?

There are no commits referencing this issue, I don't think anything has been tried.

@OttoWinter if you need a couple of AM2320 for testing I will gladly mail you some. Or if you want to try adding a delay in the code I'll be happy to test it. In all cases, thanks !

nettosama, 2019-05-15:
OK, did some testing, and found that you need a different library/code to work with the 2320 sensor. I did a rough modification of the current esphome DH12 .cpp file to mimic the specification in [https://github.com/EngDial/AM2320](url), that follows what's specified in [https://cdn-shop.adafruit.com/product-files/3721/AM2320.pdf](url), and it works. Maybe someone can make a well put together pull request?

T3m3z, 2019-05-18:
I can confirm that I have too AM2320 sensor that does not work with the ESPHome DHT12 component. And I can also confirm that the library that I previously used, seems to communicate with AM2320 in the manner as @nettosama has pointed in his comment.

It seems that we need to create a new component for these AM2320 sensors. Or would it make more sense to read the model of the sensor over I2C and act accordingly?

@nettosama do you have some code to share? I also did some "rough modifications" on the DHT12 component, but my code seems quite unrealiable. I can read the sensor only once or twice (reasonable readings) but then the sensor just returns 0 for both temperature and humidity and I can't figure out why. Possibly due to the fact that it is already 11:30 PM here :)

nettosama, 2019-05-18:
Mine has been working fine with this code, you would need to create a am2320 folder with the needed sensor.py and .h files:

```cpp
// Implementation based on:
//  - ESPEasy: https://github.com/letscontrolit/ESPEasy/blob/mega/src/_P034_DHT12.ino
//  - DHT12_sensor_library: https://github.com/xreef/DHT12_sensor_library/blob/master/DHT12.cpp
//  - Arduino - AM2320: https://github.com/EngDial/AM2320/blob/master/src/AM2320.cpp

#include "am2320.h"
#include "esphome/core/log.h"

namespace esphome {
namespace am2320 {

static const char *TAG = "am2320";

// ---=== Calc CRC16 ===---
uint16_t CRC16(uint8_t *ptr, uint8_t length)
{
uint16_t crc = 0xFFFF;
uint8_t i;
//------------------------------
  while(length--)
  {
    crc ^= *ptr++;
    for (i = 0; i < 8; i++)
      if ((crc & 0x01) != 0)
      {
        crc >>= 1;
        crc ^= 0xA001;
      }
      else
        crc >>= 1;
  }
  return crc;
}

void AM2320Component::update() {
  uint8_t data[8];
  data[0] = 0;
  data[1] = 4;
  if (!this->read_data_(data)) {
    this->status_set_warning();
    return;
  }

  float temperature = (((data[4] & 0x7F) << 8) + data[5]) / 10.0;
  temperature = (data[4] & 0x80) ? -temperature : temperature;
  float humidity = ((data[2] << 8) + data[3]) / 10.0;

  ESP_LOGD(TAG, "Got temperature=%.2f°C humidity=%.2f%%", temperature, humidity);
  if (this->temperature_sensor_ != nullptr)
    this->temperature_sensor_->publish_state(temperature);
  if (this->humidity_sensor_ != nullptr)
    this->humidity_sensor_->publish_state(humidity);
  this->status_clear_warning();
}
void AM2320Component::setup() {
  ESP_LOGCONFIG(TAG, "Setting up AM2320...");
  uint8_t data[8];
  data[0] = 0;
  data[1] = 4;
  if (!this->read_data_(data)) {
    this->mark_failed();
    return;
  }
}
void AM2320Component::dump_config() {
  ESP_LOGD(TAG, "AM2320:");
  LOG_I2C_DEVICE(this);
  if (this->is_failed()) {
    ESP_LOGE(TAG, "Communication with AM2320 failed!");
  }
  LOG_SENSOR("  ", "Temperature", this->temperature_sensor_);
  LOG_SENSOR("  ", "Humidity", this->humidity_sensor_);
}
float AM2320Component::get_setup_priority() const { return setup_priority::DATA; }

bool AM2320Component::read_bytes_(uint8_t a_register, uint8_t *data, uint8_t len, uint32_t conversion) {

  if (!this->write_bytes(a_register, data, 2)) {
    ESP_LOGW(TAG, "Writing bytes for AM2320 failed!");
    return false;
  }

  if (conversion > 0)
    delay(conversion);
  return this->parent_->raw_receive(this->address_, data, len);
}

bool AM2320Component::read_data_(uint8_t *data) {
  // Wake up
  this->write_bytes(0, data, 0);

  // Write instruction 3, 2 bytes, get 8 bytes back (2 preamble, 2 bytes temperature, 2 bytes humidity, 2 bytes CRC)
  if (!this->read_bytes_(3, data, 8, 2)) {
    ESP_LOGW(TAG, "Updating AM2320 failed!");
    return false;
  }

  uint16_t checksum;

  checksum = data[7] << 8;
  checksum += data[6];

  if (CRC16(data, 6) != checksum) {
    ESP_LOGW(TAG, "AM2320 Checksum invalid!");
    return false;
  }

  return true;
}

}  // namespace am2320
}  // namespace esphome
```

T3m3z, 2019-05-20:
I can confirm that your code works. So isn't this like almost ready for a pull request @nettosama ? I didn't quite catch what you were after when you wrote about a pull request. Do you need some help or were you just waiting for someone to confirm that your code works?

nettosama, 2019-05-21:
That comment was before I wrote that code.

If anyone thinks it's good enough, I wouldn't mind seeing it as a pull request, I just can't write it right now.

T3m3z, 2019-05-22:
I did some preliminary work and I will create pull requests tomorrow against esphome and esphome-docs. Thank you @nettosama for your work!

closed 2019-05-27 by OttoWinter