esphome/issues #193: ESPHome device always "unavailable" after Home Assistant server restart

ptmoy2, 2019-04-03:
Operating environment (Hass.io/Docker/pip/etc.):

Hass.io latest v152 running on Raspberry Pi2, Home Assistant version 0.90.2

ESP (ESP32/ESP8266/Board/Sonoff):

Sonoff SV

Affected component:

?

Description of problem:
I am using 3 Sonoff SVs with ESPHome to control 3 garage doors. Two of the devices work without problems.  The third device works without problems most of the time.  However, whenever I restart the Home Assistant server, that third device always becomes "unavailable" in Home Assistant after the restart.  A reboot of the Raspberry Pi running Hassio always remedies the situation (that is until the next server restart).  Also, the problem device always shows a weaker Wifi signal than the other two even though the 3 devices are physically located next to each other. 

Problem-relevant YAML-configuration entries:

esphome:
  name: center_garage_door
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: 'mySsid
  password: 'myPassword'

# Enable logging
logger:

api:

ota:
  password: 'myOTApassword'

switch:
  - platform: gpio
    name: "Center Garage Door Relay"
    pin: GPIO12
    id: center_garage_door_relay
    
  - platform: template
    name: "Center Garage Door Remote"
    icon: "mdi:gate"
    id: garage_remote
    turn_on_action:
    - switch.turn_on: center_garage_door_relay
    - delay: 1000ms
    - switch.turn_off: center_garage_door_relay
    
binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    name: "Sonoff SV Button"
    on_press:
      - switch.toggle: garage_remote
    
  - platform: gpio
    pin:
      number: GPIO14
      mode: INPUT_PULLUP
    name: "Center Garage Door Status"
    id: center_garage_door_stat
    
sensor:
  - platform: wifi_signal
    name: "Center Garage Door WiFi Signal Strength"
    update_interval: 60s
    
cover:
  - platform: template
    name: "Center Garage Door"
    lambda: !lambda |-
      if (id(center_garage_door_stat).state) {
        return cover::COVER_OPEN;
      } else {
        return cover::COVER_CLOSED;
      }
    open_action:
      - switch.turn_on: garage_remote
    stop_action:
      - switch.turn_on: garage_remote
    close_action:
      - switch.turn_on: garage_remote
   

Traceback (if applicable):

Home Assistant logs:

ESP logs (via USB):

Additional information:
The Yaml file for the two working Sonoff SVs are almost identical to the one shown above, except the other two files contain additional sensors.  The ESPHome dashboard logs show device is up and running fine with no errors.

closed 2019-04-03 by close-issue-app[bot]

ptmoy2, 2019-04-03:
I thought I did filled out the template.  Be nice if someone points out what I'm doing wrong to have my issue closed automatically.

mr-sneezy, 2019-04-03:
Me too. I've looked but don't see a template or any advice on the root page...