esphome/issues #198: No device type selectable in dashboard wizard.

mr-sneezy, 2019-04-03:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

HASSOS 2.11
Raspberry Pi3
HA 0.90.2

**ESP (ESP32/ESP8266, Board/Sonoff):**

NodeMCU V3 - plugged into RPi3 USB port.

**Affected component:**

ESPhome Dashboard Wizard
https://esphome.io/guides/getting_started_hassio.html#installation

**Description of problem:**
Wizard is skipping the Device Type option step when using the GUI.
In the wizard I can not select a device board as directed by the tutorials, as the GUI skips over this step every attempt. I can see the attached NodeMCU in the serial connection drop down as /dev/ttyUSB0 (USB2.0-Serial) when I plug it in and restart ESPhome add-on
I've updated to the latest HASSOS and rebooted but no change. HA is up to date also.
I am suspicious I might be using the 64bit version of HASSIO, but I can't find any indication in HA Info page or HASSIO system page as to which one I installed some time ago.

Anybody either know where I can find out if my HASSIO is 32bit or 64bit on the RPi3 ?
Or know why device selection is skipped ?

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```
Not yet achieved
**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

nickrout, 2019-04-03:
For the 32/64 bit question, ssh in and run

    uname -r

mr-sneezy, 2019-04-05:
Sadly I've been unable to run that command as suggested. Spent now some 5 hours over two nights trying to SSH into the Pi from Windows with Putty and also Hassio SSH & Web terminal component. Connection refused every time.
FIH

mr-sneezy, 2019-04-06:
I've found the image file I downloaded for hassio on my PC hard drive and it does NOT seem to be the 64bit one, so I guess my install is the 32bit. 
I have given up trying the USB over RPi method and had partial success by compiling the binary and using esphomeflasher to write to the NodeMCU.

OttoWinter, 2019-04-07:
The dashboard skipping device type has absolutely nothing to do with the OS it's running on.

The device type screen is completely handled by your browser. Please say what browser you're using and what version of it. If it's MS edge, try with chrome.

mr-sneezy, 2019-04-08:
Google Chrome
Version 73.0.3683.103 (Official Build) (64-bit)

My concern with OS was derived from the warning about compatibility issues for 64 bit HassOS somewhere in my reading up last week.

Ota works now after direct flashing so this issue is not of any concern now thanks.

meichthys, 2019-04-10:
@OttoWinter  I believe we can close this issue.

closed 2019-04-15 by OttoWinter