esphome/issues #200: Error while toggle switch on remote device Teckin Sp22

hubgitbb, 2019-04-03:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
HassIO running on Docker, ESPHome addon running in HassIO

**ESP (ESP32/ESP8266, Board/Sonoff):**
Teckin SP22 switch, ESP8266, running ESPHome v 1.11.2

**Affected component:**
https://esphome.io/guides/getting_started_hassio.html

**Description of problem:**
i have two teckin sp22 power plug (named p1 and p2). if i push the button on plug p1 i want that on plug p2 the power switch shuld toggle on/off.

this doesent work:
on_press:
then:
switch.toggle: p2_Relay

Couldn't find ID 'p2_Relay'

if i try this with the local device button it works.

**Problem-relevant YAML-configuration entries:**
```yaml
substitutions:
plug_name: p1
Higher value gives lower watt readout
current_res: "0.00182"
Lower value gives lower voltage readout
voltage_div: "788"

esphome:
name: ${plug_name}
platform: ESP8266
board: esp8285

wifi:
ssid: 'myssid'
password: '********'
Enable logging

logger:
Enable Web server

web_server:
port: 80
Enable Home Assistant API

api:
password: '*********'

ota:
password: '********'

time:
platform: homeassistant
id: homeassistant_time

binary_sensor:
platform: gpio
pin:
number: GPIO1
inverted: True
name: "${plug_name}_button"
on_press:
then:
switch.toggle: p2_Relay

switch:
platform: gpio
name: "${plug_name}_Relay"
id: "${plug_name}_Relay"
pin: GPIO14
restore_mode: ALWAYS_ON
platform: gpio
name: "${plug_name}_LED_Blue"
pin: GPIO13
inverted: True
restore_mode: ALWAYS_OFF
platform: gpio
name: "${plug_name}_LED_Red"
pin: GPIO3
inverted: True
restore_mode: ALWAYS_OFF
sensor:
platform: hlw8012
sel_pin:
number: GPIO12
inverted: True
cf_pin: GPIO04
cf1_pin: GPIO05
current_resistor: ${current_res}
voltage_divider: ${voltage_div}
current:
name: "${plug_name}_Amperage"
unit_of_measurement: A
voltage:
name: "${plug_name}_Voltage"
unit_of_measurement: V
power:
name: "${plug_name}_Wattage"
unit_of_measurement: W
id: "${plug_name}_Wattage"
change_mode_every: 8
update_interval: 10s
platform: total_daily_energy
name: "${plug_name}_Total Daily Energy"
power_id: "${plug_name}_Wattage"
filters:
# Multiplication factor from W to kW is 0.001
- multiply: 0.001
unit_of_measurement: kWh

Extra sensor to keep track of plug uptime

platform: uptime
name: ${plug_name}_Uptime Sensor

```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

redstar2k1, 2019-04-04:
You cannot directly trigger the relay from one plug to the other.
You need to setup and use MQTT or other form of communication between the two plugs.

hubgitbb, 2019-04-04:
i use gpslogger with mqtt but i have no idea on this one. can you point me to some infos or examples on how i can try this out?

OttoWinter, 2019-04-04:
Exactly, these switch.turn_on etc actions are only for devices that are on the device itself.

Since you appear to be using `api`, you need to use `homeassistant.service` to control the devices via Home Assistant with their entity ID.

closed 2019-04-04 by OttoWinter