esphome/issues #202: Validation of esp32_camera fails with a supported resolution

dattas, 2019-04-04:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io, homeassistant 0.91.0
Esphome 1.12.2
**ESP (ESP32/ESP8266, Board/Sonoff):**

m5stack camera module. ESP32, m5stack-coreesp32

**Affected component:**

https://esphome.io/components/esp32_camera.html

**Description of problem:**
Pulling a resolution listed in the documentation says it isn't valid

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: test_camera
  platform: ESP32
  board: m5stack-core-esp32

<<: !include 00-common/wifi.yaml

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: 'none'

ota:
  password: 'none'

esp32_camera:
  name: Test Camera
  external_clock:
    pin: GPIO27
    frequency: 20MHz
  i2c_pins:
    sda: GPIO25
    scl: GPIO23
  data_pins: [GPIO17, GPIO35, GPIO34, GPIO5, GPIO39, GPIO18, GPIO36, GPIO19]
  vsync_pin: GPIO22
  href_pin: GPIO26
  pixel_clock_pin: GPIO21
  reset_pin: GPIO15
  resolution: 1280x1024

```

**Traceback (if applicable):**

```
INFO Reading configuration...
Failed config

esp32_camera: [source /config/esphome/test_camera.yaml:18]
  name: Test Camera
  external_clock: 
    pin: GPIO27
    frequency: 20MHz
  i2c_pins: 
    sda: GPIO25
    scl: GPIO23
  data_pins: 
    - GPIO17
    - GPIO35
    - GPIO34
    - GPIO5
    - GPIO39
    - GPIO18
    - GPIO36
    - GPIO19
  vsync_pin: GPIO22
  href_pin: GPIO26
  pixel_clock_pin: GPIO21
  reset_pin: GPIO15
  
  Unknown value '1280X1024', must be one of 'CIF', '160X120', '640X480', 'SVGA', 'HQVGA', 'SXGA', '400X296', 'QVGA', '240X176', 'QQVGA', 'QCIF', 'VGA', '1600X1200', '320X240', 'UXGA', '128x160', 'XGA', '1024X768', '800X600', '176X144', 'QQVGA2', '1280x1024'. Got '1280x1024'
  resolution: 1280x1024
```

**Additional information and things you've tried:**
Using `SXGA` instead does seem to work properly. If I had to guess it is because it seems to be upper casing the value before testing and the valid values have some lower case in them. I'd imagine 128x160 would have this same problem. https://github.com/esphome/esphome/blob/dev/esphome/components/esp32_camera.py#L32 if I had to guess as to a line to fix this.

dattas, 2019-04-08:
Looks as though https://github.com/esphome/esphome/blob/camera-validation/esphome/components/esp32_camera.py#L17 is still an issue, although the above commit did fix the primary issue I found.

OttoWinter, 2019-04-25:
Fixed in dev. (also second issue, thanks for reporting that again!)

closed 2019-04-25 by OttoWinter