esphome/issues #203: High cpu usage and slow progress when compiling

john-arvid, 2019-04-05:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io on generic linux server
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266
**Affected component:**

Compiling

**Description of problem:**
When I want to compile a new or old yaml file it causes the cpu to use 100% and the compilation is very slow.
I have currently 6 cores on 2.83ghz available. Also 6Gb ram.

Also if I try to stop the process the cpu is still 100% active. I need to restart the esphome in hassio for it to relax again.

There is nothing in the logs for esphome, home assistant or hassio.

Maybe this is the same problem as this:  #191
But since I have more resources it actually get's somewhere. Just very slow.

Python is the process that is using all of the cpu.

OttoWinter, 2019-04-07:
> There is nothing in the logs for esphome, home assistant or hassio.

Really _nothing_ ? Please paste all logs here (compile log, addon log) even if they look like they contain only basic information.

High CPU usage during a compile is expected, the number of compile processes is automatically scaled with the number of available cores.

john-arvid, 2019-04-07:
Here is the addon log:

```[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] 00-banner.sh: executing... 

-----------------------------------------------------------
 Hass.io Add-on: ESPHome
 ESPHome Hass.io add-on for intelligently managing all your ESP8266/ESP32 devices.
-----------------------------------------------------------
 Add-on version: 1.12.2
 You are running the latest version of this add-on.
 System: Debian GNU/Linux 9 (stretch)  (amd64 / qemux86-64)
 Home Assistant version: 0.90.1
 Supervisor version: 152
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in e.g, GitHub, forums or the Discord chat.
-----------------------------------------------------------
[cont-init.d] 00-banner.sh: exited 0.
[cont-init.d] 01-log-level.sh: executing... 
[cont-init.d] 01-log-level.sh: exited 0.
[cont-init.d] 10-requirements.sh: executing... 
[cont-init.d] 10-requirements.sh: exited 0.
[cont-init.d] 20-nginx.sh: executing... 
[cont-init.d] 20-nginx.sh: exited 0.
[cont-init.d] 30-esphome.sh: executing... 
[cont-init.d] 30-esphome.sh: exited 0.
[cont-init.d] 40-migrate.sh: executing... 
[cont-init.d] 40-migrate.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[10:42:17] INFO: Starting NGINX...
[10:42:18] INFO: Starting ESPHome dashboard...
2019/04/05 10:42:19 [crit] 425#425: *1 connect() to unix:/var/run/esphome.sock failed (2: No such file or directory) while connecting to upstream, client: 192.168.1.10, server: hassio.local, request: "GET /serial-ports HTTP/1.1", upstream: "http://unix:/var/run/esphome.sock:/serial-ports", host: "192.168.1.20:6052", referrer: "http://192.168.1.20:6052/"
INFO Starting dashboard web server on unix socket /var/run/esphome.sock and configuration dir /config/esphome...
INFO 304 GET / (0.0.0.0) 30.05ms
INFO 304 GET /static/materialize.min.css?hash=ec1df3ba (0.0.0.0) 43.72ms
INFO 304 GET /static/materialize-stepper.min.css?hash=a7ad7ee2 (0.0.0.0) 0.84ms
INFO 304 GET /static/esphome.css?hash=ad6176a4 (0.0.0.0) 0.81ms
INFO 304 GET /static/jquery.min.js?hash=2f6b11a7 (0.0.0.0) 1.07ms
INFO 304 GET /static/jquery-ui.min.js?hash=a0b1425d (0.0.0.0) 1.22ms
INFO 304 GET /static/materialize.min.js?hash=87d84bf8 (0.0.0.0) 1.16ms
INFO 304 GET /static/jquery.validate.min.js?hash=24ae1ca6 (0.0.0.0) 0.77ms
INFO 304 GET /static/materialize-stepper.min.js?hash=9f34c4ff (0.0.0.0) 0.73ms
INFO 304 GET /static/ace.js?hash=8599fc94 (0.0.0.0) 1.65ms
INFO 304 GET /static/esphome.js?hash=bc135b8c (0.0.0.0) 0.77ms
INFO 304 GET /static/theme-dreamweaver.js (0.0.0.0) 20.74ms
INFO 304 GET /static/mode-yaml.js (0.0.0.0) 2.22ms
INFO 101 GET /compile (0.0.0.0) 0.85ms
INFO Running command 'esphome --dashboard /config/esphome/test.yaml compile'
```

And this is the compile log, I stopped here cause I did not want to stress my cpu that much.

```
INFO Reading configuration...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/test
Processing test (platform: espressif8266@1.8.0; framework: arduino; board: esp01_1m)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp01_1m.html
PLATFORM: Espressif 8266 > Espressif Generic ESP8266 ESP-01 1M
HARDWARE: ESP8266 80MHz 80KB RAM (1MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 27 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncTCP> 1.2.0
|   |-- <ESP8266WiFi> 1.0
|-- <Hash> 1.0
|-- <esphome-core> 1.12.2
|   |-- <ArduinoJson-esphomelib> 5.13.3
|   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266mDNS>
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESPAsyncTCP> 1.2.0
|   |   |-- <ESP8266WiFi> 1.0
Compiling /data/test/.pioenvs/test/src/main.cpp.o
Generating LD script /data/test/.pioenvs/test/ld/eagle.app.v6.common.ld
Compiling /data/test/.pioenvs/test/lib4d9/ESP8266WiFi/BearSSLHelpers.cpp.o
Compiling /data/test/.pioenvs/test/lib4d9/ESP8266WiFi/CertStoreBearSSL.cpp.o
Compiling /data/test/.pioenvs/test/lib4d9/ESP8266WiFi/ESP8266WiFi.cpp.o
Compiling /data/test/.pioenvs/test/lib4d9/ESP8266WiFi/ESP8266WiFiAP.cpp.o
Compiling /data/test/.pioenvs/test/lib4d9/ESP8266WiFi/ESP8266WiFiGeneric.cpp.o
Compiling /data/test/.pioenvs/test/lib4d9/ESP8266WiFi/ESP8266WiFiMulti.cpp.o
Compiling /data/test/.pioenvs/test/lib4d9/ESP8266WiFi/ESP8266WiFiSTA.cpp.o
Compiling /data/test/.pioenvs/test/lib4d9/ESP8266WiFi/ESP8266WiFiScan.cpp.o
Compiling /data/test/.pioenvs/test/lib4d9/ESP8266WiFi/WiFiClient.cpp.o
Compiling /data/test/.pioenvs/test/lib4d9/ESP8266WiFi/WiFiClientSecureAxTLS.cpp.o
Compiling /data/test/.pioenvs/test/lib4d9/ESP8266WiFi/WiFiClientSecureBearSSL.cpp.o
Compiling /data/test/.pioenvs/test/lib4d9/ESP8266WiFi/WiFiServer.cpp.o
Compiling /data/test/.pioenvs/test/lib4d9/ESP8266WiFi/WiFiServerSecureAxTLS.cpp.o
Compiling /data/test/.pioenvs/test/lib4d9/ESP8266WiFi/WiFiServerSecureBearSSL.cpp.o
Compiling /data/test/.pioenvs/test/lib4d9/ESP8266WiFi/WiFiUdp.cpp.o
Compiling /data/test/.pioenvs/test/lib0d3/ESP8266mDNS/ESP8266mDNS.cpp.o
```

After I Stop the process nothing more is added to the logs.

> High CPU usage during a compile is expected, the number of compile processes is automatically scaled with the number of available cores.

Yes and this is understandable, but something that used to take 1 minute on 2 core 2.1Ghz processor is now taking +30 minutes on 6 core 2.8Ghz

l3arcf, 2019-05-26:
I'm experiencing the same issue.  Slow compile.  I'm running an intel NUC and docker stats is reporting 394% CPU and 317MiB memory used of 7.77GiB.  cpp's that are slow are CertStoreBearSSL, WiFiServerSecureBearSLL.cpp so far.  After hours, I've still not fully compiled, and I have 6 devices I would like to upgrade.

Any advice, or anything I can do to hone in on the problem?  At this point I feel like I must be missing some key libraries in the container.

l3arcf, 2019-05-26:
Through process of elimination, I've downgraded from 'latest' to the version I had success with before.  Still no compile.  Given that I was successful on my RPi (before the SDCard gave up on me) and now I'm running  in a docker container I think the issue is related to me doing something wrong in docker, a software compatibility with my new hardware or not having prerequisites on the host OS (clear linux).

john-arvid, 2019-05-26:
Good to hear I'm not the only one.

Looks like that the thing we have in common is moving from one device to another. I assume you made a snapshot and imported that on the new device.
Is this correct?

I have Hassio running on a Debian virtual machine. This is were the problems started.
As a temporary solution I also set up a new VM with debian and installed ESPHome with PIP. 
REF: https://esphome.io/guides/getting_started_command_line.html

It is working as it should on the dedicated VM, but I don't get the same integration with Home Assistant. ("Auto discovery like" function)

The only thing I haven't tried is a clean install of Hassio on a VM and then installing ESPHome to check if that works. Will see if I get the time to do it someday.

l3arcf, 2019-05-26:
At the risk of getting off topic, but in the interests of discovering similarities between our setups:

I was running native (no docker or VM on the RPi). So everything was a fresh install and then moved yaml's over etc.
I've always (2+ years now) run hasbian, not hass.io, and I was using ESPHome from the commandline.
With my move to the NUC, I run clearlinux on bare-metal with docker, and am trying to run everything in docker containers.  portainer, unifi controller, home-assistant.  I'm loving docker. FWIW, I've tried different versions of ESPHome this way with similar results.  Older ones do not support esp8266_restore_from_flash AFAIK, and newer ones do, but are slow compiling and I've not successfully compiled a binary yet. 1.12.1 is what worked from the RPi and it doesn't in the docker container I have.
Current thinking is it's a docker config, or a host OS assumption that I have not met, or a system architecture issue.

I did not re-flash my ESPHome devices when I initially moved over.  I kept my old IP and home-assistant discovered fine.  I seem to recall something to do with mDNS being a possible issue from the container.  I have zero experience with hass.io so keep that in mind!

I may just rebuild the RPi on a new SDCard to get past this in the short-term.

l3arcf, 2019-06-11:
It's been a couple of weeks but work has been pretty heavy.  I got back to this last weekend and have decided that it's likely the kernel architecture that's directly (or indirectly) causing the slow compile.  I actually did get some successful compiles but very slow.  In defence of ESPHome, they are clear that not all architectures are supported so maybe I'm lucky to have something working at all.  Having said that, because it 'works' it's tough to track down the issue with slowness.

I re-established my old RPi and use that for compiling my binaries and that seems to be working well enough for now.  I'd like to have this all on my x86 box in docker but understand there are constraints I need to live with, and frankly, I think it's still worth it!

In my research, one of the biggest issues facing the future of ESPHome is platformio and the requirement on Python 2.7.  This version is sunsetting soon and so a number of newer linux variants don't even support the old commands or versions of of the box, or map them to newer ones. Clearlinux, for one was a huge struggle so I gave up trying to build my own container and just moved ESPHome back to the RPi.

I'll test each version and comment if I see any improvement in my experience in the future.

OttoWinter, 2019-07-11:
Closing as won't fix - it's probably a daemon process somewhere compiling files even after the process has been terminated. Workaround is restart the addon.

closed 2019-07-11 by OttoWinter