esphome/issues #205: Add native API to ESP32 Cam examples in documentation

mverleun, 2019-04-05:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
N/A

MacBook with ESPHome 1.12.1 in Docker container
**ESP (ESP32/ESP8266, Board/Sonoff):**

**Affected component:**
ESP32 Camera

**Description of problem:**
The documentation mentions that the native API will directly integrate with Home Assistant.

It does not mention very explicit that the native API is required and that without it the compilation will fail.
None of the shown examples have the native API included, please add these for clarity. 

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Traceback (if applicable):**

```
# The api: entry is required when compiling the esp32_camera: module
api:
  

# Example configuration entry 
# Check https://esphome.io/components/esp32_camera.html
# Don't forget to create the api: entry
esp32_camera:
  external_clock:
    pin: GPIO0
    frequency: 20MHz
  i2c_pins:
    sda: GPIO26
    scl: GPIO27
  data_pins: [GPIO5, GPIO18, GPIO19, GPIO21, GPIO36, GPIO39, GPIO34, GPIO35]
  vsync_pin: GPIO25
  href_pin: GPIO23
  pixel_clock_pin: GPIO22
```

**Additional information and things you've tried:**

OttoWinter, 2019-04-25:
The validation will now print an error when api is not active 👍

closed 2019-04-25 by OttoWinter

mverleun, 2019-04-25:
Great! Thanks for your hard work.