esphome/issues #206: NodeMCU tries to connect to set SSID but fails and restarts Wifi adapter in loop

mr-sneezy, 2019-04-06:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

HA 0.91.0 and hassos 2.11 on RPi3

**ESP (ESP32/ESP8266, Board/Sonoff):**

NodeMCU V3
**Affected component:**

WiFi connectivity

**Description of problem:**
Setting up a NodeMCU as a default esphome device for testing and finding it can't or won't connect to my home WiFi network. 
[19:52:27][I][wifi:167]: WiFi Connecting to 'Not_my_SSID'...
[19:52:33][W][wifi:399]: Error while connecting to network.
[19:52:33][W][wifi:429]: Restarting WiFi adapter...
[19:52:33][I][wifi:167]: WiFi Connecting to 'Not_my_SSID'...
[19:52:39][W][wifi:399]: Error while connecting to network.
[19:52:39][W][wifi:429]: Restarting WiFi adapter...
[19:52:45][D][wifi:271]: Starting scan...

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: test1
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: "Not_my_SSID"
  password: "Not_my_password"

  manual_ip:
# Set this to the IP of the ESP
    static_ip: 192.168.0.22
# Set this to the IP address of the router. Often ends with .1
    gateway: 191.168.0.1
# The subnet of the network. 255.255.255.0 works for most home networks.
    subnet: 255.255.255.0

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**
Two things:
1/
 Tried a static IP and matching it in the router. No change.

2/
 As a test to try see if the new module was faulty, I took this NodeMCU module and loaded the current release Tasmota sonoff-basic binary via the same esphomeflasher (handy to know that esphhomeflasher will work for other binaries as I run a bunch of Tasmota FW module on my network).
The binary flashed fine same as the esphome one, but in the case of the Tasmota binary it DOES connect to my WiFi and hang on reliably. 
I got into the AP it created as normal and changed the WiFi parameters and restarted it, then got back into it right way via it's new IP on the Wifi network. 
So there is nothing wrong with this NodeMCU module (and the esphome flasher)...

OttoWinter, 2019-04-07:
1. Did you also try with fast_connect as suggested in the FAQ?

2. Please turn on verbose logs, they include more information about why it failed to connect.

mr-sneezy, 2019-04-08:
Didn't see that in the FAQ (or didn't make a logical connection to my issue). 
**fast_connect** allows the node to connect now thanks. Maybe that is default on the other firmwares.

Verbose log in case it's of any use, of the successful connection. 
INFO Reading configuration...
INFO Starting log output from 192.168.0.22 using esphome API
INFO Connecting to 192.168.0.22:6053 (192.168.0.22)
INFO Successfully connected to 192.168.0.22
[18:56:31][I][application:097]: esphome-core version 1.12.2 compiled on Apr  8 2019, 18:46:19
[18:56:31][C][wifi:371]: WiFi:
[18:56:31][C][wifi:253]:   SSID: [redacted]
[18:56:31][C][wifi:254]:   IP Address: 192.168.0.22
[18:56:31][C][wifi:256]:   BSSID: [redacted]
[18:56:31][C][wifi:257]:   Hostname: 'test1'
[18:56:31][C][wifi:261]:   Signal strength: -69 dB ▂▄▆█
[18:56:31][C][wifi:262]:   Channel: 5
[18:56:31][C][wifi:263]:   Subnet: 255.255.255.0
[18:56:31][C][wifi:264]:   Gateway: 191.168.0.1
[18:56:31][C][wifi:265]:   DNS1: 1.1.1.1
[18:56:31][C][wifi:266]:   DNS2: 1.0.0.1
[18:56:31][C][logger:142]: Logger:
[18:56:31][C][logger:143]:   Level: VERBOSE
[18:56:31][C][logger:144]:   Log Baud Rate: 115200
[18:56:31][C][logger:145]:   Hardware UART: UART0
[18:56:31][C][api:101]: API Server:
[18:56:31][C][api:102]:   Address: 192.168.0.22:6053
[18:56:31](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[18:56:31][C][ota:128]:   Address: 192.168.0.22:8266

OttoWinter, 2019-04-09:
> Maybe that is default on the other firmwares.

Kind of, yes - fast_connect is the default way in the base sdk and therefore many firmwares use it. However, fast_connect has the big disadvantage that if you have multiple networks with the same SSID, it will connect to the one on the lowest channel (and not the one with the highest signal strength).

Anyway, I will close this issue. Further discussion should be held in the other wifi-related issues here (specifically the "No AP found" ones)

closed 2019-04-09 by OttoWinter

OttoWinter, 2019-06-18:
Moving the discussion to #455