esphome/issues #208: Error "contains duplicate key "switch", when using substitution yaml file

rradar, 2019-04-09:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip 
**ESP (ESP32/ESP8266, Board/Sonoff):**

esp8266
**Affected component:**

yaml parser?

**Description of problem:**

It's not possible for me to add a `switch:` component to my younameit.yaml while I'm injecting
`<<: !include common.yaml)` which also includes a `switch:` component already.

This is only true for the switch component and not for sensor and others. Here it works even I'm using the component in my main (younameit.yaml) and in the common.yaml

**Problem-relevant YAML-configuration entries:**

yonameit.yaml part
```yaml
switch:
  - platform: output
    name: "${upper_devicename} LED"
    output: led
    id: switch_led
```

common.yaml part
```yaml
switch:
  - platform: restart
    name: "${upper_devicename} Restart"
    id: restart
```

**Traceback (if applicable):**

```
ERROR Error while reading config: YAML file younameit.yaml contains duplicate key "switch". Check lines 11 and 111.
```

**Additional information and things you've tried:**

If I comment out (# switch: ...) one the switch components in younameit.yaml compiling works. When I comment out (# switch:  ...) in common.yaml it does not compile and starts complaining about a duplicate sensor component :tired_face:

rradar, 2019-04-15:
Okay, I finally found out what's going on.

If you have a `sensor:` part in your **nodemcu.yaml** which links the substitution yaml **common.yaml** which also includes a `sensor:` part then:
- In the best case you can compile without errors but all your sensor from the **common.yaml** are not included and you will not get any warning about this!
- In the worst case you can't compile at all and get some error like:
`ERROR Error while reading config: YAML file younameit.yaml contains duplicate key "sensor"
`

I'm going to close this one here (because now I know what's going on) and will open a feature request which makes it possible to merge different sensors/switches etc. from different yaml files to get the best case but with all sensors :tada:

closed 2019-04-15 by rradar

OttoWinter, 2019-04-15:
@rradar Yes the `<<: ` imports all keys and thus leads to this error. However, we can't work around it in esphome, as otherwise that would be invalid YAML spec (YAML explicitly states that keys must be unique, as opposed to JSON). Thus this cannot be solved by esphome, I believe there are probably other ways in working around this limitation without changing the yaml parser.