esphome/issues #209: Interval Component bricks ESP(?)

meichthys, 2019-04-10:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io running on HassOS which is running in a HyperV vm

**ESP (ESP32/ESP8266, Board/Sonoff):**

platform: ESP8266
board: esp12e

**Affected component:**

[https://esphome.io/guides/automations.html?highlight=interval#interval](url)

**Description of problem:**
I could be using this component incorrectly, but each time I use the "interval" component, my esp seems to become bricked. After adding the interval component below, the esp turns on but does not report to home assistant and does not show as 'online' in the ESPHome web interface. The esp also cannot be re-flashed or cleared. When trying to re-flash or clear the esp using the esphomeflasher program i receive the following errors:

```
Using '/dev/cu.usbserial-A900VHJQ' as serial port.
Connecting....
Detecting chip type... ESP8266
Connecting...

Chip Info:
 - Chip Family: ESP8266
 - Chip Model: ESP8266EX
 - Chip ID: 00C6C7F4
 - MAC Address: CC:50:E3:C6:C7:F4
Uploading stub...
Running stub...
Stub running...
 - Flash Size: 4MB
 - Flash Mode: dout
 - Flash Frequency: 40MHz
Erasing flash (this may take a while)...
Unexpected error: Error while erasing flash: Timed out waiting for packet content
```
Most of the time i don't even get that far and will get the following:
```
Unexpected error: Error while erasing flash: Timed out waiting for packet content
```
It's also entirely possible that i'm not using the interval component correctly.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphome:
  name: ws1
  platform: ESP8266
  board: esp12e

wifi:
  ssid: "myawesomessid"
  password: "myawesomepassword"
  fast_connect: true
  power_save_mode: high
  manual_ip:
    static_ip: 10.0.6.123
    gateway: 10.0.6.1
    subnet: 255.255.255.0
    dns1: 10.0.6.1

interval:
  interval: 1min
  then: switch.toggle: ws1shutdown

switch:
  platform: shutdown
  name: ws1 Shutdown
  id: "ws1shutdown"

binary_sensor:
  platform: status
  name: ws1 Sensor
  device_class: window

sensor:
  platform: adc
  pin: VCC
  name: ws1 Voltage

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
```

**Traceback (if applicable):**

```
I can't seem to retrieve any logs via USB :/
```

**Additional information and things you've tried:**

I've tried:

- not using any power_save_mode
- longer/shorter interval times
- multiple different ESPs (all became 'bricked')

brandond, 2019-04-10:
As far as I know is pretty much impossible to brick an ESP by writing bad firmware or running bad code. I suppose you could burn out the flash memory with excessive write cycles or something like that.

It sounds more like there's something wrong with either the serial adapter or the how you're powering it. How do you have it wired up?

meichthys, 2019-04-10:
@brandond  Thanks for the quick reply.

The serial adapter works fine for flashing/reflashing/running other similar ESP chips. I flashed 20 chips over the last few days without any isues. I'm using the same wiring as found here [https://cdn.instructables.com/FHW/O1G9/IYYSWSGM/FHWO1G9IYYSWSGM.MEDIUM.jpg](url)

The chip powers on normally, but the interval code seems to prevent it from flashing or connecting to wifi. Could it be a problem with where i added the interval yaml code? I'm not sure if it matters, but should the `interval` yaml code be below the `switch` that it references?

I've found a workaround using the `time` component, but that isn't as fitting for my purposes since it depends on the actual time and not an 'interval' since the esp has booted.

OttoWinter, 2019-04-15:
Nah, `interval` can't brick a device.

However, I'm not even sure how you managed to get it to compile with that config:
```yaml
interval:
  interval: 1min
  then: switch.toggle: ws1shutdown
```

is invalid YAML (switch.toggle should be on a separate line)

meichthys, 2019-04-15:
@OttoWinter Thanks for the input.

I probably shouldn't have used the term 'bricked' since it's not really bricked, it's just stuck in some sort of boot loop that doesn't allow me to re-flash the chip. 

I'll try the interval component again on a new chip by putting the switch.toggle on a new line.

For now i think we can close this issue.  I'll comment back if i can gather some more identifiable information that might help pinpoint the issue.

meichthys, 2019-07-10:
I havnen't tried the interval component again, but i'm closing this issue because I'm no longer actively facing the issue.

closed 2019-07-10 by meichthys