esphome/issues #213: In a recent NodeMCU ESP8266 an ADC range is 0-3.3V

epiniguin, 2019-04-10:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

NodeMCU ESP8266
**Affected component:**

https://esphome.io/components/sensor/adc.html

**Description of problem:**
In a recent NodeMCU ESP8266 an ADC range is 0-3.3V.
That's why an ADC sensor value is wrong - it should be multiplied by 3.3. 

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

HFsi, 2019-04-10:
Could you provide more info? Brand, version, revision, etc

brandond, 2019-04-10:
As far as I know, the esp8266 has always used 0.0 - 1.0 to represent 0v0 to 3v3. It's up to the user to scale that to whatever they're trying to measure.

kimonm, 2019-04-10:
The ADC range is 0-1V at the chip. The range is 0-3.3V at the board pin. There is a voltage divider  comprised of 220k and 100k resistors between the board pin and chip pin.  See schematic on https://github.com/nodemcu/nodemcu-devkit

HFsi, 2019-04-10:
This is very important, should be updated in the documentation, because Otto states in the NodeMCU board specs, that the measurement range is between 0-1 V

epiniguin, 2019-04-10:
> 
> 
> The ADC range is 0-1V at the chip. The range is 0-3.3V at the board pin. There is a voltage divider comprised of 220k and 100k resistors between the board pin and chip pin. See schematic on https://github.com/nodemcu/nodemcu-devkit

Exactly!
And now ESPHome ADC sensor value is 3.3 times less.

kimonm, 2019-04-10:
I submitted a PR for the documentation to clarify what voltage this component uses.

HFsi, 2019-04-11:
Great info and great community!

OttoWinter, 2019-04-15:
Documentation has been updated

closed 2019-04-15 by OttoWinter