esphome/issues #215: Show Logs in Dashboard gives traceback , for just 1 device.

Zixim, 2019-04-11:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io 0.91.2
**ESP (ESP32/ESP8266, Board/Sonoff):**

generic ESP32

**Affected component:**

not component related

**Description of problem:**
In the dash, I click on Show Logs for 1 specific node, the Show Logs terminal window opens, and only shows a Traceback.
The node works just fine, serial logging is fine, OTA works fine.
This is happening on only 1 of my nodes.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Traceback (if applicable):**

```
INFO Reading configuration...
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 11, in <module>
    load_entry_point('esphome', 'console_scripts', 'esphome')()
  File "/opt/esphome/esphome/__main__.py", line 484, in main
    return run_esphome(sys.argv)
  File "/opt/esphome/esphome/__main__.py", line 474, in run_esphome
    return POST_CONFIG_ACTIONS[args.command](args, config)
  File "/opt/esphome/esphome/__main__.py", line 303, in command_logs
    return show_logs(config, args, port)
  File "/opt/esphome/esphome/__main__.py", line 230, in show_logs
    raise ValueError
ValueError
```

**Additional information and things you've tried:**
Have restarted ESPHome addon & Home Assistant.

OttoWinter, 2019-04-15:
That would be because there is no logging method is available:
- MQTT and API are disabled, so no remote logging
- No serial adapter is attached - so no local logging.

The error message could be improved a bit though - the behavior itself is correct.

OttoWinter, 2019-07-11:
Fixed some versions ago

closed 2019-07-11 by OttoWinter