esphome/issues #217: VERY_VERBOSE logger level causes crash and boot loop

kll, 2019-04-11:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

PIP, compiled from source

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266EX

**Affected component:**

https://esphome.io/components/logger.html

**Description of problem:**
Setting the logging level to VERY_VERBOSE causes a crash and boot loop. 

**Problem-relevant YAML-configuration entries:**
```yaml
substitutions:
  device_name: test_light
  friendly_name: Test Light

esphome:
  name: ${device_name}
  platform: ESP8266
  board: esp01_1m
  esphome_core_version:
    local: /home/kevin/code/esphome-all/esphome-core/

  # Ensure the light turns on by default if the physical switch is actuated.
  on_boot:
    priority: 50
    then:
      - light.turn_on: light

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

web_server:
  port: 80

logger:
  level: VERY_VERBOSE

api:

ota:

output:
  - platform: esp8266_pwm
    id: output_component1
    pin: GPIO5
  - platform: esp8266_pwm
    id: output_component2
    pin: GPIO12

light:
  - platform: cwww
    name: ${friendly_name}
    id: light
    cold_white: output_component1
    warm_white: output_component2
    cold_white_color_temperature: 6500K
    warm_white_color_temperature: 2700K
    default_transition_length: 0.5s
```

**Traceback (if applicable):**

```
[... 175 lines cut ...]
[07:23:50]3fffff50:  00000000 3fff1a58 feefeffe feefeffe
[07:23:50]3fffff60:  feefeffe feefeffe feefeffe feefeffe
[07:23:50]3fffff70:  feefeffe feefeffe feefeffe feefeffe
[07:23:50]3fffff80:  feefeffe feefeffe feefeffe feefeffe
[07:23:50]3fffff90:  feefeffe 3fff0680 feefeffe feefeffe
[07:23:50]3fffffa0:  feefeffe feefeffe feefeffe 3fff05c8
[07:23:50]3fffffb0:  3fffdad0 00000000 3fff05c1 402143a0
WARNING Decoded 0x402143a0: loop_wrapper() at core_esp8266_main.cpp
[07:23:50]3fffffc0:  feefeffe feefeffe 3ffe8654 40100a2d
WARNING Decoded 0x40100a2d: cont_wrapper
[07:23:50]<<<stack<<<
[07:23:50]
[07:23:50] ets Jan  8 2013,rst cause:1, boot mode:(3,7)
[07:23:50]
[07:23:50]load 0x4010f000, len 1384, room 16
[07:23:50]tail 8
[07:23:50]chksum 0x2d
[07:23:50]csum 0x2d
[07:23:50]vbb28d4a3
[07:23:50]~ld

```

**Additional information and things you've tried:**
I flashed a Feit RGBWW light bulb OTA which essentially bricked it. I wanted to figure out what happened though so I ripped off the diffuser, removed the LED driver board which was blocking access to the tuya module, and traced the PCB to figure out which test points gave me access to the right pins and managed to solder on some wires to allow me to access the UART and GPIO0 directly and was able to revive it. After several tests I narrowed it down to changing the logging level from VERBOSE to VERY_VERBOSE. I tested with the same YAML only changing that one thing and also tested on both the master and dev branches with the same results. The logs pasted above is from the test on the dev branch.

Zixim, 2019-04-11:
on esp32, i don't get the crash / bootloop.
I did notice these warnings in the compilation sequence : 
```
Compiling /data/esp32_dev/.pioenvs/esp32_dev/FrameworkArduino/esp32-hal-i2c.c.o
/root/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-i2c.c: In function 'i2cDumpFifo':
/root/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-i2c.c:415:35: warning: ' ' flag used with '%X' gnu_printf format [-Wformat=]
} else k += sprintf(&buf[k],"% 4X ",fifoBuffer[i]);
^
/root/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-i2c.c:415:35: warning: ' ' flag used with '%X' gnu_printf format [-Wformat=]
Compiling /data/esp32_dev/.pioenvs/esp32_dev/FrameworkArduino/esp32-hal-ledc.c.o
```

They are there only for `VERY_VERBOSE` , not for `VERBOSE`.

glmnet, 2019-04-12:
Me too. Did see the problem on ESP8266 I was debugging some i2c stuff and eventually it stopped working. Sorry I cannot give more details.

OttoWinter, 2019-04-15:
Ah yes I see what it is now, some places in esphome use `%zu` as format which works fine in arduino-core@2.5.0, but crashes before. (and on esp32 this is no problem)

The `%zu`s just need to be replaced with `%u` (and all those lines need to get a `// NOLINT` flag, otherwise clang-tidy will correctly complain that the format is wrong)

OttoWinter, 2019-04-25:
Fixed in dev.

closed 2019-04-25 by OttoWinter