esphome/issues #218: OTA Update Failed: ERROR[8]: Flash config wrong real: 4194304 IDE: 16777216

lblokland, 2019-04-11:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Using Hassio and ESPHome

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 ,D1 Mini by Robotdyn

**Affected component:**

OTA is failing

**Description of problem:**
Hi,

I'm using a D1 Mini (RobotDyn) with 32Mb and when trying to flash OTA the debug on the webinterface shows
`Update Failed: ERROR[8]: Flash config wrong real: 4194304 IDE: 16777216`
Flashing manually using 'esptool -cp COM3 -bm dout -cf image.bin -v' works just fine.

The error message on ESPhome console is:
'
INFO Uploading /data/d1_1/.pioenvs/d1_1/firmware.bin (386752 bytes)
ERROR Error binary size: Error: ESP has been flashed with wrong flash size. Please choose the correct 'board' option (esp01_1m always works) and then flash over USB.
WARNING OTA v2 method failed. Trying with legacy OTA...
INFO Starting on 0.0.0.0:46964
INFO Upload size: 386752
INFO Sending invitation to 192.168.178.16 
..........
ERROR No response from the ESP
'

Does anyone know how to solve this?

Cheers,
Leon

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphome:
  name: d1_1
  platform: ESP8266
  board: d1_mini
#  board: Generic 4M
web_server:
  port: 80
## Example configuration entry
output:
  - platform: gpio
    pin: D1
    id: gpio_d1
    
## Example configuration entry
light:
  - platform: binary
    name: "LED Lamp"
    output: gpio_d1
    
## Example configuration entry
#mqtt:
#  broker: 192.168.178.15
#  username: jj
#  password: jj

wifi:
  ssid: "jj"
  password: "jj"
  # Optional manual IP
  manual_ip:
    static_ip: 192.168.178.16
    gateway: 192.168.178.1
    subnet: 255.255.255.0
    dns1: 192.168.178.1
# Enable logging
logger:

# Enable Home Assistant API
api:
  password: 'jj'

ota:
  password: 'jj'
```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

brandond, 2019-04-11:
It looks like 
>Update Failed: ERROR[8]: Flash config wrong real: 4194304 IDE: 16777216

It tells you in the error message what to do:

> ERROR Error binary size: Error: ESP has been flashed with wrong flash size. Please choose the correct 'board' option (esp01_1m always works) and then flash over USB.

Did you perhaps initially flash via USB using another board definition that is configured for only 1mb?

lblokland, 2019-04-12:
Hi Brandon,

Not sure what I used to flash it initially, thats a few weeks ago.
Today I build the image with ESPhome and downloaded the binary. Just to be sure I tried the image with the d1_mini_pro board as it’s a 32Mb board.
Flashed it from the command line with esptool and then it was online.

So, tried a new OTA update, but no go:

INFO Successfully compiled program.

INFO Connecting to 192.168.178.16

INFO Uploading /data/d1_1/.pioenvs/d1_1/firmware.bin (415376 bytes)

ERROR Error binary size: Error: ESP has been flashed with wrong flash size. Please choose the correct 'board' option (esp01_1m always works) and then flash over USB.

WARNING OTA v2 method failed. Trying with legacy OTA...

INFO Starting on 0.0.0.0:34482

INFO Upload size: 415376

INFO Sending invitation to 192.168.178.16

..........

ERROR No response from the ESP

Leon

Verzonden vanuit Mail<https://go.microsoft.com/fwlink/?LinkId=550986> voor Windows 10

OttoWinter, 2019-04-15:
> Update Failed: ERROR[8]: Flash config wrong real: 4194304 IDE: 16777216

This log string is from the arduino `Update` library. And what it says is: your board setting is for `16777216` bytes (4MB), however the actual hw only has 4194304 bytes of flash (1MB).

You probably got a cheap module that doesn't have the flash size it advertised - the log string is very likely correct (don't think the arduino library would have an error there).

Please try with `esp01_1m` board option

closed 2019-04-15 by OttoWinter

lblokland, 2019-04-15:
Hi Otto,

that's a bit too easy...there is a Winbond 25Q32JVSIQ on the board, and that is a 4MB chip...
It was cheap, but all the ESP2866 are right? 

Is there an easy way to read hardware using USB?

OttoWinter, 2019-04-16:
> that's a bit too easy...there is a Winbond 25Q32JVSIQ on the board, and that is a 4MB chip...
> It was cheap, but all the ESP2866 are right?

Well the log output is showing otherwise - and as I said I don't think that would be incorrect.

Probably you can check the size with esptool.py somehow, but you'll have to look that up with google

lblokland, 2019-04-16:
Hi Otto,

But stilll..
When using ESPtool.exe I can write the 4MB ESPeasy file just fine:

C:\temp>esptool.exe -cp COM3 -cb 115200 -bm dout -cf ESPEasy_R120_4096.bin -v
esptool v0.4.13 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
setting flash mode from qio to dout
opening bootloader
resetting board
trying to connect
trying to connect
Uploading 411264 bytes from ESPEasy_R120_4096.bin to flash at 0x00000000
................................................................................ [ 19% ]
................................................................................ [ 39% ]
................................................................................ [ 59% ]
................................................................................ [ 79% ]
................................................................................ [ 99% ]
..                                                                               [ 100% ]
starting app without reboot
closing bootloader

After that I can also flash with the image created with board ‘d1_mini_pro’ :

C:\temp>esptool.exe -cp COM3 -cb 115200 -bm dout -cf d1_1-d1_mini_pro.bin -v
esptool v0.4.13 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
setting flash mode from qio to dout
opening bootloader
resetting board
trying to connect
trying to connect
trying to connect
resetting board
trying to connect
Uploading 387408 bytes from d1_1-d1_mini_pro.bin to flash at 0x00000000
................................................................................ [ 21% ]
................................................................................ [ 42% ]
................................................................................ [ 63% ]
................................................................................ [ 84% ]
...........................................................                      [ 100% ]
starting app without reboot
closing bootloader

This one boots just fine, just OTA isn’t working.
Also not when using the board ‘esp01_1m’
So, when I want to change something, I need to do it using a serial upload. Works fine, but OTA is better.

Verzonden vanuit Mail<https://go.microsoft.com/fwlink/?LinkId=550986> voor Windows 10

OttoWinter, 2019-04-21:
> When using ESPtool.exe I can write the 4MB ESPeasy file just fine:

That tool doesn't check if the chip actually supports firmware sizes. You could also try uploading firmware images for 16MB systems no problem unless the *file* size is bigger than the chip size.

> Also not when using the board ‘esp01_1m’

So to clarify, the error message means that you need to flash an esp01_1m image over USB, and then you can upload esp01_1m images OTA too. You cannot upload esp01_1m images OTA with a system that is set to 4MB of flash.

lblokland, 2019-04-23:
Hi Otto,

Finally found it.
Looks like I was using the wrong board all the time. When trying your solution I was able to perform the OTA, but pins wouldn’t work.
When I took a new look at the back of the board I noticed it said D1 MINI without the PRO….
So, went to ‘http://docs.platformio.org/en/latest/boards/espressif8266/d1_mini.html#board-espressif8266-d1-mini’ and found the correct settings.

Sorry I have bothered you, but maybe we can help others when they read this ‘resolution

Thanks again,

Leon