esphome/issues #219: SoftwareSerial on Custom Text Sensor

webcharlyx, 2019-04-11:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 (d1 mini)

**Affected component:**

https://esphome.io/components/text_sensor/custom.html

**Description of problem:**
Tryed to include and use SoftwareSerial.h for custom text sensor

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  includes:
    - lg_e350_ir.h
  libraries:
    - "SoftwareSerial"
```

[lg_e350_ir.h.txt](https://github.com/esphome/issues/files/3070834/lg_e350_ir.h.txt)

**Traceback (if applicable):**

```
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Library Storage: xyz\esphomeyaml\lg_e350_ir\.piolibdeps
LibraryManager: Installing id=346 @ 0.8.2
AsyncMqttClient @ 0.8.2 is already installed
Installing dependencies
LibraryManager: Installing id=305 @ ^1.1.0
ESPAsyncTCP @ 1.2.0 is already installed
LibraryManager: Installing id=1826 @ ^1.0.0
AsyncTCP @ 1.0.3 is already installed
Looking for ESP8266WiFi library in registry
Warning! Library `{'requirements': None, 'name': 'ESP8266WiFi'}` has not been found in PlatformIO Registry.
You can ignore this message, if `{'requirements': None, 'name': 'ESP8266WiFi'}` is a built-in library (included in framework, SDK). E.g., SPI, Wire, etc.
Looking for ESP8266mDNS library in registry
Warning! Library `{'requirements': None, 'name': 'ESP8266mDNS'}` has not been found in PlatformIO Registry.
You can ignore this message, if `{'requirements': None, 'name': 'ESP8266mDNS'}` is a built-in library (included in framework, SDK). E.g., SPI, Wire, etc.
LibraryManager: Installing id=305 @ 1.2.0
ESPAsyncTCP @ 1.2.0 is already installed
Looking for Hash library in registry
Warning! Library `{'requirements': None, 'name': 'Hash'}` has not been found in PlatformIO Registry.
You can ignore this message, if `{'requirements': None, 'name': 'Hash'}` is a built-in library (included in framework, SDK). E.g., SPI, Wire, etc.
LibraryManager: Installing id=2728
SoftwareSerial @ 25b087be29 is already installed
LibraryManager: Installing v1.12.2
esphome-core @ 1.12.2 is already installed
Installing dependencies
LibraryManager: Installing id=3837 @ 5.13.3
ArduinoJson-esphomelib @ 5.13.3 is already installed
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp01_1m.html
PLATFORM: Espressif 8266 > Espressif Generic ESP8266 ESP-01 1M
HARDWARE: ESP8266 80MHz 80KB RAM (1MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Ignored library xyz\esphomeyaml\lg_e350_ir\.piolibdeps\AsyncTCP_ID1826
Framework incompatible library xyz\esphomeyaml\lg_e350_ir\.piolibdeps\SoftwareSerial_ID2728
Ignored library abc\.platformio\lib\AsyncTCP_ID1826
Ignored library abc\.platformio\lib\ESP Async WebServer_ID306
Ignored library abc\.platformio\lib\ESP Async WebServer_ID306@1.1.1
Ignored library abc\.platformio\lib\FastLED
Ignored library abc\.platformio\lib\FastLED_ID126
Ignored library abc\.platformio\lib\FastLED_ID126@3.2.0
Framework incompatible library abc\.platformio\lib\SoftwareSerial_ID2728
Ignored library abc\.platformio\packages\framework-arduinoespressif8266@2.20402.4\libraries\ArduinoOTA
Ignored library abc\.platformio\packages\framework-arduinoespressif8266@2.20402.4\libraries\Wire
Collected 39 compatible librariesMore details about "Library Compatibility Mode": https://docs.platformio.org/page/librarymanager/ldf.html#ldf-compat-mode
...
In file included from src\main.cpp:5:0:
src\..\..\lg_e350_ir.h:2:28: fatal error: SoftwareSerial.h: No such file or directory

************************************************************************
* Looking for SoftwareSerial.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:SoftwareSerial.h"
* Web  > https://platformio.org/lib/search?query=header:SoftwareSerial.h
*
************************************************************************

#include "SoftwareSerial.h"
^
compilation terminated.
*** [.pioenvs\lg_e350_ir\src\main.cpp.o] Error 1
```

**Additional information and things you've tried:**
OK...
`Framework incompatible library abc\.platformio\lib\SoftwareSerial_ID2728`
tells me, that installed lib is found but ESPhome does not like it...
within [uart_component.cpp](https://esphome.io/api/uart__component_8cpp_source) and 
[your answer here](https://github.com/esphome/feature-requests/issues/49#issuecomment-474304513) i found that SoftwareSerial is already used within ESPhome.
but I can't find any information how to use it in my own code for the custom sensor.
Did I miss any docu anywhere?

brandond, 2019-04-11:
https://esphome.io/custom/uart.html

webcharlyx, 2019-04-11:
hey brandond, THX!

I already had a look into custom uart before I wrote the issue, but it was not really clear to me, how I could implement this special style of UART communication there.
What I want to realize is: first to send some bits to the device for init, and then parse it's feedback. The tricky thing about it is it's 7E1. But I'm no expert on uart right now and any help is welcome.
What I first tried, was to get data on an custom text sensor with this
[lg_e350_ir.h.txt](https://github.com/esphome/issues/files/3070958/lg_e350_ir.h.txt)
mechanism.

Can I just push all stuff from my loop part into the loop part of a new my_custom_uart.h?

webcharlyx, 2019-04-12:
OK. Did so -with a few adjustments- and it compiled in the end.
Hopefully I will have some time next days to setup HW and give it a try.

brandond, 2019-04-12:
Here's a more complicated example of reading and writing bytes or byte arrays, checking to see if any data is waiting with available(), etc.

https://github.com/brandond/esphome-tuya_pir/blob/master/sb1_uart.h#L93

webcharlyx, 2019-04-12:
that's great! a lot to learn from it :-) Many thanks to you

OttoWinter, 2019-04-15:
Yeah the uart component is preferred because it's esphome-native.

If you do want to use stock softwareserial you need to correct the platformio lib. You need to use [espsoftwareserial](https://platformio.org/lib/show/168/EspSoftwareSerial) - just "softwareserial" points at an incompatible lib.

closed 2019-04-15 by OttoWinter