esphome/issues #220: Adding Servo component (and ESP8266_pwm) to code stops wifi connecting.

mr-sneezy, 2019-04-12:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

HA on hassOS on RPi3

**ESP (ESP32/ESP8266, Board/Sonoff):**

NodeMCU v3
**Affected component:**

Servo component

**Description of problem:**
I have a project that has a PIR sensor and LED working fine and reporting motion to HA.

When I add a Servo component to that code, and then upload, the ESPhome node drops offline and won't reconnect ( but the actual NodeMCU hardware responds to the PIR iva LED on/off & servo PWM changes on the pwm pin when viewed on oscilloscope).

Using ESPhomeFlasher to upload same code again gets same result (OTA not possible after issue occurs), logs show the ESP8266 not getting BSSID and Channel like a normal connection. Again the NodeMCU works via it's pin responses and pwm on CRO but node wifi won't connect to the router.

Oddly, reverting to the previous good YAML code (ripping out the Servo and pwm stuff) does NOT restore wifi connection right way after upload via ESPhomeFlasher and rebooting!
What does restore wifi is reflashing with Sonoff Tasmota firmware and checking wifi is working with that, then uploading the previous good ESPhome code after that again and rebooting. 

So to my untrained eye it appears that adding the Servo component is screwing with the wifi code somehow (but I'll be surprised if that's actually right).

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
**wifi killing code**
esphome:
  name: remote_actuator_node1
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: "not_my_ssid"
  password: "not_my_password"

  manual_ip:
# Set this to the IP of the ESP
    static_ip: 192.168.0.22
# Set this to the IP address of the router. Often ends with .1
    gateway: 191.168.0.1
# The subnet of the network. 255.255.255.0 works for most home networks.
    subnet: 255.255.255.0

  fast_connect: true
# Enable logging

logger:
  level: VERBOSE
# Enable Home Assistant API
api:

ota:

servo:
  - id: my_servo
    output: pwm_output

output:
  - platform: gpio
    pin: D1
    inverted: False
    id: gpio_d1_led
    
  - platform: esp8266_pwm
    id: pwm_output
    pin: D6
    frequency: 50 Hz

binary_sensor:
  - platform: gpio
    pin: D5
    name: "Multisensor Motion"
    device_class: motion
    
    on_press:
      then:
        - output.turn_on: gpio_d1_led
        - servo.write:
            id: my_servo
            level: 100.0%          
    on_release:
      then:
        - output.turn_off: gpio_d1_led
        - servo.write:
            id: my_servo
            level: -100.0%

**previous code that wifi works fine with**
esphome:
  name: remote_actuator_node1
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: "not_my_ssid"
  password: "not_my_password"

  manual_ip:
# Set this to the IP of the ESP
    static_ip: 192.168.0.22
# Set this to the IP address of the router. Often ends with .1
    gateway: 191.168.0.1
# The subnet of the network. 255.255.255.0 works for most home networks.
    subnet: 255.255.255.0
  fast_connect: true
  
# Enable logging
logger:
#  level: VERBOSE

# Enable Home Assistant API
api:

ota:

output:
  - platform: gpio
    pin: D1
    inverted: False
    id: gpio_d1_led

binary_sensor:
  - platform: gpio
    pin: D5
    name: "Multisensor Motion"
    device_class: motion
    
    on_press:
      then:
        - output.turn_on: gpio_d1_led
        
    on_release:
      then:
        - output.turn_off: gpio_d1_led

```

**Traceback (if applicable):**

```
Compilation and verification is passing with no errors.
```

**Additional information and things you've tried:**
I've repeated the process three times to get the Servo component working and then back tracking to the previous working code (after the Tasmota step to check wifi module still working on that FW).
Not sure what to ask to fix my issue. I have recorded the logs from the failing code and the passing code if needed...

OttoWinter, 2019-04-15:
> So to my untrained eye it appears that adding the Servo component is screwing with the wifi code somehow (but I'll be surprised if that's actually right).

Very unlikely. If anything the ESP would crash. But otherwise the servo component is just writing values to the PWM output.

Sounds more like a power supply problem to me - are you using the 3.3V or 5V from the ESP8266 directly for powering the servo?

mr-sneezy, 2019-04-16:
I'm using a 1Amp USB wall-wart, and a 1000uF capacitor on Vin/Gnd on the NodeMCU.
Just tried a 2Amp USB wall-wart with no change, removed the capacitor also, no change. 
So I'm now trying a second NodeMCU module and rerunning the two codes (with and without the servo usage).

mr-sneezy, 2019-04-16:
The

closed 2019-04-16 by mr-sneezy

mr-sneezy, 2019-04-16:
The repeat of test on another NodeMCU also failed once I added the servo code (with some other outputs).
This is the basic connection test that passes the WiFi logon. 
'''
esphome:
  name: servo_test2
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

# Enable logging
logger:
  level: VERBOSE

# Enable Home Assistant API
api:

ota:
''''
The successful log from ESPhomeflasher
-------------------------------------------------
Using 'COM3' as serial port.
Connecting....
Detecting chip type... ESP8266
Connecting....

Chip Info:
 - Chip Family: ESP8266
 - Chip Model: ESP8266EX
 - Chip ID: 00264080
 - MAC Address: 2C:3A:E8:26:40:80
Uploading stub...
Running stub...
Stub running...
 - Flash Size: 4MB
 - Flash Mode: dout
 - Flash Frequency: 40MHz
Erasing flash (this may take a while)...
Chip erase completed successfully in 7.2s
Compressed 336272 bytes to 232027...
Wrote 336272 bytes (232027 compressed) at 0x00000000 in 20.5 seconds (effective 131.1 kbit/s)...
Hash of data verified.

Leaving...
Hard Resetting...
Done! Flashing is complete!

Showing logs:
[17:56:16];l
[17:56:16]SDK:2.2.1(cfd48f3)/Core:2.4.2/lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1-13-g163bb82)/BearSSL:6d1cefc
[17:56:16][I][logger:119]: Log initialized
[17:56:16][C][ota:484]: There have been 0 suspected unsuccessful boot attempts.
[17:56:16][I][application:055]: Running through setup()...
[17:56:16][V][application:056]: Sorting components by setup priority...
[17:56:16][C][wifi:027]: Setting up WiFi...
[17:56:16][V][wifi_esp8266:034]: Enabling STA.
[17:56:16][V][wifi_esp8266:044]: Disabling AP.
[17:56:16]bcn 0
[17:56:16]del if1
[17:56:16]usl
[17:56:16]mode : sta(2c:3a:e8:26:40:80)
[17:56:16]add if0
[17:56:16][V][wifi_esp8266:392]: Event: Changed Mode old=AP new=STA
[17:56:16]wifi evt: 8
[17:56:16]sleep disable
[17:56:16][D][wifi:271]: Starting scan...
[17:56:22]scandone
[17:56:22][D][wifi:286]: Found networks:
[17:56:22][I][wifi:321]: - 'not_my_ssid' (BC:30:D9:BA:4B:E3) ?_?¦
[17:56:22][D][wifi:322]:     Channel: 5
[17:56:22][D][wifi:323]:     RSSI: -61 dB
[17:56:22][D][wifi:325]: - '' (7A:30:D9:BA:4B:E5) ?_?¦
[17:56:22][D][wifi:325]: - 'TelstraDC773D' (E0:B9:E5:DC:77:3D) ?_?¦
[17:56:22][D][wifi:325]: - 'Telstra Air' (E2:B9:E5:DC:77:3F) ?_?¦
[17:56:22][D][wifi:325]: - 'Fon WiFi' (E2:B9:E5:DC:77:30) ?_?¦
[17:56:22][I][wifi:167]: WiFi Connecting to 'not_my_ssid'...
[17:56:22][V][wifi:169]: Connection Params:
[17:56:22][V][wifi:170]:   SSID: 'not_my_ssid'
[17:56:22][V][wifi:173]:   BSSID: BC:30:D9:BA:4B:E3
[17:56:22][V][wifi:177]:   Password: 'not_my_pw'
[17:56:22][V][wifi:179]:   Channel: 5
[17:56:22][V][wifi:189]:   Using DHCP IP
[17:56:22][V][wifi:191]:   Hidden: NO
[17:56:23][V][wifi_esp8266:360]: Event: Changed AuthMode old=OPEN new=WPA2 PSK
[17:56:23]wifi evt: 2
[17:56:23]scandone
[17:56:24]state: 0 -> 2 (b0)
[17:56:24]state: 2 -> 3 (0)
[17:56:24]state: 3 -> 5 (10)
[17:56:24]add 0
[17:56:24]aid 10
[17:56:24]cnt 
[17:56:27]
[17:56:27]connected with not_my_ssid, channel 5
[17:56:27]dhcp client start...
[17:56:27][V][wifi_esp8266:345]: Event: Connected ssid='not_my_ssid' bssid=BC:30:D9:BA:4B:E3 channel=5
[17:56:27]wifi evt: 0
[17:56:28]ip:192.168.0.51,mask:255.255.255.0,gw:192.168.0.1
[17:56:28][V][wifi_esp8266:366]: Event: Got IP static_ip=192.168.0.51 gateway=192.168.0.1 netmask=255.255.255.0
[17:56:28]wifi evt: 3
[17:56:28][I][wifi:379]: WiFi connected!
[17:56:28][C][wifi:253]:   SSID: 'not_my_ssid'
[17:56:28][C][wifi:254]:   IP Address: 192.168.0.51
[17:56:28][C][wifi:256]:   BSSID: BC:30:D9:BA:4B:E3
[17:56:28][C][wifi:257]:   Hostname: 'servo_test2'
[17:56:28][C][wifi:261]:   Signal strength: -62 dB ?_?¦
[17:56:28][C][wifi:262]:   Channel: 5
[17:56:28][C][wifi:263]:   Subnet: 255.255.255.0
[17:56:28][C][wifi:264]:   Gateway: 192.168.0.1
[17:56:28][C][wifi:265]:   DNS1: 192.168.0.1
[17:56:28][C][wifi:266]:   DNS2: 0.0.0.0
[17:56:28][C][api:023]: Setting up Home Assistant API server...
[17:56:28][C][ota:127]: Over-The-Air Updates:
[17:56:28][C][ota:128]:   Address: servo_test2.local:8266
[17:56:28][I][application:089]: setup() finished successfully!
[17:56:28][I][application:097]: esphome-core version 1.12.2 compiled on Apr 16 2019, 17:50:49
[17:56:28][C][wifi:371]: WiFi:
[17:56:28][C][wifi:253]:   SSID: 'not_my_ssid'
[17:56:28][C][wifi:254]:   IP Address: 192.168.0.51
[17:56:28][C][wifi:256]:   BSSID: BC:30:D9:BA:4B:E3
[17:56:28][C][wifi:257]:   Hostname: 'servo_test2'
[17:56:28][C][wifi:261]:   Signal strength: -60 dB ?_?¦
[17:56:28][C][wifi:262]:   Channel: 5
[17:56:28][C][wifi:263]:   Subnet: 255.255.255.0
[17:56:28][C][wifi:264]:   Gateway: 192.168.0.1
[17:56:28][C][wifi:265]:   DNS1: 192.168.0.1
[17:56:28][C][wifi:266]:   DNS2: 0.0.0.0
[17:56:28][C][logger:142]: Logger:
[17:56:28][C][logger:143]:   Level: VERBOSE
[17:56:28][C][logger:144]:   Log Baud Rate: 115200
[17:56:28][C][logger:145]:   Hardware UART: UART0
[17:56:28][C][api:101]: API Server:
[17:56:28][C][api:102]:   Address: servo_test2.local:6053
[17:56:28][C][ota:127]: Over-The-Air Updates:
[17:56:28][C][ota:128]:   Address: servo_test2.local:8266
[17:56:28][I][application:109]: Running through first loop()
[17:56:28][I][application:136]: First loop finished successfully!
[17:56:34]pm open,type:0 0
[17:58:03][V][api:515]: Hello from client: 'aioesphomeapi (192.168.0.5)'
[17:58:03][D][api:073]: Disconnecting aioesphomeapi (192.168.0.5)
[17:58:16][I][ota:152]: Boot seems successful, resetting boot loop counter.
Serial port closed!

mr-sneezy, 2019-04-16:
This is the additional code that WiFi fails to connect with. I have also tried without the pin multisensor gpio code (just servo code) and that failed also on the first module...
For this repeat test there are NO PIR sensor or servo connected, so load on the NodeMCU is only the modules own .
The failing code (additions after ota:)
----------------------------------------------------
esphome:
  name: servo_test2
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

# Enable logging
logger:
  level: VERBOSE

# Enable Home Assistant API
api:

ota:

servo:
  - id: my_servo
    output: pwm_output

output:
  - platform: gpio
    pin: D1
    inverted: False
    id: gpio_d1_led
    
  - platform: esp8266_pwm
    id: pwm_output
    pin: D6
    frequency: 50 Hz

binary_sensor:
  - platform: gpio
    pin: D5
    name: "Multisensor Motion"
    device_class: motion
    
    on_press:
      then:
        - output.turn_on: gpio_d1_led
        - servo.write:
            id: my_servo
            level: 80.0%          
    on_release:
      then:
        - output.turn_off: gpio_d1_led
        - servo.write:
            id: my_servo
----------------------------------------------
The failing log from ESPhomeFlasher up until I yanked the plug on it. 
Using 'COM3' as serial port.
Connecting....
Detecting chip type... ESP8266
Connecting....

Chip Info:
 - Chip Family: ESP8266
 - Chip Model: ESP8266EX
 - Chip ID: 00264080
 - MAC Address: 2C:3A:E8:26:40:80
Uploading stub...
Running stub...
Stub running...
 - Flash Size: 4MB
[... 224 lines cut ...]
[18:28:28][V][wifi_esp8266:354]: Event: Disconnected ssid='not_my_ssid' bssid=00:00:00:00:00:00 reason='AP Not Found'
[18:28:28]wifi evt: 1
[18:28:28]STA disconnect: 201
[18:28:28][W][wifi:399]: Error while connecting to network.
[18:28:28][W][wifi:429]: Restarting WiFi adapter...
[18:28:28][V][wifi_esp8266:036]: Disabling STA.
[18:28:28]del if0
[18:28:28]usl
[18:28:28]mode : null
[18:28:28][V][wifi_esp8266:392]: Event: Changed Mode old=STA new=OFF
[18:28:28]wifi evt: 8
[18:28:28][I][wifi:167]: WiFi Connecting to 'not_my_ssid'...
[18:28:28][V][wifi:169]: Connection Params:
[18:28:28][V][wifi:170]:   SSID: 'not_my_ssid'
[18:28:28][V][wifi:173]:   BSSID: BC:30:D9:BA:4B:E3
[18:28:28][V][wifi:177]:   Password: 'not_my_pw'
[18:28:28][V][wifi:179]:   Channel: 5
[18:28:28][V][wifi:189]:   Using DHCP IP
[18:28:28][V][wifi:191]:   Hidden: NO
[18:28:28][V][wifi_esp8266:034]: Enabling STA.
[18:28:28]mode : sta(2c:3a:e8:26:40:80)
[18:28:28]add if0
[18:28:28][V][wifi_esp8266:392]: Event: Changed Mode old=OFF new=STA
[18:28:28]wifi evt: 8
Serial port closed!

OttoWinter, 2019-04-16:
Have you tried `fast_connect`?

mr-sneezy, 2019-04-17:
I've repeated the servo code test again anyway on the second NodeMCU with **fast_connect** active, as I can't think of anything much else to try, or what I might be repeatedly getting wrong. Servo control is the main reason I started this particular project so I can't go further till I sort it out.
Result log below.
( BTW I have purchased a ESP32 board to try out when it arrives in three weeks, maybe it works OK on the 32bit boards)
-------------------------------------------------------------
Using 'COM3' as serial port.
Connecting....
Detecting chip type... ESP8266
Connecting....

Chip Info:
 - Chip Family: ESP8266
 - Chip Model: ESP8266EX
 - Chip ID: 00264080
 - MAC Address: 2C:3A:E8:26:40:80
Uploading stub...
Running stub...
Stub running...
 - Flash Size: 4MB
 - Flash Mode: dout
 - Flash Frequency: 40MHz
Erasing flash (this may take a while)...
Chip erase completed successfully in 6.9s
Compressed 347376 bytes to 238288...
Wrote 347376 bytes (238288 compressed) at 0x00000000 in 21.1 seconds (effective 132.0 kbit/s)...
Hash of data verified.

Leaving...
Hard Resetting...
Done! Flashing is complete!

Showing logs:
[20:00:06];l
[20:00:06]SDK:2.2.1(cfd48f3)/Core:2.4.2/lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1-13-g163bb82)/BearSSL:6d1cefc
[20:00:06][I][logger:119]: Log initialized
[20:00:06][C][ota:484]: There have been 1 suspected unsuccessful boot attempts.
[20:00:06][I][application:055]: Running through setup()...
[20:00:06][V][application:056]: Sorting components by setup priority...
[20:00:06][C][output.gpio:019]: Setting up GPIO Binary Output...
[20:00:06][C][output.esp8266_pwm:024]: Setting up ESP8266 PWM Output...
[20:00:06][C][binary_sensor.gpio:016]: Setting up GPIO binary sensor 'Multisensor Motion'...
[20:00:06][D][binary_sensor:037]: 'Multisensor Motion': Sending state ON
[20:00:06][C][wifi:027]: Setting up WiFi...
[20:00:06][V][wifi_esp8266:034]: Enabling STA.
[20:00:06][V][wifi_esp8266:044]: Disabling AP.
[20:00:06]bcn 0
[20:00:06]del if1
[20:00:06]usl
[20:00:06]mode : sta(2c:3a:e8:26:40:80)
[20:00:06]add if0
[20:00:06][V][wifi_esp8266:392]: Event: Changed Mode old=AP new=STA
[20:00:06]wifi evt: 8
[20:00:06]sleep disable
[20:00:06][I][wifi:167]: WiFi Connecting to 'not_my_ssid'...
[20:00:06][V][wifi:169]: Connection Params:
[20:00:06][V][wifi:170]:   SSID: 'not_my_ssid'
[20:00:06][V][wifi:175]:   BSSID: Not Set
[20:00:06][V][wifi:177]:   Password: 'not_my_pw'
[20:00:06][V][wifi:181]:   Channel: Not Set
[20:00:06][V][wifi:189]:   Using DHCP IP
[20:00:06][V][wifi:191]:   Hidden: NO
[20:00:09]scandone
[20:00:09][V][wifi_esp8266:354]: Event: Disconnected ssid='not_my_ssid' bssid=00:00:00:00:00:00 reason='AP Not Found'
[20:00:09]wifi evt: 1
[20:00:09]STA disconnect: 201
[20:00:09][W][wifi:399]: Error while connecting to network.
[20:00:09][W][wifi:429]: Restarting WiFi adapter...
[20:00:09][V][wifi_esp8266:036]: Disabling STA.
[20:00:09]del if0
[20:00:09]usl
[20:00:09]mode : null
[... 91 lines cut ...]
[20:00:31][V][wifi_esp8266:354]: Event: Disconnected ssid='not_my_ssid' bssid=00:00:00:00:00:00 reason='AP Not Found'
[20:00:31]wifi evt: 1
[20:00:31]STA disconnect: 201
[20:00:31][W][wifi:399]: Error while connecting to network.
[20:00:31][W][wifi:429]: Restarting WiFi adapter...
[20:00:31][V][wifi_esp8266:036]: Disabling STA.
[20:00:31]del if0
[20:00:31]usl
[20:00:31]mode : null
[20:00:31][V][wifi_esp8266:392]: Event: Changed Mode old=STA new=OFF
[20:00:31]wifi evt: 8
[20:00:31][I][wifi:167]: WiFi Connecting to 'not_my_ssid'...
[20:00:31][V][wifi:169]: Connection Params:
[20:00:31][V][wifi:170]:   SSID: 'not_my_ssid'
[20:00:31][V][wifi:175]:   BSSID: Not Set
[20:00:31][V][wifi:177]:   Password: 'not_my_pw'
[20:00:31][V][wifi:181]:   Channel: Not Set
[20:00:31][V][wifi:189]:   Using DHCP IP
[20:00:31][V][wifi:191]:   Hidden: NO
[20:00:31][V][wifi_esp8266:034]: Enabling STA.
[20:00:31]mode : sta(2c:3a:e8:26:40:80)
[20:00:31]add if0
[20:00:31][V][wifi_esp8266:392]: Event: Changed Mode old=OFF new=STA
[20:00:31]wifi evt: 8
Serial port closed!

mr-sneezy, 2019-04-22:
Could my issue be caused by the Servo codes 50hz PWM output repetition interrupting WiFi too often ?
I assume other people have had success with WiFi and Servo running together...

OttoWinter, 2019-04-22:
> Could my issue be caused by the Servo codes 50hz PWM output repetition interrupting WiFi too often ?

I don't think so. There's plenty of people using ESP8266 PWM at 1kHz or even above with no issue. Plus if that were the issue then it would show up after the connection is established. The initial connection attempt (probe request in this case) is done at a low datarate.

mr-sneezy, 2019-04-23:
Weird then. This is my 6th or 7th NodeMCU on my HA, and have not encountered WiFi connect issues before. I don't think I'm doing anything different physically with this one that wasn't done with the others either. All the testing in the last week has been with no servo connected to eliminate noise/loading issues. When I get back I'll try faster/slower PWM rates and percentages, and maybe try another version of ESP8266 module (when I have one).

mr-sneezy, 2019-04-29:
It's still weird but I know a bit more tonight after another 4 hours of testing... 
I borrowed a genuine Adafruit Hazzah ESP board and a generic ESP8266MOD breakout board, and tried the same YAML on those (on a new created entities in the dashboard) and stuff me both work. And I hammered both by making trivial changes to the YAML and uploading via OTA, and all iterations connected and ran fine with the servo code included. Tried the same YAML on the two NodeMCU's again, and they still fail to connect (unless I stip out the servo code). 
What's also interesting is that if I compile & download the working binary from the Huzzah or other breakout dashboard, and load it to either NodeMCU via ESPhomeflasher, they fail to connect with it... 
 So now that I have the other two boards to test with successfully it looks like the two NodeMCU's have some strange issue when the servo code is added. Maybe they are defective or have faked memory or something. Is that sort of thing possible with these things ?

OttoWinter, 2019-05-01:
> Maybe they are defective or have faked memory or something. Is that sort of thing possible with these things ?

Certainly possible, as to the probability of that happening to exactly two units I don't know. However I also don't have other reasonable explanations at this time.

OttoWinter, 2019-06-18:
Moving the discussion to #455