esphome/issues #221: ESPHome + ODORID-C2 + HASSIO | Cannot install.

JasonSwindle, 2019-04-12:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

HASSIO 0.91.0

_arch_: aarch64

**ESP (ESP32/ESP8266, Board/Sonoff):**

None yet, installing software to get started.

**Affected component:**

**Description of problem:**
Trying to use ESPHOME on my ODROID-C2 powered by HASSIO, I am being told:

```
This addon is not available on your system.
```

but I see there is an docker image for https://hub.docker.com/r/esphome/esphome-hassio-base-aarch64

**Additional information and things you've tried:**

N\A

deece, 2019-04-14:
I have the same setup, but on an Orange Pi Prime (also aarch64) and see the same problem.

OttoWinter, 2019-04-15:
Mostly a duplicate of https://github.com/esphome/issues/issues/129

closed 2019-04-15 by OttoWinter

JasonSwindle, 2019-04-15:
These are not a duplicate, as this error does not allow the install in HASSIO vs #129 is a firmware compile issue.

OttoWinter, 2019-04-15:
@JasonSwindle Yes, that's why it's "mostly" a duplicate. I manually turned off aarch64 support in the hass.io config file for this addon because _it does not work_ as per #129. So #129 is the reason for this