esphome/issues #222: Sonoff 4CH Pro Rev2 button 4 not working

lblokland, 2019-04-12:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

using a 4ch Sonoff with ESPHome and Home Assistant

**ESP (ESP32/ESP8266, Board/Sonoff):**

Sonoff 4CH Pro rev2

**Affected component:**

https://esphome.io/devices/sonoff_4ch.html

**Description of problem:**

Button 4 is not doing anything

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

esphome:
  name: sonoff_4pro_1
  platform: ESP8266
  board: esp01_1m
  
web_server:
  port: 80

wifi:
  ssid: "10"
  password: ".."
  manual_ip:
    static_ip: 192.168.178.17
    gateway: 192.168.178.1
    subnet: 255.255.255.0
    dns1: 192.168.178.1
    
binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    name: "Sonoff 4pro1 Button 1"
  - platform: gpio
    pin:
      number: GPIO9
      mode: INPUT_PULLUP
      inverted: True
    name: "Sonoff 4pro1 Button 2"
  - platform: gpio
    pin:
      number: GPIO10
      mode: INPUT_PULLUP
      inverted: True
    name: "Sonoff 4pro1 Button 3"
  - platform: gpio
    pin:
      number: GPIO14
      mode: INPUT_PULLUP
      inverted: True
    name: "Sonoff 4pro1 Button 4"
  - platform: status
    name: "Sonoff 4pro1 Status"

switch:
  - platform: gpio
    name: "Sonoff 4pro1 Relay 1"
    pin: GPIO12
  - platform: gpio
    name: "Sonoff 4pro1 Relay 2"
    pin: GPIO5
  - platform: gpio
    name: "Sonoff 4pro1 Relay 3"
    pin: GPIO4
  - platform: gpio
    name: "Sonoff 4pro1 Relay 4"
    pin: GPIO15

output:
  # Register the blue LED as a dimmable output ....
  - platform: esp8266_pwm
    id: blue_led
    pin: GPIO13
    inverted: True

light:
  # ... and then make a light out of it.
  - platform: monochromatic
    name: "Sonoff 4pro1 Blue LED"
    output: blue_led

debug:
# Enable logging
logger:
  level: debug

# Enable Home Assistant API
api:
  password: '...'

ota:
  password: '..'

```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

Enabled debug and the webinterface, when pressing button 4 nothing is happening on the debog part of the web interface.

OttoWinter, 2019-04-15:
Likely you're using the wrong pin - don't know what version of the sonoff 4ch the guide was written against.

Please verify GPIO14 is the correct pin for button 4

lblokland, 2019-04-16:
I cleared all memory and uploaded the image again and now its working jus! fine!

closed 2019-04-16 by lblokland