esphome/issues #224: ESP32 Cam W OV2640 OTA issue

giuarb, 2019-04-13:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP32S AI Thinker with OV2640

**Affected component:**
https://esphome.io/components/esp32_camera.html?highlight=camera

**Description of problem:**
I flashed it using my FTDI adapter connected to RP3 USB, it worked, it shows up in integration, very stable (left on for 24 hours) I then tried to upload via OTA.
I did flash and camera worked inconsistently, I then reflashed via FTDI and it was stable again

This is the configuration I have used, the only board that worked for me is the esp-wrover-kit

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphome:
name: esp32cam
platform: ESP32
board: esp-wrover-kit

wifi:
networks:
- ssid: 'yournetwork'
password: 'your password'
- ssid: 'yournetworkbackup'
password: 'yourpassword'

# Enable logging
logger:

# Enable Home Assistant API
api:
password: '123456'

ota:
password: '123456'

# Example configuration entry
esp32_camera:
external_clock:
pin: GPIO0
frequency: 20MHz
i2c_pins:
sda: GPIO26
scl: GPIO27
data_pins: [GPIO5, GPIO18, GPIO19, GPIO21, GPIO36, GPIO39, GPIO34, GPIO35]
vsync_pin: GPIO25
href_pin: GPIO23
pixel_clock_pin: GPIO22
power_down_pin: GPIO32

# Image settings
name: Esp32 Cam``

**Traceback (if applicable):**
<!--
Please copy the traceback here if compilation is failing. If possible, also connect to the ESP over USB and copy its logs into the backticks.
-->
```

```

**Additional information and things you've tried:**
Link to the ESP32S and OV2640 bundle I am using
https://usa.banggood.com/Geekcreit-ESP32-CAM-WiFi-bluetooth-Camera-Module-Development-Board-ESP32-With-Camera-Module-OV2640-p-1394679.html?rmmds=myorder&fbclid=IwAR0wMe4_TF-wOKmKADSA2O45a3zK7w0X21XM1LtbFPgoOvIiS4GLvqcoYeE&cur_warehouse=CN

<!-- LEAVE THIS LINE AS-IS AND DON'T DELETE IT, OTHERWISE THE ISSUE WILL BE CLOSED AUTOMATICALLY. -->

OttoWinter, 2019-04-15:
Could be the camera overheating - from my past experience the symptoms are similar (doesn't work after being restarted).

Have you tried the test pattern as the documentation says? If that works, then it's the camera overheating.

OttoWinter, 2019-07-11:
Closing due to inactivity

closed 2019-07-11 by OttoWinter