esphome/issues #226: Displaying ASCII characters on PCF8574 LCD Display

anybody84, 2019-04-13:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 (d1_mini)
**Affected component:**

https://esphome.io/components/display/lcd_display.html

**Description of problem:**
There is an issue with displaying ASCII characters on PCF8574 LCD Display. I copied an example  for displaying `°C` from here:
[https://esphome.io/components/display/index.html#display-printf](url)

The example is:
```yaml
display:
  - platform: ...
    # ...
    lambda: |-
      // %% - literal % sign
      it.printf(0, 0, id(my_font), "Temperature: %.1f°C, Humidity: %.1f%%", id(temperature).state, id(humidity).state);
```
However, when I'm using this expression `%.1f°C` it is displaying this:
[https://drive.google.com/file/d/1ioubeic1sQeoN9DOMCFNpmWAYAiVPTMK/view](url)

**Problem-relevant YAML-configuration entries:**
```yaml
sensor:
  - platform: bmp280
    temperature:
      name: "Salon temp"
      id: temp
      unit_of_measurement: "°C"
      accuracy_decimals: 2
      oversampling: 16x
      filters:
        - offset: 0.0
display:
  - platform: lcd_pcf8574
    dimensions: 16x2
    address: 0x27
    lambda: |-
      it.printf(0, 0, "Home: %.2f°C", id(temp).state);
```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

HFsi, 2019-04-13:
Yep, I've the same issue

cyberplant, 2019-04-15:
I'm having the same issue, populating the LCD from a text field from home assistant.

I've tried several ASCII codes over chr(128) and found that some symbols repeat all the time: 
[image]

[image]

OttoWinter, 2019-04-15:
These displays don't support all characters.

ESPHome just directs the raw character data directly to the display - if the display doesn't support a character it will not be displayed.

You can google the character table of your display. Some displays have extended character sets to display special characters (like japanese characters in the image of @cyberplant).

ESPHome just forwards this data in raw, it can't make the display stuff it doesn't support.

closed 2019-04-15 by OttoWinter