esphome/issues #227: NeoPixelBus doesn't get updated for addressable effects

ActuallyRuben, 2019-04-14:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 ESP-12 type board (using nodemcu as platform in the config)

**Affected component:**

[NeoPixelBus light](https://esphome.io/components/light/neopixelbus.html)

**Description of problem:**
Addressable light effects do not get updated properly. They only seem to update at random (couple second) intervals, or immediately when sending a turn_on event.
Normal effects do work as intended.

**Problem-relevant YAML-configuration entries:**
```yaml
light:
 - platform: neopixelbus
   name: "Complete LED strip"
   id: ledstrip
   num_leds: 93
   pin: GPIO3
 - platform: partition
   name: "LED strip top"
   segments:
    - id: ledstrip
      from: 0
      to: 87
   effects:
    - addressable_rainbow:
    - addressable_color_wipe:
    - random:
         transition_length: 10s
         update_interval: 10s
```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**
I tried fixing this issue myself. By inserting `ESP_LOGD(TAG, "Updating LEDstrip");` in the loop method of the component, just before `this->controller_->Show();`. (On line 90 in [this file](https://github.com/esphome/esphome-core/blob/dev/src/esphome/light/neo_pixel_bus_light_output.tcc) to be specific), the issue resolves itself, but of course then your log gets filled with useless 'Updating LEDstrip' statements. I'm not sure how it's related, but the bug seems to be related to the logger. I found issue #182, could it be related?

ActuallyRuben, 2019-04-14:
Looking further into the issue, I've found some more peculiarities:

1. Disabling logging output over serial:
This seems to have no effect at all.
2. Disabling logging completely:
The strip can't be updated anymore through sending turn_on events, it only updates after random intervals.
3. Placing a logging statement after `this->controller_->Show();` instead of in front of it:
This makes the random updates happen more often, but it doesn't animate smoothly at all.

I'm still looking into it, but it seems a very low-level conflict somewhere.

ActuallyRuben, 2019-04-14:
The delay caused by the logger seems to fix the issue. I'm still not 100% certain why.
I've found it can be avoided by not enabling the high frequency mode at the start of an addressable effect, I made a pull request with that change. esphome/esphome-core#586

OttoWinter, 2019-07-11:
Has been fixed some time ago.

closed 2019-07-11 by OttoWinter