esphome/issues #228: Documentation improvements for Mirabella Genio bulbs

deece, 2019-04-14:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Mirabella Genio Cold + Warm White Bulb

Mirabella Genio Cold + Warm White Bulb
https://esphome.io/cookbook/mirabella-genio-bulb.html

The example provided did not work for me.
The warm white LED should be on GPIO13, and the on_boot stanza to turn the light on at boot did not work. Setting the priority to 49 fixed this.

OttoWinter, 2019-04-15:
CC @cryptelli (https://github.com/esphome/esphome-docs/pull/184)

cryptelli, 2019-04-16:
@deece Thanks for bringing this to my attention, and the timing is perfect since I happened to reflash my bulbs last night and encountered the same issue with the `on_boot` priority no longer working. The fix (as you've found out) is to lower it, so I've set mine all to `50.0`. The upside is the transition is smoother compared to it being set to `100`.

Also, in my original configs the warm white GPIO was set to 13, accidental typo in my configuration. 🤦‍♂️ 

@OttoWinter did something change in 0.12.1 causing the priority to no longer function when set to `100`?

I'll submit a PR to address these issues.

OttoWinter, 2019-04-16:
> @OttoWinter did something change in 0.12.1 causing the priority to no longer function when set to 100?

Yes it did actually, now lights set the state to OFF when restoring values (as they should). However, I don't consider boot priorities to be breaking changes, since then there would be many breaking changes per release (and it's mostly internal anyway)

cryptelli, 2019-04-20:
Understood, except restoring the light to on when it's been accidentally turned off via the switch is preferred. All my LIFX bulbs work this way.

Moving forward is setting `on_boot` to `50.0` going to be future-proofed or can you recommend a better way of handling this?

OttoWinter, 2019-04-22:
> Moving forward is setting on_boot to 50.0 going to be future-proofed or can you recommend a better way of handling this?

I can't really guarantee it - and I don't really know of a way to do this in a consistent way. What *would* work is basically setting the boot priority to something very low - then the chances of getting hit by this again are at least very slim.

OttoWinter, 2019-07-11:
Closing due to inactivity

closed 2019-07-11 by OttoWinter

cryptelli, 2019-07-16:
I wasn't sure where to go with this, hence the inactivity. 

However it seems you've now introduced a `restore_mode` option in 1.13.0 and updated my config to reflect this change, works like a charm and smoother than `on_boot`.

Awesome work, thank you 👍