esphome/issues #229: Template switch ignoring id, converting name instead

gitolicious, 2019-04-14:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**

Sonoff, Wemos D1 mini

**Affected component:**

https://esphome.io/components/switch/template.html

**Description of problem:**
When creating a template switch with both, `id` and `name`, the `id` is ignored. Instead an `id` is auto-generated from the `name`.
According to the [documentation](https://esphome.io/components/switch/template.html#configuration-variables), it should be possible to set the `id` manually:

> **id** (_Optional_, ID): Manually specify the ID used for code generation.

In below example, the switch will get the id `name`. This can be seen in the web server source code and in the HomeAssistant integrations and entities list.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: test
  platform: ESP8266
  board: d1_mini

wifi: !include templates/wifi.yaml
api:
web_server:

switch:
  - platform: template
    id: id
    name: Name
```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**
When only an `id` is given, it is properly used for the switch id.

gitolicious, 2019-04-14:
Actually after digging a bit deeper into this, I think I misunderstood the meaning of `id`. It's not meant to be reflected in the HomeAssistant `entity_id` but only used internally within the ESPHome yaml, right?
Is there any way to influence the `entity_id`? I think the `name` should be something descriptive whereas the `entity_id` should be concise. Would be nice to configure them individually.

HFsi, 2019-04-14:
That's right, id is completely internal, not changing anything in HA

gitolicious, 2019-04-14:
Thanks for the explanation. I will close this and probably open a feature request.

closed 2019-04-14 by gitolicious

OttoWinter, 2019-04-15:
That's correct, ID is for internal use - HA builds the entity_id based off of the name itself. ESPHome does not (and will not) have any influence over this.

gitolicious, 2019-04-15:
Too bad. But I will find my way around it for now.