esphome/issues #230: sensor: -platform: homeassistant not receiving values from HASSIO

ddean6232, 2019-04-14:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

HASSIO on RPI3B+ with ESPHOME 
Wemos ESP32 with OLED display 
Wemos D1 mini pro with DHT11 sensor
Sonoff POW with Current/Voltage/Power sensor

Wemos ESP32 with OLED ("esp32test" node)

- platform: homeassistant
https://esphome.io/components/sensor/homeassistant.html?highlight=homeassistant
https://esphome.io/components/api.html

**Description of problem:**
doesn't seem like ESPHOME is receiving values from HA.  The HA time will sync and display but the 'volt' value is reporting as NAN
have used CURL on the HA API and can poll the HA sensor and get the value.

**Problem-relevant YAML-configuration entries:**
```yaml
# Enable logging
logger:
  level: VERBOSE

# Enable Home Assistant API
api:

ota:

font:
  - file: "OpenSans-Regular.ttf"
    id: my_font
    size: 12

time:
  - platform: homeassistant
    id: time
    
sensor:
  - platform: homeassistant
    entity_id: sensor.multisensor_temperature
    id: temp
  - platform: homeassistant
    entity_id: sensor.multisensor_humidity
    id: humid
  - platform: homeassistant
    entity_id: sensor.hlw8012_voltage
    id: volt

i2c:
  sda: GPIO5
  scl: GPIO4
  scan: True

display:
  - platform: ssd1306_i2c
    model: "SSD1306 128x32"
    address: 0x3C
    lambda: |-
        it.strftime(0, 0, id(my_font), "%H:%M", id(time).now());
        it.printf(0, 20, id(my_font), "sensor: %.1f", id(volt).state);

```

**Traceback (if applicable):**

```
[13:49:31][C][time.homeassistant:032]: Home Assistant Time:
[13:49:31][C][time.homeassistant:033]:   Timezone: 'MST7MDT6,M3.2.0/2,M11.1.0/2'
[13:49:31][C][sensor.homeassistant:031]: Homeassistant Sensor 'temp'
[13:49:31][C][sensor.homeassistant:031]:   Unit of Measurement: ''
[13:49:31][C][sensor.homeassistant:031]:   Accuracy Decimals: 0
[13:49:31][C][sensor.homeassistant:032]:   Entity ID: 'sensor.multisensor_temperature'
[13:49:31][C][sensor.homeassistant:031]: Homeassistant Sensor 'humid'
[13:49:31][C][sensor.homeassistant:031]:   Unit of Measurement: ''
[13:49:32][C][sensor.homeassistant:031]:   Accuracy Decimals: 0
[13:49:32][C][sensor.homeassistant:032]:   Entity ID: 'sensor.multisensor_humidity'
[13:49:32][C][sensor.homeassistant:031]: Homeassistant Sensor 'volt'
[13:49:32][C][sensor.homeassistant:031]:   Unit of Measurement: ''
[13:49:32][C][sensor.homeassistant:031]:   Accuracy Decimals: 0
[13:49:32][C][sensor.homeassistant:032]:   Entity ID: 'sensor.hlw8012_voltage'
[13:49:32][C][api:101]: API Server:
[13:49:32][C][api:102]:   Address: esp32test.local:6053
[13:49:32][C][ota:127]: Over-The-Air Updates:
[13:49:32][C][ota:128]:   Address: esp32test.local:3232
[13:49:32][D][time.homeassistant:029]: Synchronized time: Sun Apr 14 13:49:31 2019
```

**Additional information and things you've tried:**
removed SSL on HASSIO
tried getting values from many different sensors from HA on different devices
using no passwords on the APIs
CURL results from HA API returning expected values/results

ddean6232, 2019-04-14:
Tried another ESP32 module.... nada
Tried another NODEMCU ..... nada
Got it to work on a d1_mini...   ????  Why not ESP32 or NODEMCU???

ddean6232, 2019-04-14:
maybe the ESPHOME compile is not linking the correct libraries for the ESP32 boards to integrated the HA API...?

OttoWinter, 2019-04-15:
If it's synchronizing time that means the connection is working - that also rules out any error in the connection related code (and thus a difference between esp8266 and esp32).

Most likely this is just a user error somewhere (maybe mistyped entity ID etc). Try restarting Hass.io and enabling esphome logging in your HA config in logger->logs section `aioesphomeapi.client: debug`

ddean6232, 2019-04-15:
The EXACT working code on a 8266 base and everything functions. I don't believe I have a user error with the code

ddean6232, 2019-04-15:
This is what I get from the 8266 base device log when I add the HA entity polling code to the esphome device config

Polling code
```
sensor:
  - platform: homeassistant
    entity_id: sensor.multisensor_temperature
    id: temp
  - platform: homeassistant
    entity_id: sensor.multisensor_humidity
    id: humid
  - platform: homeassistant
    entity_id: sensor.hlw8012_voltage
    id: volt
```
Device Log
```
INFO Connecting to 192.168.0.61:6053 (192.168.0.61)
INFO Successfully connected to 192.168.0.61
[15:30:18][I][application:097]: esphome-core version 1.12.2 compiled on Apr 15 2019, 15:27:45
[15:30:18][C][wifi:371]: WiFi:
[15:30:18][C][wifi:253]:   SSID: [redacted]
[15:30:18][C][wifi:254]:   IP Address: 192.168.0.61
[15:30:18][C][wifi:256]:   BSSID: [redacted]
[15:30:18][C][wifi:257]:   Hostname: 'multisensor'
[15:30:18][C][wifi:261]:   Signal strength: -43 dB ▂▄▆█
[15:30:18][C][wifi:262]:   Channel: 4
[15:30:18][C][wifi:263]:   Subnet: 255.255.255.0
[15:30:18][C][wifi:264]:   Gateway: 192.168.0.1
[15:30:18][C][wifi:265]:   DNS1: 1.1.1.1
[15:30:18][C][wifi:266]:   DNS2: 1.0.0.1
[15:30:18][C][binary_sensor.gpio:022]: GPIO Binary Sensor 'Multisensor Motion'
[15:30:18][C][binary_sensor.gpio:022]:   Device Class: 'motion'
[15:30:18][C][binary_sensor.gpio:023]:   Pin: GPIO14 (Mode: INPUT)
[15:30:18][C][output.esp8266_pwm:029]: ESP8266 PWM:
[15:30:18](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[15:30:18][C][output.esp8266_pwm:031]:   Frequency: 1000.0 Hz
[15:30:18][C][output.esp8266_pwm:029]: ESP8266 PWM:
[15:30:18][C][output.esp8266_pwm:030]:   Pin: GPIO4 (Mode: OUTPUT)
[15:30:18][C][output.esp8266_pwm:031]:   Frequency: 1000.0 Hz
[15:30:18][C][output.esp8266_pwm:029]: ESP8266 PWM:
[15:30:18][C][output.esp8266_pwm:030]:   Pin: GPIO0 (Mode: OUTPUT)
[15:30:18][C][output.esp8266_pwm:031]:   Frequency: 1000.0 Hz
[15:30:18][C][logger:142]: Logger:
[15:30:18][C][logger:143]:   Level: DEBUG
[15:30:18][C][logger:144]:   Log Baud Rate: 115200
[15:30:18][C][logger:145]:   Hardware UART: UART0
[15:30:18][C][light.state:225]: Light 'Multisensor Light'
[15:30:18][C][light.state:227]:   Default Transition Length: 1000 ms
[15:30:18][C][light.state:228]:   Gamma Correct: 2.80
[15:30:18][C][sensor.dht:035]: DHT:
[15:30:18][C][sensor.dht:036]:   Pin: GPIO13 (Mode: OUTPUT)
[15:30:18][C][sensor.dht:040]:   Model: DHT11
[15:30:18][C][sensor.dht:045]:   Update Interval: 15000 ms
[15:30:18][C][sensor.dht:047]:   Temperature 'Multisensor Temperature'
[15:30:18][C][sensor.dht:047]:     Unit of Measurement: '°C'
[15:30:18][C][sensor.dht:047]:     Accuracy Decimals: 1
[15:30:18][C][sensor.dht:048]:   Humidity 'Multisensor Humidity'
[15:30:18][C][sensor.dht:048]:     Unit of Measurement: '%'
[15:30:18][C][sensor.dht:048]:     Accuracy Decimals: 0
[15:30:18][C][sensor.dht:048]:     Icon: 'mdi:water-percent'
[15:30:18][C][sensor.homeassistant:031]: Homeassistant Sensor 'temp'
[15:30:18][C][sensor.homeassistant:031]:   Unit of Measurement: ''
[15:30:18][C][sensor.homeassistant:031]:   Accuracy Decimals: 0
[15:30:18][C][sensor.homeassistant:032]:   Entity ID: 'sensor.multisensor_temperature'
[15:30:18][C][sensor.homeassistant:031]: Homeassistant Sensor 'humid'
[15:30:18][C][sensor.homeassistant:031]:   Unit of Measurement: ''
[15:30:18][C][sensor.homeassistant:031]:   Accuracy Decimals: 0
[15:30:18][C][sensor.homeassistant:032]:   Entity ID: 'sensor.multisensor_humidity'
[15:30:18][C][sensor.homeassistant:031]: Homeassistant Sensor 'volt'
[15:30:18][C][sensor.homeassistant:031]:   Unit of Measurement: ''
[15:30:18][C][sensor.homeassistant:031]:   Accuracy Decimals: 0
[15:30:18][C][sensor.homeassistant:032]:   Entity ID: 'sensor.hlw8012_voltage'
[15:30:18][C][api:101]: API Server:
[15:30:18][C][api:102]:   Address: 192.168.0.61:6053
[15:30:18][C][ota:127]: Over-The-Air Updates:
[15:30:18][C][ota:128]:   Address: 192.168.0.61:8266
[15:30:18][D][sensor.homeassistant:026]: 'sensor.multisensor_temperature': Got state 22.00
[15:30:18][D][sensor.homeassistant:026]: 'sensor.multisensor_humidity': Got state 21.00
[15:30:18][D][sensor.homeassistant:026]: 'sensor.hlw8012_voltage': Got state 118.00
[15:30:19][D][sensor.dht:066]: Got Temperature=22.0°C Humidity=20.0%
[15:30:19][D][sensor.homeassistant:026]: 'sensor.multisensor_humidity': Got state 20.00
[15:30:34][D][sensor.dht:066]: Got Temperature=22.0°C Humidity=21.0%
[15:30:34][D][sensor.homeassistant:026]: 'sensor.multisensor_humidity': Got state 21.00
[15:30:49][D][sensor.dht:066]: Got Temperature=22.0°C Humidity=20.0%
[15:30:49][D][sensor.homeassistant:026]: 'sensor.multisensor_humidity': Got state 20.00
[15:31:00][D][sensor.homeassistant:026]: 'sensor.hlw8012_voltage': Got state 117.80
[15:31:04][D][sensor.dht:066]: Got Temperature=22.0°C Humidity=20.0%
[15:31:10][D][sensor.homeassistant:026]: 'sensor.hlw8012_voltage': Got state 117.60
[15:31:19][D][sensor.dht:066]: Got Temperature=22.0°C Humidity=20.0%
[15:31:34][D][sensor.dht:066]: Got Temperature=22.0°C Humidity=20.0%
[15:31:49][D][sensor.dht:066]: Got Temperature=22.0°C Humidity=21.0%
[15:31:49][D][sensor.homeassistant:026]: 'sensor.multisensor_humidity': Got state 21.00
[15:32:04][D][sensor.dht:066]: Got Temperature=22.0°C Humidity=21.0%
[15:32:09][I][ota:152]: Boot seems successful, resetting boot loop counter.
[15:32:10][D][sensor.homeassistant:026]: 'sensor.hlw8012_voltage': Got state 117.70
[15:32:19][D][sensor.dht:066]: Got Temperature=22.0°C Humidity=21.0%
[15:32:34][D][sensor.dht:066]: Got Temperature=22.0°C Humidity=21.0%
[15:32:49][D][sensor.dht:066]: Got Temperature=22.0°C Humidity=20.0%
[15:32:49][D][sensor.homeassistant:026]: 'sensor.multisensor_humidity': Got state 20.00
help_outline 
```

ddean6232, 2019-04-15:
Here is the log from a ESP32 base esphome device
```
INFO Reading configuration...
INFO Detected timezone 'MST' with UTC offset -7 and daylight savings time from 03/10/19 02:00:00 to 11/03/19 02:00:00
INFO Starting log output from tester1.local using esphome API
INFO Connecting to tester1.local:6053 (192.168.0.203)
INFO Successfully connected to tester1.local
[15:37:46][D][time.homeassistant:029]: Synchronized time: Mon Apr 15 15:37:46 2019
[15:37:47][I][application:097]: esphome-core version 1.12.2 compiled on Apr 14 2019, 17:12:12
[15:37:47][C][wifi:371]: WiFi:
[15:37:47][C][wifi:253]:   SSID: [redacted]
[15:37:47][C][wifi:254]:   IP Address: 192.168.0.203
[15:37:47][C][wifi:256]:   BSSID: [redacted]
[15:37:47][C][wifi:257]:   Hostname: 'tester1'
[15:37:47][C][wifi:261]:   Signal strength: -46 dB ▂▄▆█
[15:37:47][C][wifi:262]:   Channel: 4
[15:37:47][C][wifi:263]:   Subnet: 255.255.255.0
[15:37:47][C][wifi:264]:   Gateway: 192.168.0.1
[15:37:47][C][wifi:265]:   DNS1: 8.8.8.8
[15:37:47][C][wifi:266]:   DNS2: 8.8.4.4
[15:37:47][C][i2c:035]: I2C Bus:
[15:37:47][C][i2c:036]:   SDA Pin: GPIO5
[15:37:47][C][i2c:037]:   SCL Pin: GPIO4
[15:37:47][C][i2c:038]:   Frequency: 50000 Hz
[15:37:47][I][i2c:040]: Scanning i2c bus for active devices...
[15:37:47][I][i2c:047]: Found i2c device at address 0x3C
[15:37:47][C][logger:142]: Logger:
[15:37:47][C][logger:143]:   Level: DEBUG
[15:37:47][C][logger:144]:   Log Baud Rate: 115200
[15:37:47][C][logger:145]:   Hardware UART: UART0
[15:37:47][C][display.ssd1306:312]: I2C SSD1306
[15:37:47][C][display.ssd1306:312]:   Rotations: 0 °
[15:37:47][C][display.ssd1306:312]:   Dimensions: 128px x 32px
[15:37:47][C][display.ssd1306:313]:   Address: 0x3C
[15:37:47][C][display.ssd1306:314]:   Model: SSD1306 128x32
[15:37:47][C][display.ssd1306:316]:   External VCC: NO
[15:37:47][C][display.ssd1306:317]:   Update Interval: 5000 ms
[15:37:47][C][time.homeassistant:032]: Home Assistant Time:
[15:37:47][C][time.homeassistant:033]:   Timezone: 'MST7MDT6,M3.2.0/2,M11.1.0/2'
[15:37:47][C][sensor.homeassistant:031]: Homeassistant Sensor 'temp'
[15:37:47][C][sensor.homeassistant:031]:   Unit of Measurement: ''
[15:37:47][C][sensor.homeassistant:031]:   Accuracy Decimals: 0
[15:37:47][C][sensor.homeassistant:032]:   Entity ID: 'sensor.multisensor_temperature'
[15:37:47][C][sensor.homeassistant:031]: Homeassistant Sensor 'humid'
[15:37:47][C][sensor.homeassistant:031]:   Unit of Measurement: ''
[15:37:47][C][sensor.homeassistant:031]:   Accuracy Decimals: 0
[15:37:47][C][sensor.homeassistant:032]:   Entity ID: 'sensor.multisensor_humidity'
[15:37:47][C][sensor.homeassistant:031]: Homeassistant Sensor 'volt'
[15:37:47][C][sensor.homeassistant:031]:   Unit of Measurement: ''
[15:37:47][C][sensor.homeassistant:031]:   Accuracy Decimals: 0
[15:37:47][C][sensor.homeassistant:032]:   Entity ID: 'sensor.hlw8012_voltage'
[15:37:47][C][api:101]: API Server:
[15:37:47][C][api:102]:   Address: tester1.local:6053
[15:37:47][C][ota:127]: Over-The-Air Updates:
[15:37:47][C][ota:128]:   Address: tester1.local:3232
```

ddean6232, 2019-04-15:
I have attached the HASSIO system log with
```
logger:
  default: debug
  logs:
    aioesphomeapi.client: debug
```
in the configuration.yaml 
I don't see anything obvious on why the ESP32 would not be picking up the values
[HA_SYS_LOG.txt](https://github.com/esphome/issues/files/3082175/HA_SYS_LOG.txt)

erazor666, 2019-04-15:
I made a test, ran a brand new Wemos32 and pulled sensors, I'm not getting NAN values. However I don't run HASSIO, but Home Assistant in a docker environment on UNRAID. However, is there any difference between HASSIO and Home Assistant ran in a docker environment? (API)

@ddean6232 has ran with my yaml code for the ESP and substituted the senors names for his own, he is getting NAN for his sensors. If we try to use curl and query the Home Assistant API on the same sensors we get the json as expected.

I did some further testing (ran same yaml with board changed), I'm getting the NAN problem on a board with the WROOM32 chip but no on boards with a chip labeled ESPRESSIF (ESP-32s) (see enclosed photo)
[image]

ddean6232, 2019-04-15:
it is that all the 32s I have are ESP-WROOM-32 boards... I have been working with @e_razor and he has a ESP-32 board working.... maybe something in the board defs?

OttoWinter, 2019-04-16:
@ddean6232 I assure you it's not the board defs. There are very few places in the esphome codebase that are platform dependent - the homeassistant sensor is not one of those. If anything would happen, then it would be that the board simply doesn't connect to HA.

What I do see from the logs is that you don't appear to have the node set up in HA (at least there are no messages with `tester1` in them) - not even connect attempts from what I see.

Have you looked in the HA integrations screen if the node is added to HA?

ddean6232, 2019-04-16:
Wow!... that worked... I have never had to do that!  All my boards have just worked without a manual setup in 'Integrations'... sorry for the trouble!... all our (@erazor666) testing never lead us to that path.  I will make it standard practice to 'Integrate' every time now... Thanks

erazor666, 2019-04-16:
It seems, when i did the testing, I recycled the yaml and ip address for the node (I did remove it from Home Assistant and re-add it when changing board). I got some sort of "double integration" for a lack of a better word, noticed this morning that Home Assistant that I had two nodes integrated on top of each other in Home Assistant. Usually I restart HA when adding and removing nodes, I didn't on this occasion and somehow got a monkey in the works. This whole situation wasn't making a whole lot of sense, but I think its cleared up now.

closed 2019-04-17 by ddean6232