esphome/issues #231: homeassistant.service data templating / data_template variables not working

mmotley999, 2019-04-14:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

HomeAssistant venv, ESPHome Docker

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32
**Affected component:**

homeassistant.service

**Description of problem:**
I'm writing an app for controlling a ceiling fan (iFan02 running Tasmota), using a Nextion as an control panel.  Turn light off, turn fan off, increase/decrease fan.  That's all working just fine.

HomeAssistant expects the fan speed to be a string, like "low", "medium", or "high" (as far as I know).  I'm exposing the current speed setting to the app via a HomeAssistant template sensor (since it's an attribute in HA), and via a text_sensor in ESPHome.  I'm storing the value in a temporary variable, so I can manipulate it based on input from the Nextion.

```yaml
globals:
  - id: fan_speed
    type: char[10]
```

I'm trying to pass that string to a homeassistant.service call so that the speed can be set when the speed up/speed down buttons are pushed.

I appears that the "data:" section is not able to be lambda templated, because this doesn't work:

```yaml
- homeassistant.service
    service: fan.set_speed
    data:
      entity_id: $fan_entity
      speed: !lambda 'return id(fan_speed).c_str();'

Also, tried to use the variables outlined in https://esphome.io/components/api.html#homeassistant-service-action:

```yaml
- homeassistant.service:
    service: fan.set_speed
    variables:
    my_variable: |-
      return id(fan_speed).c_str();
                data:
                  entity_id: $fan_entity
                data_template:
                  speed: {{ my_variable }}
```
The variables method throws an error on validation:

```
ERROR Error while reading config: invalid key: "OrderedDict([('my_variable', None)])"
```

It's possible I'm simply doing this wrong.

**Problem-relevant YAML-configuration entries:**
```yaml
 # Button Up, ID 5
  - platform: nextion
    page_id: 0
    component_id: 5
    internal: yes
    name: "Speed Up"
    id: button_up
    on_press:
      then:
        if:
          condition:
            lambda: 'return(strcmp(id(fan_speed),"high") != 0);'
          then:
            - logger.log: "Increasing Fan Speed"
            - lambda: |-
                if(strcmp(id(fan_speed),"off") == 0) {
                  strcpy(id(fan_speed),"low");
                  ESP_LOGD("Changing to fan speed low");
                }
                else if(strcmp(id(fan_speed),"low") == 0) {
                  strcpy(id(fan_speed),"medium");
                  ESP_LOGD("Changing to fan speed medium");
                }
                else if(strcmp(id(fan_speed),"medium") == 0) {
                  strcpy(id(fan_speed),"high");
                  ESP_LOGD("Changing to fan speed high");
                }
            - homeassistant.service:
                service: fan.set_speed
                variables:
                  my_variable: |-
                      return id(fan_speed).c_str();
                data:
                  entity_id: $fan_entity
                data_template:
                  speed: {{ my_variable }}
                
```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

mmotley999, 2019-04-14:
Update....

I found out that the template string needs quotes (doh! on my part).

I went ahead and changed the global variable to an int (to remove some level of string handling).  Here's the relevant YAML:

```yaml
globals:
  - id: fan_speed
    type: int
    initial_value: '0'

  # Button Up, ID 5
  - platform: nextion
    page_id: 0
    component_id: 5
    internal: yes
    name: "Speed Up"
    id: button_up
    on_press:
      then:
        if:
          condition:
            lambda: 'return (id(fan_speed) != 3);'
          then:
            - logger.log: "Increasing Fan Speed"
            - lambda: |-
                id(fan_speed) += 1;
              
            - homeassistant.service:
                service: fan.set_speed
                data_template:
                  entity_id: $fan_entity
                  speed: "{{ my_speed }}"
                variables:
                  my_speed: |-
                    if(id(fan_speed) == 0) {
                      return("off");
                    }
                    if (id(fan_speed) == 1) {
                      return("low");
                    }
                    if (id(fan_speed) == 2) {
                      return("medium");
                    }
                    if (id(fan_speed) == 3) {
                      return("high");
                    }
```

However, now I'm getting a rather lengthy error when compiling.

Here's the offending code from the generated 'main.cpp':

```
  action_17->set_variables({api::TemplatableKeyValuePair("my_speed", [=]() {
        if(fan_speed->value() == 0) {
          return("off");
        }
        if (fan_speed->value() == 1) {
          return("low");
        }
        if (fan_speed->value() == 2) {
          return("medium");
        }
        if (fan_speed->value() == 3) {
          return("high");
        }
  })});
```
[... 59 lines cut ...]
In file included from .piolibdeps/esphome-core/src/esphome/component.h:7:0,
from .piolibdeps/esphome-core/src/esphome/api/api_server.h:8,
from .piolibdeps/esphome-core/src/esphome/application.h:6,
from .piolibdeps/esphome-core/src/esphome.h:1,
from src/main.cpp:3:
.piolibdeps/esphome-core/src/esphome/helpers.h:50:13: note: candidate: std::__cxx11::string esphome::to_string(std::__cxx11::string)
std::string to_string(std::string val);
^
.piolibdeps/esphome-core/src/esphome/helpers.h:51:13: note: candidate: std::__cxx11::string esphome::to_string(String)
std::string to_string(String val);
^
In file included from .piolibdeps/esphome-core/src/esphome/component.h:4:0,
from .piolibdeps/esphome-core/src/esphome/api/api_server.h:8,
from .piolibdeps/esphome-core/src/esphome/application.h:6,
from .piolibdeps/esphome-core/src/esphome.h:1,
from src/main.cpp:3:
/root/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:2134:2: error: 'std::function<_Res(_ArgTypes ...)>::_Requires<std::__and_<std::__not_<std::is_same<typename std::decay<_Tp>::type, std::function<_Res(_ArgTypes ...)> > >, std::__or_<std::is_void<_Tp>, std::is_convertible<decltype (std::__callable_functor(declval<typename std::decay<_Tp>::type&>())((declval<_ArgTypes>)()...)), _Res> > >, std::function<_Res(_ArgTypes ...)>&> std::function<_Res(_ArgTypes ...)>::operator=(_Functor&&) [with _Functor = esphome::api::TemplatableKeyValuePair::TemplatableKeyValuePair(std::__cxx11::string, T) [with T = setup()::<lambda()>; std::__cxx11::string = std::__cxx11::basic_string<char>]::<lambda()>; _Res = std::__cxx11::basic_string<char>; _ArgTypes = {}; std::function<_Res(_ArgTypes ...)>::_Requires<std::__and_<std::__not_<std::is_same<typename std::decay<_Tp>::type, std::function<_Res(_ArgTypes ...)> > >, std::__or_<std::is_void<_Tp>, std::is_convertible<decltype (std::__callable_functor(declval<typename std::decay<_Tp>::type&>())((declval<_ArgTypes>)()...)), _Res> > >, std::function<_Res(_ArgTypes ...)>&> = std::function<std::__cxx11::basic_string<char>()>&]', declared using local type 'esphome::api::TemplatableKeyValuePair::TemplatableKeyValuePair(std::__cxx11::string, T) [with T = setup()::<lambda()>; std::__cxx11::string = std::__cxx11::basic_string<char>]::<lambda()>', is used but never defined [-fpermissive]
operator=(_Functor&& __f)
^
/root/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:2134:2: error: 'std::function<_Res(_ArgTypes ...)>::_Requires<std::__and_<std::__not_<std::is_same<typename std::decay<_Tp>::type, std::function<_Res(_ArgTypes ...)> > >, std::__or_<std::is_void<_Tp>, std::is_convertible<decltype (std::__callable_functor(declval<typename std::decay<_Tp>::type&>())((declval<_ArgTypes>)()...)), _Res> > >, std::function<_Res(_ArgTypes ...)>&> std::function<_Res(_ArgTypes ...)>::operator=(_Functor&&) [with _Functor = esphome::api::TemplatableKeyValuePair::TemplatableKeyValuePair(std::__cxx11::string, T) [with T = setup()::<lambda()>; std::__cxx11::string = std::__cxx11::basic_string<char>]::<lambda()>; _Res = std::__cxx11::basic_string<char>; _ArgTypes = {}; std::function<_Res(_ArgTypes ...)>::_Requires<std::__and_<std::__not_<std::is_same<typename std::decay<_Tp>::type, std::function<_Res(_ArgTypes ...)> > >, std::__or_<std::is_void<_Tp>, std::is_convertible<decltype (std::__callable_functor(declval<typename std::decay<_Tp>::type&>())((declval<_ArgTypes>)()...)), _Res> > >, std::function<_Res(_ArgTypes ...)>&> = std::function<std::__cxx11::basic_string<char>()>&]', declared using local type 'esphome::api::TemplatableKeyValuePai [... cut ...] red using local type 'esphome::api::TemplatableKeyValuePair::TemplatableKeyValuePair(std::__cxx11::string, T) [with T = setup()::<lambda()>; std::__cxx11::string = std::__cxx11::basic_string<char>]::<lambda()>', is used but never defined [-fpermissive]
Compiling .pioenvs/touchpanel32_office/lib5a9/esphome-core/esphome/api/basic_messages.cpp.o
Compiling .pioenvs/touchpanel32_office/lib5a9/esphome-core/esphome/api/command_messages.cpp.o
*** [.pioenvs/touchpanel32_office/src/main.cpp.o] Error 1
```

OttoWinter, 2019-04-15:
> I appears that the "data:" section is not able to be lambda templated, because this doesn't work:

Yes, that's correct. See variables for that.

>     type: char[10]

char arrays do not have `.c_str()` method in C++. Please use `std::string` if you intend to use that method (see google).

> if(id(fan_speed) == 0) {

This is also incorrect. You're comparing a char array against an int, that will not result in the intended behavior (use google again and search for C++ strings)

mmotley999, 2019-04-15:
I changed the code so that fan_speed is an integer to see if that helped, hence the == comparator (see the second post).

The error in compiling was with comparing an int but returning a string from the lambda.

Im an old C programmer from early 90s university days so i struggle with the C++ stuff a bit... was trying to make sure a char pointer wasnt returned vs the string.

OttoWinter, 2019-04-16:
Ah yes true, I missed that.

The ambiguous function call is indeed a problem - both `std::string` and `String` have implicit `const char *` constructors. You can work around it by returning `return std::string("high");` from the function - that will explicitly define the type.

mmotley999, 2019-04-18:
Well, it compiles using the "return std::string" nomenclature!

I'll test it on the real ESP32 (and an ESP8266 since I have both) when I get home.  Also need to make my code a little less hacky.

Thanks for putting up with my C++ ignorance / learning process.

OttoWinter, 2019-04-22:
Fixed in latest version (see dev branch)

closed 2019-04-22 by OttoWinter