esphome/issues #232: platform: adc filters ignored?

bboser, 2019-04-14:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

**Affected component:**

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
sensor:
  - platform: adc
    pin: 39
    name: "Z4 Test"
    attenuation: 11db
    update_interval: 1s
    filters:
      - delta: 0.3
```

**Traceback (if applicable):**

Per log, values are reported every second (update_interval), despite being closer than 0.3V (delta) of the previous value. I would expect only the first value to be reported.
```
[14:20:00][D][sensor.adc:087]: 'Z4 Test': Got voltage=1.37V
[14:20:00][D][sensor.adc:087]: 'Z4 Test': Got voltage=1.41V
[14:20:01][D][sensor.adc:087]: 'Z4 Test': Got voltage=1.30V
[14:20:02][D][sensor.adc:087]: 'Z4 Test': Got voltage=1.42V
```

**Additional information and things you've tried:**

I've also tried adding a `switch` to monitor changes. The switch is never closed (or opened):

```
switch:
  - platform: template
    name: "Z4"
    id: z4

sensor:
  - platform: adc
    pin: 39
    name: "Z4 Test"
    attenuation: 11db
    on_value_range:
      - above: 1.40
        then:
          - switch.turn_on: z4
      - below: 1.32
        then:
          - switch.turn_off: z4
```

OttoWinter, 2019-04-15:
The debug messages you see there are the raw values from the adc sensor. If you make the delta filter trigger, you will see a second message from the sensor core saying that a new value is being "published" - viewing the value in HA will also confirm this.

closed 2019-04-15 by OttoWinter

bboser, 2019-04-15:
Thanks, I understand now.

I’d set `update_interval: 0s`. That froze homeassistant (became
unresponsive). Perhaps it was overwhelmed by these debug messages. I’ll try
reducing (or disabling) logging.

Bernhard