esphome/issues #233: Waveshare 2.13inch black screen

nunofgs, 2019-04-14:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 - https://www.banggood.com/Wemos-T5-V2_0-TTGO-WiFi-Wireless-Module-bluetooth-Base-ESP-32-2_13-ePaper-Display-Development-Board-p-1332909.html?rmmds=myorder&cur_warehouse=CN

**Affected component:**

waveshare_epaper

**Description of problem:**

The display seems to blink a little, then just displays black with white spots. I can post a picture if needed.

Note that if I flash the original firmware (found [here](https://github.com/lewisxhe/TTGO-Electronic-Badge)) it works again, so it's not a hardware issue.

**Problem-relevant YAML-configuration entries:**
```yaml
spi:
  clk_pin: 18
  mosi_pin: 23

display:
  platform: waveshare_epaper
  cs_pin: 5
  dc_pin: 17
  busy_pin: 4
  reset_pin: 16
  model: 2.13in
  lambda: |-
    it.rectangle(10, 10, it.get_width() - 10, it.get_height() - 10);
```

**Traceback (if applicable):**

```
[00:27:31][C][display.waveshare_epaper:162]: Waveshare E-Paper
[00:27:31][C][display.waveshare_epaper:162]:   Rotations: 0
[00:27:31][C][display.waveshare_epaper:162]:   Dimensions: 250px x 128px
[00:27:31][C][display.waveshare_epaper:168]:   Model: 2.13in
[00:27:31][C][display.waveshare_epaper:174]:   Full Update Every: 30
[00:27:31][C][display.waveshare_epaper:175]:   Reset Pin: GPIO16 (Mode: OUTPUT)
[00:27:31][C][display.waveshare_epaper:176]:   DC Pin: GPIO17 (Mode: OUTPUT)
[00:27:31][C][display.waveshare_epaper:177]:   Busy Pin: GPIO4 (Mode: INPUT)
[00:27:31][C][display.waveshare_epaper:178]:   Update Interval: 10000 ms
```

**Additional information and things you've tried:**

Is there any way to debug this further?

OttoWinter, 2019-04-15:
There are several versions of the 2.13inch model (see [note](https://github.com/lewisxhe/TTGO-Electronic-Badge#note)) - could be that the original fw uses a display that is not supported.

nunofgs, 2019-04-17:
Hi @OttoWinter. In my case I flashed the `GxGDE0213B72` version so it seems I have the newest(ish) version of the 2.13inch model.

I'll see if I can play around this weekend and figure this out. Will submit a PR if I do.

OttoWinter, 2019-07-11:
Closing due to inactivity

closed 2019-07-11 by OttoWinter