esphome/issues #236: ESP ble beacon

tamet83, 2019-04-16:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Home Assistant Raspberry Pi 3 B+ addon

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP-WROOM-32

**Affected component:**

https://esphome.io/components/esp32_ble_beacon.html

**Description of problem:**

Hi, I have a problem in configuring the ESP as beacon using your code. Each time i try to upload the configuration the procedure seems to go in loop. I've pasted the log while the process runs.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphome:
  name: ibeacon_test
  platform: ESP32
  board: nodemcu-32s

# Enable logging
logger:
  level: debug

# Enable Home Assistant API
api:

ota:

esp32_ble_beacon:
  type: iBeacon
  uuid: 'dd27ea68-aab6-f700-3926-0b3a99622744'
  major: 10
  minor: 1
```

**Traceback (if applicable):**

```
======================== [SUCCESS] Took 100.73 seconds ========================
INFO Successfully uploaded program.
INFO Starting log output from /dev/ttyUSB1 with baud rate 115200
[08:13:14][I][logger:119]: Log initialized
[08:13:14][C][ota:484]: There have been 7 suspected unsuccessful boot attempts.
[08:13:14][I][application:055]: Running through setup()...
[08:13:14][C][esp32_ble_beacon:036]: Setting up ESP32 BLE beacon...
[08:13:14][C][api:023]: Setting up Home Assistant API server...
[08:13:14]/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/queue.c:1441 (xQueueGenericReceive)- assert failed!
[08:13:14]abort() was called at PC 0x4009043d on core 1
[08:13:14]
[08:13:14]Backtrace: 0x4008dc10:0x3ffc98a0 0x4008de41:0x3ffc98c0 0x4009043d:0x3ffc98e0 0x4017695a:0x3ffc9920 0x40176b5a:0x3ffc9940 0x40169dec:0x3ffc9960 0x40169e55:0x3ffc9980 0x4016c614:0x3ffc99a0 0x4016c700:0x3ffc99d0 0x4017b9a7:0x3ffc99f0 0x400d5ed2:0x3ffc9a40 0x400d2c06:0x3ffc9aa0 0x400d275d:0x3ffc9ac0 0x400d167d:0x3ffc9b10 0x400d885f:0x3ffc9b50 0x40092885:0x3ffc9b70
WARNING Found stack trace! Trying to decode it
INFO Need to fetch platformio IDE-data, please stand by
INFO Running:  platformio run -d /config/esphome/ibeacon_test -t idedata
WARNING Decoded 0x4008dc10: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c:707
WARNING Decoded 0x4008de41: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c:707
WARNING Decoded 0x4009043d: xQueueGenericReceive at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/queue.c:2519
WARNING Decoded 0x4017695a: sys_mutex_lock at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/port/esp32/freertos/sys_arch.c:444
WARNING Decoded 0x40176b5a: sys_arch_protect at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/port/esp32/freertos/sys_arch.c:469
WARNING Decoded 0x40169dec: do_memp_malloc_pool at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/memp.c:459
WARNING Decoded 0x40169e55: memp_malloc at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/memp.c:459
WARNING Decoded 0x4016c614: tcp_alloc at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/tcp.c:1765
WARNING Decoded 0x4016c700: tcp_new_ip_type at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/tcp.c:1782
WARNING Decoded 0x4017b9a7: AsyncServer::begin()
WARNING Decoded 0x400d5ed2: esphome::api::APIServer::setup()
WARNING Decoded 0x400d2c06: esphome::Component::call_setup()
[... 101 lines cut ...]
[08:13:27]/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/queue.c:1441 (xQueueGenericReceive)- assert failed!
[08:13:27]abort() was called at PC 0x4009043d on core 1
[08:13:27]
[08:13:27]Backtrace: 0x4008dc10:0x3ffc98a0 0x4008de41:0x3ffc98c0 0x4009043d:0x3ffc98e0 0x4017695a:0x3ffc9920 0x40176b5a:0x3ffc9940 0x40169dec:0x3ffc9960 0x40169e55:0x3ffc9980 0x40178076:0x3ffc99a0 0x40176dcc:0x3ffc99c0 0x40167e2c:0x3ffc9a00 0x4017cd37:0x3ffc9a20 0x400d40b1:0x3ffc9a60 0x400d2c06:0x3ffc9aa0 0x400d42a0:0x3ffc9ac0 0x400d1642:0x3ffc9b10 0x400d885f:0x3ffc9b50 0x40092885:0x3ffc9b70
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x4008dc10: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c:707
WARNING Decoded 0x4008de41: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c:707
WARNING Decoded 0x4009043d: xQueueGenericReceive at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/queue.c:2519
WARNING Decoded 0x4017695a: sys_mutex_lock at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/port/esp32/freertos/sys_arch.c:444
WARNING Decoded 0x40176b5a: sys_arch_protect at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/port/esp32/freertos/sys_arch.c:469
WARNING Decoded 0x40169dec: do_memp_malloc_pool at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/memp.c:459
WARNING Decoded 0x40169e55: memp_malloc at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/memp.c:459
WARNING Decoded 0x40178076: netconn_alloc at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/api/api_msg.c:742
WARNING Decoded 0x40176dcc: netconn_new_with_proto_and_callback at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/api/api_lib.c:950 (discriminator 4)
WARNING Decoded 0x40167e2c: lwip_socket at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/api/sockets.c:3532
WARNING Decoded 0x4017cd37: WiFiServer::begin(unsigned short)
WARNING Decoded 0x400d40b1: esphome::OTAComponent::setup()
WARNING Decoded 0x400d2c06: esphome::Component::call_setup()
WARNING Decoded 0x400d42a0: esphome::OTAComponent::start_safe_mode(unsigned char, unsigned int)
WARNING Decoded 0x400d1642: setup()
WARNING Decoded 0x400d885f: loopTask(void*)
WARNING Decoded 0x40092885: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c:355 (discriminator 1)
```

**Additional information and things you've tried:**

OttoWinter, 2019-04-16:
You have OTA and API enabled, but those only work when a network is configured. You manually removed WiFi out of the config - thus OTA is failing to set up because WiFi is not on.

When removing WiFi, also remove OTA and API - the two make no sense when there's no network connection after all

closed 2019-04-16 by OttoWinter

tamet83, 2019-04-17:
Hi, sorry to bother you again but I corrected the error in the configuration without any results.

This time I used esphomeflasher compiling the .bin from ESPHome Dashboard.

The yaml code I used is this one:

```
esphome:
  name: beacon
  platform: ESP32
  board: nodemcu-32s

# Enable logging
logger:

esp32_ble_beacon:
  type: iBeacon
  uuid: 'c29ce823-e67a-4e71-bff2-abaa32e77a98'
```
And the log is the following:
```
Using 'COM3' as serial port.
Connecting........_____....._____...
Detecting chip type... ESP32
Connecting......

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32D0WDQ6 (revision 1)
 - Number of Cores: 2
 - Max CPU Frequency: 240MHz
 - Has Bluetooth: YES
 - Has Embedded Flash: NO
 - Has Factory-Calibrated ADC: YES
 - MAC Address: 3C:71:BF:6A:D1:60
Uploading stub...
Running stub...
Stub running...
 - Flash Size: 4MB
 - Flash Mode: dout
 - Flash Frequency: 40MHz
Erasing flash (this may take a while)...
Chip erase completed successfully in 7.1s
Compressed 13696 bytes to 8973...
Wrote 13696 bytes (8973 compressed) at 0x00001000 in 0.8 seconds (effective 135.6 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 143...
Wrote 3072 bytes (143 compressed) at 0x00008000 in 0.0 seconds (effective 630.2 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 2321.8 kbit/s)...
Hash of data verified.
Compressed 1030336 bytes to 574669...
Wrote 1030336 bytes (574669 compressed) at 0x00010000 in 51.5 seconds (effective 159.9 kbit/s)...
Hash of data verified.

Leaving...
Hard Resetting...
Done! Flashing is complete!

Showing logs:
[22:28:56]ets Jun  8 2016 00:22:57
[22:28:56]
[22:28:56]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[22:28:56]configsip: 0, SPIWP:0xee
[22:28:56]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[22:28:56]mode:DOUT, clock div:2
[22:28:56]load:0x3fff0018,len:4
[22:28:56]load:0x3fff001c,len:808
[22:28:56]load:0x40078000,len:6084
[22:28:56]load:0x40080000,len:6704
[22:28:56]entry 0x400802e4
[22:28:57][I][logger:119]: Log initialized
[22:28:57][I][application:055]: Running through setup()...
[22:28:57][C][esp32_ble_beacon:036]: Setting up ESP32 BLE beacon...
[22:28:57][I][application:089]: setup() finished successfully!
[22:28:57][I][application:097]: esphome-core version 1.12.2 compiled on Apr 17 2019, 22:22:58
[22:28:57][C][logger:142]: Logger:
[22:28:57][C][logger:143]:   Level: DEBUG
[22:28:57][C][logger:144]:   Log Baud Rate: 115200
[22:28:57][C][logger:145]:   Hardware UART: UART0
[22:28:57][I][application:109]: Running through first loop()
[22:28:57][I][application:136]: First loop finished successfully!

```

But at the end I can't find the beacon neither using following apps:

- [Beacon test](https://itunes.apple.com/it/app/rl-beacon/id1191040734?mt=8)
- [nRF Connect](https://itunes.apple.com/it/app/nrf-connect/id1054362403?mt=8)

Do you have any further advice?

Many thanks in advance