esphome/issues #243: esp32_ble_beacon is ignored

voltivo, 2019-04-17:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
pip install

**ESP (ESP32/ESP8266, Board/Sonoff):**
  platform: esp32
  board: lolin32

**Affected component:**
https://esphome.io/components/esp32_ble_beacon.html

**Description of problem:**
device yaml has the below config, compile and upload runs without errors.
However after device reboots, no BLE related debug statements are seen and it just looks like that the beacon code is not executed at all.

```
INFO Starting log output from /dev/cu.wchusbserial146240 with baud rate 115200
[01:33:32]�n  8 2016 00:22:57
[01:33:32]
[01:33:32]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[01:33:32]configsip: 0, SPIWP:0xee
[01:33:32]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[01:33:32]mode:DOUT, clock div:2
[01:33:32]load:0x3fff0018,len:4
[01:33:32]load:0x3fff001c,len:928
[01:33:32]ho 0 tail 12 room 4
[01:33:32]load:0x40078000,len:9280
[01:33:32]load:0x40080400,len:5860
[01:33:32]entry 0x40080698
[01:33:33][I][logger:119]: Log initialized
[01:33:33][C][ota:484]: There have been 1 suspected unsuccessful boot attempts.
[01:33:33][I][application:055]: Running through setup()...
[01:33:33][D][binary_sensor:037]: 'BTENV Status': Sending state OFF
[01:33:33][C][esp32_ble_beacon:036]: Setting up ESP32 BLE beacon...
[01:33:33][C][sensor.dht:029]: Setting up DHT...
[01:33:33][C][wifi:027]: Setting up WiFi...
[01:33:33][I][wifi:167]: WiFi Connecting to 'EXTREME'...
[01:33:33][D][sensor.dht:066]: Got Temperature=27.1°C Humidity=56.7%
[01:33:34][I][wifi:379]: WiFi connected!
[01:33:34][C][wifi:253]:   SSID: 'EXTREME'
[01:33:34][C][wifi:254]:   IP Address: 10.1.1.242
[01:33:34][C][wifi:256]:   BSSID: 12:80:63:73:90:14
[01:33:34][C][wifi:257]:   Hostname: 'btenv1'
[01:33:34][C][wifi:261]:   Signal strength: -11 dB ▂▄▆█
[01:33:34][C][wifi:262]:   Channel: 7
[01:33:34][C][wifi:263]:   Subnet: 255.255.255.0
[01:33:34][C][wifi:264]:   Gateway: 10.1.1.1
[01:33:34][C][wifi:265]:   DNS1: 10.1.1.1
[01:33:34][C][wifi:266]:   DNS2: 1.0.0.1
[01:33:34][C][api:023]: Setting up Home Assistant API server...
[01:33:34][C][ota:127]: Over-The-Air Updates:
[01:33:34][C][ota:128]:   Address: 10.1.1.242:3232
[01:33:34][C][mqtt.client:027]: Setting up MQTT...
[01:33:34][I][mqtt.client:173]: Connecting to MQTT...
[01:33:34][I][mqtt.client:213]: MQTT Connected!
[01:33:34][I][application:089]: setup() finished successfully!
[01:33:34][I][application:097]: esphome-core version 1.12.2 compiled on Apr 18 2019, 01:31:29
[01:33:34][C][wifi:371]: WiFi:
[01:33:34][C][wifi:253]:   SSID: 'XXXXX'
[01:33:34][C][wifi:254]:   IP Address: 10.1.1.xxx
[01:33:34][C][wifi:256]:   BSSID: 12:80:63:73:xx:xx
[01:33:34][C][wifi:257]:   Hostname: 'btenv1'
[01:33:34][C][wifi:261]:   Signal strength: -8 dB ▂▄▆█
[01:33:34][C][wifi:262]:   Channel: 7
[01:33:34][C][wifi:263]:   Subnet: 255.255.255.0
[01:33:34][C][wifi:264]:   Gateway: 10.1.1.x
[01:33:34][C][wifi:265]:   DNS1: 10.1.1.x
[01:33:34][C][wifi:266]:   DNS2: 1.0.0.x
[01:33:34][C][binary_sensor.status:035]: Status Binary Sensor 'BTENV Status'
[01:33:34][C][binary_sensor.status:035]:   Device Class: 'connectivity'
[01:33:34][C][logger:142]: Logger:
[01:33:34][C][logger:143]:   Level: DEBUG
[01:33:34][C][logger:144]:   Log Baud Rate: 115200
[01:33:34][C][logger:145]:   Hardware UART: UART0
[01:33:34][C][sensor.dht:035]: DHT:
[01:33:34][C][sensor.dht:036]:   Pin: GPIO13 (Mode: OUTPUT)
[01:33:34][C][sensor.dht:042]:   Model: DHT22 (or equivalent)
[01:33:34][C][sensor.dht:045]:   Update Interval: 10000 ms
[01:33:34][C][sensor.dht:047]:   Temperature 'Living Room Temperature'
[01:33:34][C][sensor.dht:047]:     Unit of Measurement: '°C'
[01:33:34][C][sensor.dht:047]:     Accuracy Decimals: 1
[01:33:34][C][sensor.dht:048]:   Humidity 'Living Room Humidity'
[01:33:34][C][sensor.dht:048]:     Unit of Measurement: '%'
[01:33:34][C][sensor.dht:048]:     Accuracy Decimals: 0
[01:33:34][C][sensor.dht:048]:     Icon: 'mdi:water-percent'
[01:33:34][C][api:101]: API Server:
[01:33:34][C][api:102]:   Address: 10.1.1.xxx:6053
[01:33:34][C][ota:127]: Over-The-Air Updates:
[01:33:34][C][ota:128]:   Address: 10.1.1.xxx:3232
[01:33:34][C][mqtt.client:062]: MQTT:
[01:33:34][C][mqtt.client:064]:   Server Address: 10.1.1.xxx:1883 (10.1.1.21)
[01:33:34][C][mqtt.client:065]:   Username: 'homeassistant'
[01:33:34][C][mqtt.client:066]:   Client ID: 'btenv1-30aea44d0xxx'
[01:33:34][C][mqtt.client:071]:   Topic Prefix: 'btenv1'
[01:33:34][C][mqtt.client:073]:   Log Topic: 'btenv1/debug'
[01:33:34][C][mqtt.client:076]:   Availability: 'btenv1/status'
[01:33:34][C][binary_sensor.mqtt:021]: MQTT Binary Sensor 'BTENV Status':
[01:33:34][C][binary_sensor.mqtt:022]:   State Topic: 'btenv1/status'
[01:33:34][C][sensor.mqtt:025]: MQTT Sensor 'Living Room Temperature':
[01:33:34][C][sensor.mqtt:027]:   Expire After: 50s
[01:33:34][C][sensor.mqtt:029]:   State Topic: 'btenv1/sensor/living_room_temperature/state'
[01:33:34][C][sensor.mqtt:025]: MQTT Sensor 'Living Room Humidity':
[01:33:34][C][sensor.mqtt:027]:   Expire After: 50s
[01:33:34][C][sensor.mqtt:029]:   State Topic: 'btenv1/sensor/living_room_humidity/state'
[01:33:34][I][application:109]: Running through first loop()
[01:33:34][D][binary_sensor:037]: 'BTENV Status': Sending state ON
[01:33:34][I][application:136]: First loop finished successfully!
[01:33:36][D][api:546]: Client 'Home Assistant 0.91.4 (10.1.1.xxx)' connected successfully!
[01:33:40][D][sensor.dht:066]: Got Temperature=27.1°C Humidity=56.3%
[01:33:50][D][sensor.dht:066]: Got Temperature=27.0°C Humidity=56.6%
...
...
```

**Problem-relevant YAML-configuration entries:**
```yaml
esp32_ble_beacon:
  type: iBeacon
  uuid: '12a13a53-905b-0654-3468-3956014a7888'
  major: 60042
  minor: 43814
```

**Traceback (if applicable):**

```
N/A
```

**Additional information and things you've tried:**

- use different boards
- remove all other sensors, components from YAML

tamet83, 2019-04-17:
I have the same problem here.
I did a mistake at the first, leaving the api and ota in the code, I deleted all wifi related component as suggested from @OttoWinter.

So i tried again, as you can see in issue #[236](https://github.com/esphome/issues/issues/236), but I still cant detect the beacon from any kind of app available in the iOS store.

voltivo, 2019-04-18:
I got it to work. No changes needed on the other config.

The config does not seem to value the major and minor setting, which was why my BLE scanner did not find the beacon.

With above config the Major was 234 instead of 60042, and the Minor was 138 instead of 43814.
I adjusted the the "receiver" of the beacon to the real observed values.

closed 2019-04-18 by voltivo