esphome/issues #245: ESP32-CAM Stream Access ?

ycardon, 2019-04-18:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

N/A

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32-CAM

**Affected component:**

https://esphome.io/components/esp32_camera.html

**Description of problem:**

I'd like to capture a camera stream without going through the HA camera proxy stream ([http://<my_ha_server>/api/camera_proxy_stream/camera.<my_esphome_device_name>?token=<my_ha_token_for_the_device>]()).

Is there a URL I can point to directly on my ESP32 Camera device ?

**Problem-relevant YAML-configuration entries:**
```yaml
N/A
```

**Traceback (if applicable):**

```
N/A
```

**Additional information and things you've tried:**

same as https://github.com/esphome/issues/issues/237

OttoWinter, 2019-04-22:
There is not, and it's not really a goal of the ESP32 camera platform. There are two reasons for this:
- With native API, the ESP can know when a client requests a full stream, a single image or nothing at all - helps with overheating issues.
- This requires more access to the internals of the HTTP web server, that is not possible with the library used for the `web_server` component.

ycardon, 2019-04-22:
Thanks for the insight !

closed 2019-04-22 by ycardon