esphome/issues #246: TTGO ESP32 Cam has incorrect still image update rate after streaming

Svurk, 2019-04-19:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**
TTGO T-Camera ESP32 WROVER OV2640 Camera Module 0.96 OLED

**Affected component:**
ESP32 Camera Component

**Description of problem:**
ESP32 cam installs correctly in ESPHome. Integrates with Home Assistant and shows up as camera in HA frontend. In the ESPHome serial log the camera initially updates a still image every 10 seconds. When stream starts this works fine in HA too. After stream is stopped the ESP32 cam does not revert back to updating once per 10s, but spills out images at approximately 5-10 images per sec according to serial log (which fills up quickly). HA still updates image approximately once per 10s in frontend. After that connection with HA is lost after a few minutes (HA reports: image not available). ESP32 must be restarted as well as ESPHome to get back function.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: ttgo_cam_one
  platform: ESP32
  board: esp-wrover-kit

wifi:
  manual_ip:
    static_ip: 192.168.21.8
    gateway: 192.168.21.1
    subnet: 255.255.255.0
  ssid: '---'
  password: '---'  

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: '---'

ota:
  password: '---'
  
esp32_camera:
  name: min_kamera
  external_clock:
    pin: GPIO32
    frequency: 20MHz
  i2c_pins:
    sda: GPIO13
    scl: GPIO12
  data_pins: [GPIO5, GPIO14, GPIO4, GPIO15, GPIO18, GPIO23, GPIO36, GPIO39]
  vsync_pin: GPIO27
  href_pin: GPIO25
  pixel_clock_pin: GPIO19
  power_down_pin: GPIO26
  resolution: 320x240
  max_framerate: 10 fps
  idle_framerate: 0.1 fps
```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**
Uploading firmware by cable or OTA does not matter. Tried fresh uploads to ESP32 cam several times by cable and OTA.

OttoWinter, 2019-04-22:
Yeah that is a bug in HA (I think). While developing the integration I also saw that HA would sometimes leave mjpeg connections open despite the stream being closed. Using chrome as the browser fixed the issue (usually I use firefox).

If you enable debug logs in HA, you should see the stream being closed after a while. If that's not the case, then it is indeed an issue in HA.

Svurk, 2019-04-22:
Ok, I see. ESPHome is a truly awesome addon in HA. But, I still cannot get this particular ESP cam to function with HA via ESPHome in a stable way. Changing browser did not help. Perhaps the stream continuing after it should have stopped will be addressed in future updates in HA. 
Turned out the best way to get a stable camera function in my setting with this board was to use this firmware:
[https://github.com/Xinyuan-LilyGo/esp32-camera-bme280](url) 
and set it up as a Generic MJPEG IP Camera in HA:
```
camera:
  - platform: mjpeg
    mjpeg_url: http://192.168.21.32:81/stream
    still_image_url: http://192.168.21.32/capture?    
    name: my_camera
    verify_ssl: false
```
Stable and works fine for now. Although, this makes accessing the PIR and the OLED on the board more difficult. Needless to say, looking forward to future bug fix so I can integrate with ESPHome and fully exploit all the board components.

OttoWinter, 2019-07-11:
Closing due to inactivity

closed 2019-07-11 by OttoWinter