esphome/issues #249: v1.13 CustomComponents lambda changed captures

glmnet, 2019-04-20:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
v1.13

**ESP (ESP32/ESP8266, Board/Sonoff):**

**Affected component:**
custom components, e.g. CustomBinarySensor

**Description of problem:**
custom binary sensor lambdas returns an instance (or an array of instances) this lambda used to capture all context in main `[=]` in v1.13 it is not capturing that, it is `[]` instead.

this is a breaking change, is there a good reason for this?
changing the code to this fixes the problem:
```python
def to_code(config):
    template_ = yield cg.process_lambda(
        config[CONF_LAMBDA], [], '=',
        return_type=cg.std_vector.template(binary_sensor.BinarySensorPtr))
```

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

OttoWinter, 2019-04-22:
The reason was that now that all variables are stored globally, there should be no need for any lambda captures anymore. And lambda capture errors can often be difficult to track down (see also mutable lambda), plus there really shouldn't be any need to modify your `main.cpp` anymore - all that should be done with separate `includes`.

Can you show the code with which you see compile errors? If you have modified your setup() method then that's probably it.

glmnet, 2019-04-22:
No `main.cpp` hand modified.
The problem arises when you want to use a custom component from another custom component. 
More precisely using a custom component from an e.g. binary custom component
My custom component will be an interface to custom i2c device.
This custom component, created by a lambda, return a CustomComponentConstructor, this is not generated as a global, but as stack which is not captured by the binary custom component lambda, may be another solution would be if it was generated as a global variable too.

See generated code below: `ape` is no global:
(`ape_binary_output` is a macro which expands to something returning a binary output)

````yaml
custom_component:
  - id: ape
    lambda: |-
      auto ape_component = new ArduinoPortExtender(i2ccomponent, 0x08);
      return {ape_component};

binary_sensor:
  - platform: custom
    lambda: |-
      return {ape_binary_sensor(ape, 2), // ribbon purple
              ape_binary_sensor(ape, 3), // ribbon white
              ape_binary_sensor(ape, 8), // ribbon verde
              ape_binary_sensor(ape, 15), // A1 ribbon orange
              ape_binary_sensor(ape, 13), // A1 ribbon orange
              ape_binary_sensor(ape, 14)}; // ribbon red
````

````cpp
CustomComponentConstructor ape = CustomComponentConstructor([=]() -> std::vector<Component *> {
      auto ape_component = new ArduinoPortExtender(i2ccomponent, 0x08);
      return {ape_component};
  });
  output::CustomBinaryOutputConstructor output_custombinaryoutputconstructor = output::CustomBinaryOutputConstructor([=]() -> std::vector<output::BinaryOutput *> {
      return {ape_binary_output(ape, 7),
              ape_binary_output(ape, 6),
              ape_binary_output(ape, 5),
              ape_binary_output(ape, 4)};
  });
````

OttoWinter, 2019-04-22:
Aha ok, yeah non-pointer variables are not global. I guess in that case you're right that we can make `=` the default again.

OttoWinter, 2019-04-25:
Fixed in dev/reverted to old method.

closed 2019-04-25 by OttoWinter