esphome/issues #250: Problems with Partition Platform and NeopixelBus

pinxaolives, 2019-04-21:
Trying the NeopixelBus platform with the platform: partition, I'm having problems during the 'Compile and Updload' process.

The process suddently stops.

# Yaml File
esphome:
  name: laundry_shelf1_rack1
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: “xxx”
  password: “xxx”
  manual_ip:
    static_ip: 192.168.1.200
    gateway: 192.168.1.1
    subnet: 255.255.255.0

logger:
  level: DEBUG

api:
  password: ‘xxx’
ota:
  password: ‘xxx’

# Configuration entry
light:
  - platform: partition
    name: "Partition Light"
    segments:
      - id: light1
        from: 0
        to: 0
      - id: light2
        from: 1
        to: 10 
  - platform: neopixelbus
    id: light1
    num_leds: 20
    pin: GPIO3
    name: "NeoPixel Light1"             
    
  - platform: neopixelbus
    id: light2
    num_leds: 20
    pin: GPIO3
    name: "NeoPixel Light2"

# Compile and Upload Log
...
INFO Successfully uploaded program.
INFO Starting log output from /dev/ttyUSB0 with baud rate 115200
[19:22:04]�c��oo�doo���cp��dslslp�o��l��co�|l��c��oo�l��l`�nnd`ns���nc�ls��nc�l�ccc���d`��o�[I][logger:119]: Log initialized
[19:22:04][C][ota:484]: There have been 1 suspected unsuccessful boot attempts.
[19:22:04][I][application:055]: Running through setup()...
[19:22:04][C][light.state:138]: Setting up light 'NeoPixel Light1'...
[19:22:04][D][light.state:418]: 'NeoPixel Light1' Turning OFF
[19:22:04][D][light.state:425]:   Brightness: 100%
[19:22:04][D][light.state:449]:   Red=100%, Green=100%, Blue=100%
[19:22:04][D][light.state:459]:   Transition Length: 1000 ms
[19:22:04][C][light.state:138]: Setting up light 'NeoPixel Light2'...
[19:22:04][D][light.state:418]: 'NeoPixel Light2' Turning OFF
[19:22:04][D][light.state:425]:   Brightness: 100%
[19:22:04][D][light.state:449]:   Red=100%, Green=100%, Blue=100%
[19:22:04][D][light.state:459]:   Transition Length: 1000 ms
[19:22:04][C][light.state:138]: Setting up light 'Partition Light'...
[19:22:04][D][light.state:418]: 'Partition Light' Turning OFF
[19:22:04][D][light.state:425]:   Brightness: 100%
[19:22:04][D][light.state:449]:   Red=100%, Green=100%, Blue=100%
[19:22:04][D][light.state:459]:   Transition Length: 1000 ms
[19:22:04][C][wifi:027]: Setting up WiFi...
[19:22:04][D][wifi:271]: Starting scan...
--------------------------------------------------
The 'Compile and Upload' freeze here.

OttoWinter, 2019-04-22:
GPIO3 is the RX pin of the ESP8266, and also deals with the USB logging connection.

It could well be that the neopixelbus library stops the serial interface when enabled - but that would then just be a limitation of the platform (and an upstream issue).

Can you try accessing the remote logs?

OttoWinter, 2019-07-11:
Closing due to inactivity

closed 2019-07-11 by OttoWinter