esphome/issues #251: Unable to get consistent Remote Receiver data

tr1plus, 2019-04-21:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

DOCKER: esphome/esphome-armhf:latest 

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266/NodeMCU
**Affected component:**

Remote Receiver

**Description of problem:**
Using the simple code below, I am unable to get any consistent data from the receiver. I have tried 4 remotes from different brands (where the Philips one should be most reliable from the lot). It is not providing me with any protocol (also used all protocols separately with and without the "inverted" flag.

The log below is me holding the same button for a few seconds

**Problem-relevant YAML-configuration entries:**
```yaml
remote_receiver:
  pin:
    number: D1
    inverted: true
    mode: INPUT_PULLUP
  dump: all
```

**Traceback (if applicable):**

```
[17:38:18][D][remote.raw:067]: Received Raw: 2648, -888, 441, -885, 440, -443, 614, -270, 1322, -1327, 440, -442, 442, -442, 441, -443, 526, -357, 442, -442, 441, -443, 441, -442, 441, -442, 883, -886, 521, -361, 442, -443, 441, -442, 441, -443, 527, -355, 574, -10000
[17:38:18][D][remote.raw:067]: Received Raw: 2650, -889, 692, -632, 634, -250, 573, -309, 1404, -1246, 441, -442, 681, -205, 686, -196, 441, -442, 529, -354, 636, -249, 573, -309, 602, -283, 1127, -641, 606, -277, 441, -442, 636, -249, 688, -195, 440, -448, 436, -10000
[17:38:19][D][remote.raw:067]: Received Raw: 2817, -719, 554, -772, 440, -442, 441, -443, 556, -760, 967, -365, 441, -443, 443, -442, 462, -420, 441, -443, 440, -443, 441, -443, 441, -443, 440, -444, 881, -885, 441, -447, 437, -444, 581, -301, 553, -331, 526, -358, 584, -10000
[17:38:19][D][remote.raw:067]: Received Raw: 2734, -802, 444, -881, 440, -443, 500, -384, 554, -763, 1025, -306, 441, -447, 551, -329, 589, -295, 440, -443, 555, -328, 648, -237, 584, -299, 440, -443, 1054, -713, 584, -301, 440, -443, 585, -299, 583, -301, 609, -274, 615, -10000
[17:38:20][D][remote.raw:067]: Received Raw: 2823, -714, 548, -776, 440, -442, 693, -193, 1633, -1015, 441, -442, 441, -442, 442, -442, 635, -249, 441, -441, 552, -337, 437, -442, 441, -441, 1103, -667, 719, -140, 551, -358, 696, -215, 603, -252, 441, -442, 669, -10000
[17:38:20][D][remote.raw:067]: Received Raw: 2649, -888, 725, -601, 440, -441, 698, -215, 1494, -1127, 441, -443, 441, -442, 441, -443, 441, -442, 533, -355, 603, -277, 603, -280, 442, -442, 1137, -631, 717, -142, 656, -252, 441, -442, 441, -443, 692, -193, 439, -10000
[17:38:20][D][remote.raw:067]: Received Raw: 2648, -890, 439, -884, 605, -278, 442, -443, 440, -876, 882, -450, 687, -197, 441, -442, 441, -443, 441, -442, 441, -443, 441, -443, 441, -446, 437, -442, 883, -885, 441, -442, 442, -442, 441, -443, 441, -443, 574, -308, 604, -10000
[17:38:20][D][remote.raw:067]: Received Raw: 2648, -888, 441, -885, 439, -442, 442, -443, 440, -877, 882, -477, 413, -444, 440, -442, 441, -443, 441, -442, 603, -280, 442, -441, 443, -442, 441, -443, 882, -885, 445, -438, 442, -443, 441, -442, 441, -443, 441, -442, 441, -10000
[17:38:21][D][remote.raw:067]: Received Raw: 2648, -888, 442, -883, 440, -443, 442, -441, 1324, -1327, 440, -443, 441, -442, 658, -226, 441, -443, 441, -442, 441, -443, 441, -444, 439, -443, 882, -885, 441, -442, 656, -227, 442, -443, 441, -443, 440, -445, 439, -10000
[17:38:21][D][remote.raw:067]: Received Raw: 2649, -888, 441, -883, 441, -443, 572, -311, 1324, -1325, 442, -442, 441, -442, 526, -358, 441, -443, 441, -443, 441, -442, 605, -279, 632, -252, 1100, -667, 441, -443, 440, -443, 444, -438, 527, -358, 441, -442, 580, -10000
[17:38:22][D][remote.raw:067]: Received Raw: 2648, -888, 440, -884, 441, -443, 441, -442, 441, -877, 881, -451, 440, -443, 607, -277, 528, -354, 605, -279, 528, -356, 441, -443, 444, -439, 529, -355, 1074, -693, 633, -252, 607, -276, 604, -279, 443, -440, 528, -356, 633, -10000
[17:38:22][D][remote.raw:067]: Received Raw: 2650, -889, 663, -661, 440, -443, 441, -443, 441, -877, 881, -449, 441, -444, 571, -313, 634, -249, 440, -443, 441, -442, 441, -442, 529, -355, 441, -443, 1053, -715, 1488, -279, 441, -442, 442, -442, 441, -444, 440, -10000
[17:38:22][D][remote.raw:067]: Received Raw: 2824, -713, 616, -709, 646, -240, 610, -271, 1505, -1144, 645, -241, 614, -268, 671, -218, 434, -443, 616, -270, 628, -254, 528, -355, 584, -300, 940, -828, 644, -267, 412, -443, 588, -297, 440, -442, 641, -245, 615, -10000
[17:38:22][D][remote.raw:067]: Received Raw: 278, -559, 2648, -889, 440, -884, 571, -312, 441, -442, 1325, -1326, 440, -443, 441, -446, 571, -309, 441, -443, 663, -220, 578, -305, 612, -273, 441, -442, 882, -886, 440, -442, 442, -442, 442, -442, 441, -443, 441, -443, 440, -10000
[17:38:23][D][remote.raw:067]: Received Raw: 2648, -888, 441, -884, 441, -442, 441, -443, 633, -685, 881, -450, 441, -443, 440, -444, 440, -442, 615, -270, 440, -443, 606, -278, 440, -443, 441, -443, 882, -884, 442, -443, 441, -442, 664, -222, 568, -312, 661, -225, 440, -10000
[17:38:23][D][remote.raw:067]: Received Raw: 2845, -691, 659, -666, 572, -310, 442, -443, 441, -876, 885, -447, 574, -309, 441, -443, 662, -223, 439, -443, 441, -443, 440, -443, 442, -441, 441, -444, 882, -885, 440, -158, 889, -279, 442, -442, 609, -275, 685, -199, 665, -10000
[17:38:23][D][remote.raw:067]: Received Raw: 2648, -888, 604, -722, 440, -442, 441, -442, 1324, -1326, 441, -442, 442, -442, 441, -443, 574, -308, 638, -248, 630, -253, 660, -223, 528, -356, 882, -885, 603, -280, 658, -226, 571, -312, 633, -251, 574, -310, 632, -10000
[17:38:23][D][remote.raw:067]: Received Raw: 2734, -801, 663, -661, 604, -278, 443, -442, 1323, -1326, 689, -196, 440, -443, 441, -442, 441, -443, 637, -247, 440, -444, 605, -278, 637, -246, 882, -885, 441, -443, 441, -441, 443, -442, 441, -158, 726, -442, 520, -10000
[17:38:24][D][remote.raw:067]: Received Raw: 2648, -888, 607, -718, 700, -185, 635, -248, 439, -877, 1994, -223, 1575, -191, 577, -307, 694, -190, 439, -442, 442, -442, 442, -442, 1099, -669, 440, -444, 577, -306, 441, -442, 698, -213, 604, -252, 636, -10000
[17:38:24][D][remote.raw:067]: Received Raw: 2874, -662, 555, -770, 1456, -311, 725, -593, 881, -452, 662, -220, 674, -210, 692, -191, 525, -360, 440, -441, 441, -442, 442, -442, 441, -443, 1049, -718, 551, -333, 444, -441, 633, -249, 609, -275, 607, -278, 778, -10000
[17:38:24][D][remote.raw:067]: Received Raw: 225, -10000
[17:38:24][D][remote.raw:067]: Received Raw: 336, -6601, 226, -10000
[17:38:24][D][remote.raw:067]: Received Raw: 224, -10000
[17:38:24][D][remote.raw:067]: Received Raw: 2649, -889, 525, -799, 441, -442, 657, -225, 1325, -1327, 440, -442, 442, -443, 526, -356, 443, -442, 440, -443, 441, -443, 440, -444, 440, -445, 1042, -724, 529, -354, 441, -442, 441, -443, 441, -445, 438, -444, 440, -10000
[17:38:25][D][remote.raw:067]: Received Raw: 2649, -888, 613, -712, 580, -304, 440, -443, 1492, -1158, 440, -442, 608, -276, 441, -443, 441, -441, 528, -357, 441, -442, 441, -443, 612, -272, 882, -885, 441, -443, 441, -442, 441, -443, 578, -305, 529, -355, 527, -10000
```

**Additional information and things you've tried:**
All protocols separately (lg, nec, samsung, ...)
Different remotes
with and without the inverted flag

HFsi, 2019-04-21:
I'm with the same issue, very difficult to get something that could make any sense.
What kind of receiver are you using?

tr1plus, 2019-04-21:
I am using "KY-022 Infrared IR Sensor Receiver Module" 
When using this with some "normal" Arduino code (one of the examples that comes with the IR arduino library) everything goes okay but... not via here it seems...

HFsi, 2019-04-22:
The raws seems pretty consistent...your issue is that the specific protocols are not dumped?

tr1plus, 2019-04-22:
I haven't found one line that is exactly the same as a before one.
Problem is that this way I am unable to determine if a certain button has been pressed... Unless I am misunderstanding.

But this is what I tried to do. I held the button for half a minute or so and it never got triggered.

```
binary_sensor:
  - platform: remote_receiver
    name: "Remote Plus Sign"
    raw: [2648, -888, 442, -883, 440, -443, 442, -441, 1324, -1327, 440, -443, 441, -442, 658, -226, 441, -443, 441, -442, 441, -443, 441, -444, 439, -443, 882, -885, 441, -442, 656, -227, 442, -443, 441, -443, 440, -445, 439, -10000]
```

OttoWinter, 2019-04-22:
> I haven't found one line that is exactly the same as a before one.

That is normal, most remotes have pretty crappy chips with unreliable timing. However, all receivers accept a wide tolerance on these inputs. If you connect another ESP8266 with a remote_transmitter with a raw code you will see that the remote_receiver has almost perfect timing (I've only seen 4µs difference at most)

In this case your button press is definitely sending different signals though (see 9th column), your remote could be incrementing an integer internally to differentiate signals.

In the raw remote reciver, try removing the -10000 at the end.

tr1plus, 2019-04-22:
But in that case, how can I make the binary sensor work reliably? Won't it need the exact same "raw" data always?

HFsi, 2019-04-22:
I believe there's some tolerance in the codes....did you get anything to work? I don´t

tr1plus, 2019-04-22:
Nope, unable to get my binary sensor to trigger.

I also removed the "-10000" at the end of the raw string but no luck so far.

pkozul, 2019-05-04:
I've had good success sniffing RF codes using pilight-debug on a Raspberry Pi.

```
sudo pilight-debug
```

I was hoping my ESP8266 (with ESPHome) would be able to sniff the same codes. It came pretty close, but I kept on seeing a lot of other 'noise', and it wasn't consistent.

What I was really interesting in was getting ESPHome to transmit the codes (i.e. simulate my fireplace remote), but according to the codes I was seeing in pilight-debug (on the Raspberry Pi), they were all over the place.

I did find a solution in the end, using OpenMQTTGateway. I installed that on my ESP8266 and was able to mimic the exact same codes that my 433MHz fireplace remote was sending. Looking at the output from pilight-debug (on the Raspberry Pi), the codes being sent from OpenMQTTGateway were identical to the ones from my remote.

Having said that, it would be good to get this working in ESPHome as well. I believe OpenMQTTGateway includes an enhanced version of the rc-switch library, that contains a lot more protocols.

https://github.com/1technophile/OpenMQTTGateway/tree/development/lib/rc-switch

Maybe ESPHome could use version of the rc-switch library as well?

tr1plus, 2019-05-04:
It's good to see that I'm not the only one getting bad consistency with this!
I may also go for openmqttgateway if by the time I receive my esp I don't have another solution :)
Thanks!

HFsi, 2019-05-04:
For sure you are not the only one! Altough I'm struggling with RF instead of IR, I'm in the same situation.
All the guns point to OMG, as I'm seeing in different threads, this will be my next attempt.

tr1plus, 2019-05-30:
Okay, so almost a month into the future:

I tested things using **openMQTTGateway** and **IRremoteESP8266: IRrecvDumpV2** (on which OMG is also based)

Long story short; it seems they are both able to translate my remote signals into a usable hex code. OMG dev branch example(the hex conversion is not in a live version yet):
`{"value":551494365,"protocol":3,"bits":32,"hex":"20DF22DD","raw":"9002,4476,580,536,606,510,554,1680,554,564,578,538,580,536,580,536,582,536,580,1654,554,1680,580,538,606,1628,580,1654,580,1654,606,1628,580,1654,580,538,580,538,580,1654,578,538,580,538,578,538,578,1658,578,538,580,1654,580,1654,578,538,580,1654,608,1628,580,1654,552,564,554,1680,554"}`

I had a quick look into this code, but I'm not sure what "behind the scenes" IR library ESPHome is using, don't see anything resembling IRremoteESP8266. (and there is probably a reason for that :))

Would it be possible to try adding a translation from raw to hex? @OttoWinter My feeling is that this will be a final push to make things work better.

While playing around with the IRremoteESP8266 library I also played around with the following, which would be nice to make configurable via ESPHome:
`
// Some A/C units have gaps in their protocols of ~40ms. e.g. Kelvinator
// A value this large may swallow repeats of some protocols
const uint8_t kTimeout = 50;
`

Hope this gives some insight. In the meantime I'll try to delve into the code a little more to see if I can find something else :)

Thanks!

OttoWinter, 2019-07-11:
Closing due to inactivity - and I think the issue has been somewhat fixed in 1.13

closed 2019-07-11 by OttoWinter

tr1plus, 2019-07-11:
I will try the new release as soon as I can. Need to verify if the arm docker has also been updated.

Keep you posted