esphome/issues #252: Unable to get initial flash of ESP32-CAM working

nhorvath, 2019-04-21:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hassio addion on x86_64 virtual machine, downloaded binary and trying to flash with esptool on windows. Also tried NodeMCU-pyflasher-3.0

**ESP (ESP32/ESP8266, Board/Sonoff):**

AiThinker clone ESP32-CAM
This one specifically: https://www.aliexpress.com/item/ESP32-CAM-WiFi-Module-Module-ESP32-with-OV2640/33001043678.html?spm=a2g0s.9042311.0.0.27424c4d18eQdP

**Affected component:**

esp32cam

**Description of problem:**
Flashing says it's successful but after resetting the device the led just flashes and never connects to wifi.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: esp32_cam
  platform: ESP32
  board: esp-wrover-kit

wifi:
  ssid: "horvath"
  password: "redacted"
  domain: '.horvath'

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: 'redacted'

ota:
  password: 'redacted'

esp32_camera:
  external_clock:
    pin: GPIO0
    frequency: 20MHz
  i2c_pins:
    sda: GPIO26
    scl: GPIO27
  data_pins: [GPIO5, GPIO18, GPIO19, GPIO21, GPIO36, GPIO39, GPIO34, GPIO35]
  vsync_pin: GPIO25
  href_pin: GPIO23
  pixel_clock_pin: GPIO22
  power_down_pin: GPIO32

  # Image settings
  name: ESP32-CAM
  #resolution: 1600x1200
```

**Traceback (if applicable):**

```
n/a
```

**Additional information and things you've tried:**
```esptool.py.exe --port COM10 write_flash --flash_mode dio 0x0 f:\Downloads\esp32_cam.bin```
I tried a few different flashers, tried erasing the flash. Double checked my wifi config. Tried a different power supply. Nothing seems to get it to connect.

Something else I noticed is I tried to use board: esp32cam as would be indicated here, but it said it was invalid: http://docs.platformio.org/en/latest/boards/espressif32/esp32cam.html#board-espressif32-esp32cam
When that didn't work I just used the one for generic ESP32.

nhorvath, 2019-04-21:
I installed esphome on my linux laptop and copied the yaml file from hassio and did esphome run on it and it started to work for about 10 seconds. then I got a bunch of restart loops and brownout detect lines. So I unplugged it from my usb flasher and connected it to a dedicated 5V supply. I can ping it but I can't pull images or connect with esphome logs. Any ideas?

This was the original logs from USB serial:
```
========================================================================================================================== [SUCCESS] Took 53.10 seconds ==========================================================================================================================
INFO Successfully uploaded program.
INFO Starting log output from /dev/ttyUSB0 with baud rate 115200
[19:22:42]ets Jun  8 2016 00:22:57
[19:22:42]
[19:22:42]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[19:22:42]configsip: 0, SPIWP:0xee
[19:22:42]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[19:22:42]mode:DOUT, clock div:2
[19:22:42]load:0x3fff0018,len:4
[19:22:42]load:0x3fff001c,len:928
[19:22:42]ho 0 tail 12 room 4
[19:22:42]load:0x40078000,len:9280
[19:22:42]load:0x40080400,len:5860
[19:22:42]entry 0x40080698
[19:22:43]�[I][logger:119]: Log initialized
[19:22:43][E][Preferences.cpp:457] getBytes(): nvs_get_blob len fail: 0 NOT_FOUND
[19:22:43][C][ota:484]: There have been 0 suspected unsuccessful boot attempts.
[19:22:43][I][application:055]: Running through setup()...
[19:22:44]SCCB_Write [22]=1a failed
[19:22:44][E][camera.c:1054] camera_init(): Failed to set frame size
[19:22:44][E][camera.c:1104] esp_camera_init(): Camera init failed with error 0x20002
[19:22:44][E][esp32_camera:018]: esp_camera_init failed: ERROR
[19:22:44][E][component:151]: Component was marked as failed.
[19:22:44][C][wifi:027]: Setting up WiFi...
[19:22:44][D][wifi:271]: Starting scan...
[19:22:46][D][wifi:286]: Found networks:
[19:22:46][I][wifi:321]: - 'horvath' (B4:FB:E4:CB:4A:9F) ▂▄▆█
[19:22:46][D][wifi:322]:     Channel: 3
[19:22:46][D][wifi:323]:     RSSI: -73 dB
[19:22:46][D][wifi:325]: - '' (B6:FB:E4:9B:4A:9F) ▂▄▆█
[19:22:46][I][wifi:167]: WiFi Connecting to 'horvath'...
[19:22:47]
[19:22:47]Brownout detector was triggered
[19:22:47]
[19:22:47]ets Jun  8 2016 00:22:57
[19:22:47]
[19:22:47]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[19:22:47]configsip: 0, SPIWP:0xee
[19:22:47]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[19:22:47]mode:DOUT, clock div:2
[19:22:47]load:0x3fff0018,len:4
[19:22:47]load:0x3fff001c,len:928
[19:22:47]ho 0 tail 12 room 4
[19:22:47]load:0x40078000,len:9280
[19:22:47]load:0x40080400,len:5860
[19:22:47]entry 0x40080698
[19:22:49][I][logger:119]: Log initialized
[19:22:49][C][ota:484]: There have been 1 suspected unsuccessful boot attempts.
[19:22:49][I][application:055]: Running through setup()...
[19:22:49][C][wifi:027]: Setting up WiFi...
[19:22:50][D][wifi:271]: Starting scan...
[19:22:52][D][wifi:286]: Found networks:
[19:22:52][I][wifi:321]: - 'horvath' (B4:FB:E4:CB:4A:9F) ▂▄▆█
[19:22:52][D][wifi:322]:     Channel: 3
[19:22:52][D][wifi:323]:     RSSI: -67 dB
[19:22:52][D][wifi:325]: - '' (B6:FB:E4:9B:4A:9F) ▂▄▆█
[19:22:52][I][wifi:167]: WiFi Connecting to 'horvath'...
[19:22:53][I][wifi:379]: WiFi connected!
[19:22:53][C][wifi:253]:   SSID: 'horvath'
[19:22:53][C][wifi:254]:   IP Address: 192.168.75.99
[19:22:53][C][wifi:256]:   BSSID: B4:FB:E4:CB:4A:9F
[19:22:53][C][wifi:257]:   Hostname: 'esp32_cam'
[19:22:53][C][wifi:261]:   Signal strength: -65 dB ▂▄▆█
[19:22:53][C][wifi:262]:   Channel: 3
[19:22:53][C][wifi:263]:   Subnet: 255.255.255.0
[... 291 lines cut ...]
[19:25:43]load:0x40078000,len:9280
[19:25:43]load:0x40080400,len:5860
[19:25:43]entry 0x40080698
```
And this is what it says now with esphome logs
```
nhorvath@nhor-ul2:~/Arduino$ esphome esp32_cam.yaml logs
INFO Reading configuration...
Found multiple options, please choose one:
  [1] /dev/ttyUSB0 (FT232R USB UART)
  [2] Over The Air (esp32_cam.horvath)
(number): 2
INFO Starting log output from esp32_cam.horvath using esphome API
INFO Connecting to esp32_cam.horvath:6053 (192.168.75.99)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.75.99: [Errno 111] Connection refused). Re-Trying in 1 seconds
INFO Connecting to esp32_cam.horvath:6053 (192.168.75.99)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.75.99: [Errno 111] Connection refused). Re-Trying in 2 seconds
INFO Connecting to esp32_cam.horvath:6053 (192.168.75.99)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.75.99: [Errno 111] Connection refused). Re-Trying in 4 seconds
INFO Connecting to esp32_cam.horvath:6053 (192.168.75.99)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.75.99: [Errno 111] Connection refused). Re-Trying in 8 seconds
INFO Connecting to esp32_cam.horvath:6053 (192.168.75.99)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.75.99: [Errno 111] Connection refused). Re-Trying in 16 seconds
```
I can ping it and telnet to 6053 (but nothing shows up on 6053, i don't know if there should be).

nhorvath, 2019-04-22:
It started working after a few minutes. Strange. I can't believe it would be overheating after 10 seconds. But I have heard the camera modules overheat so I added a heatsink to the back and it's running at 1600x1200 just fine now.

nhorvath, 2019-04-23:
Reopening because I unplugged the camera to move it to a different spot and now it says Setup Failed: ERROR.

```
[10:14:34][C][wifi:371]: WiFi:
[10:14:34][C][wifi:253]:   SSID: [redacted]
[10:14:34][C][wifi:254]:   IP Address: 192.168.75.99
[10:14:34][C][wifi:256]:   BSSID: [redacted]
[10:14:34][C][wifi:257]:   Hostname: 'esp32_cam'
[10:14:34][C][wifi:261]:   Signal strength: -77 dB ▂▄▆█
[10:14:34][C][wifi:262]:   Channel: 3
[10:14:34][C][wifi:263]:   Subnet: 255.255.255.0
[10:14:34][C][wifi:264]:   Gateway: 192.168.75.1
[10:14:34][C][wifi:265]:   DNS1: 192.168.75.25
[10:14:34][C][wifi:266]:   DNS2: 192.168.75.1
[10:14:34][C][logger:142]: Logger:
[10:14:34][C][logger:143]:   Level: DEBUG
[10:14:34][C][logger:144]:   Log Baud Rate: 115200
[10:14:34][C][logger:145]:   Hardware UART: UART0
[10:14:34][C][esp32_camera:044]: ESP32 Camera:
[10:14:34][C][esp32_camera:045]:   Name: ESP32-CAM
[10:14:34][C][esp32_camera:046]:   Board Has PSRAM: YES
[10:14:34][C][esp32_camera:048]:   Data Pins: D0:5 D1:18 D2:19 D3:21 D4:36 D5:39 D6:34 D7:35
[10:14:34][C][esp32_camera:049]:   VSYNC Pin: 25
[10:14:34][C][esp32_camera:050]:   HREF Pin: 23
[10:14:34][C][esp32_camera:051]:   Pixel Clock Pin: 22
[10:14:34][C][esp32_camera:052]:   External Clock: Pin:0 Frequency:20000000
[10:14:34][C][esp32_camera:053]:   I2C Pins: SDA:26 SCL:27
[10:14:34][C][esp32_camera:054]:   Reset Pin: -1
[10:14:34][C][esp32_camera:087]:   Resolution: 1600x1200 (UXGA)
[10:14:34][E][esp32_camera:092]:   Setup Failed: ERROR
[10:14:34][C][api:101]: API Server:
[10:14:34][C][api:102]:   Address: esp32_cam.horvath:6053
[10:14:34][C][ota:127]: Over-The-Air Updates:
[10:14:34][C][ota:128]:   Address: esp32_cam.horvath:3232
[10:14:34][C][ota:130]:   Using Password.
```

nhorvath, 2019-04-23:
Closing again, may have been a power issue. I changed power adapters and it went away.

closed 2019-04-23 by nhorvath

bwanajag, 2019-06-01:
I'm getting the setup failed error, but it's not resolving.  I have the TTGO-Camera with microphone, so I used the pins from [here](https://github.com/lewisxhe/esp32-camera-series)

```
[21:11:03][C][logger:137]: Logger:
[21:11:03][C][logger:138]:   Level: DEBUG
[21:11:03][C][logger:139]:   Log Baud Rate: 115200
[21:11:03][C][logger:140]:   Hardware UART: UART0
[21:11:03][C][esp32_camera:043]: ESP32 Camera:
[21:11:03][C][esp32_camera:044]:   Name: Hall Camera
[21:11:03][C][esp32_camera:045]:   Board Has PSRAM: YES
[21:11:03][C][esp32_camera:047]:   Data Pins: D0:34 D1:13 D2:14 D3:35 D4:39 D5:12 D6:15 D7:36
[21:11:03][C][esp32_camera:048]:   VSYNC Pin: 5
[21:11:03][C][esp32_camera:049]:   HREF Pin: 27
[21:11:03][C][esp32_camera:050]:   Pixel Clock Pin: 25
[21:11:03][C][esp32_camera:051]:   External Clock: Pin:4 Frequency:20000000
[21:11:03][C][esp32_camera:052]:   I2C Pins: SDA:21 SCL:22
[21:11:03][C][esp32_camera:053]:   Reset Pin: -1
[21:11:03][C][esp32_camera:074]:   Resolution: 640x480 (VGA)
[21:11:03][E][esp32_camera:091]:   Setup Failed: ERROR
```