esphome/issues #253: FastLED crazy flickering

Diaoul, 2019-04-21:
I have a custom made FastLED ESP8266 device which connects to my MQTT and integrates well with HomeAssistant using MQTT JSON Template component. I wanted something easier for my ESP8266 fleet so I tried ESPHome.
I had first some trouble uploading through OTA from Home Assistant so I switch to my PC and I'll sort this one out later.
However, the first thing that I immediately notice is that the Addressable Rainbow Effect flickers badly.
It does it on all addressable effects, it's more or less visible depending on the number of leds on at this moment.
I know clockless is a PITA and very sensitive but still. Is this a known limitation or a bug that can be solved?
**Operating environment/Installation (Hass.io/Docker/pip/etc.):** pip

**ESP (ESP32/ESP8266, Board/Sonoff):** ESP8266

**Affected component:** FastLED Light

**Description of problem:** Addressable Rainbow Effect is flickering

**Problem-relevant YAML-configuration entries:**
```yaml
sensor:
  - platform: dht
    pin: GPIO2
    temperature:
      name: "Temperature"
    humidity:
      name: "Humidity"
    update_interval: 60s

light:
  - platform: fastled_clockless
    chipset: NEOPIXEL
    pin: GPIO14
    num_leds: 60
    name: "Light"
    effects:
      - addressable_rainbow:
```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

OttoWinter, 2019-04-22:
> I know clockless is a PITA and very sensitive but still. Is this a known limitation or a bug that can be solved?

Limitation, you can try with different pins (gpio1/3) and the neopixelbus integration, but basically this is a limitation of these devices. With ESP32s flickering is almost always gone though.

closed 2019-04-22 by OttoWinter

Diaoul, 2019-04-22:
It worked with my code, same effects, less "integration" features. Is there
a way I can troubleshoot this and make it work or is it already under tons
of layers of code?

OttoWinter, 2019-04-22:
> Is there a way I can troubleshoot this and make it work or is it already under tons of layers of code?

It is. You can try removing stuff from the config (for example DHT) and playing with the `max_refresh_rate` option. Otherwise I mean esphome is also just using fastled internally as your project does.

Diaoul, 2019-04-22:
So I've disabled DHT with no effect and `max_refresh_rate` changes only slows the flickering.
IMO, there is something in the code messing with the timings, this is not good. I had the same issues.

I've compared with my code, [this is rather standard FastLED](https://github.com/Diaoul/LEDEffect/blob/master/src/LEDEffect/LEDEffect.hpp#L168-L188): [update LED array](https://github.com/Diaoul/LEDEffect/blob/master/src/LEDEffect/Effects/RainbowEffect.hpp#L35-L38), show, delay, repeat.

I've tried enabling debug logs but it seems that network activity isn't showed. Maybe ESPHome generates a lot of network activity for some reason ? Or other parts of ESPHome have blocking operations causing delay?

> With ESP32s flickering is almost always gone though.

Upgrading hardware to solve a software issue doesn't seem like a good idea.

Why did you close this issue? I think it's worth mentioning in the docs that clockless doesn't play well with ESP8266 if you don't plan of fixing it. Or is it just happening to me?

davericher, 2019-04-24:
^ arrogant responses like this on free projects make people not want to help you. A hardware limitation is a hardware limitation, he has already told you to upgrade to an ESP32, you can either do that or find a fix and submit a PR.  Your opinion on what is a good idea or what is not a good idea is absolutely irrelevant

Diaoul, 2019-04-24:
I didn't mean to be arrogant, I'm trying to get more info on the project to debug it. @OttoWinter mentioned that the issue is "almost gone", not fixed. It's also working for me with the same hardware, just different software. My reasoning (and what I meant with this comment) is that upgrading the hardware may only make the problem invisible to the human eye, but still there.
As those clockless LEDs require precise timing, this probably means that other component that would have the same requirements (microsecond precision) wouldn't work with ESPHome.
I don't own an ESP32, maybe it has just more power and hides the issue or maybe a different design making it insensitive to those things.

There is probably a library in the dependency tree messing with the interrupts, I seem to remember that the ESP8266 stops everything to run network stuff. Hopefully this can be solved upstream but this may be an upstream design decision.

I'd understand one wouldn't want to change the whole network layer of the project just because some component requires precise timings but this is not what @OttoWinter said before closing the ticket.

Closing an relevant ticket without a definitive answer feels harsh. I don't feel welcome to contribute ideas, code or docs update.

OttoWinter, 2019-04-24:
> So I've disabled DHT with no effect and max_refresh_rate changes only slows the flickering.

To what levels did you set max_refresh_rate? Your code has 30fps update rate, which would equate to 33ms max_refresh_rate.

> Maybe ESPHome generates a lot of network activity for some reason ?

It does not.

> Or other parts of ESPHome have blocking operations causing delay?

Nah, not the case (esphome's main loop is intentionally fully async). Plus that wouldn't result in flickering.

> Why did you close this issue?

Because it sounded like a setup and/or upstream issue.