esphome/issues #254: ESP32 freeze and hang

celeroll, 2019-04-22:
**Operating environment/Installation (/Docker/.):**

**ESP32** v1.12.2

**Affected component:**
ESP32

**Description of problem:**
I'm using an ESP32 to control my lowboard LEDs and a fan.
I have four LED strips (APA102) with just 11 leds on every strip. Additionally I have a fan which I have installed inside the lowboard to cool the inside area.
When I was testing the setup everything worked perfect. But not since I have installed it into the place after 10-15 min the ESP freezes. After  that there is no way to get it online. Only solution is to physically disconnect every component (4 Led strips + fan) and wait! again 10 - 15 min and only then the ESP will come online. 
But after couple of minutes it will  freeze again and the game starts from beginning.

**Problem-relevant YAML-configuration entries:**
Here is my yaml file 
```
esphome:
  name: esp32_node1
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: !secret wifi_id
  password: !secret wifi_pass
  manual_ip:
    static_ip: 192.168.1.10
    gateway: 192.168.1.1
    subnet: 255.255.255.0

# Enable logging
logger:
  # level: VERBOSE

# Enable Home Assistant API
api:

ota:

#====== fan start
fan:
  - platform: speed
    output: fanlowboard1
    name: "Lowboard Fan"
    # speed:
    #   low: 0.1
    #   medium: 0.3
    #   high: 1
      # very_low: 0.1

output:
  - platform: ledc
    pin: GPIO21
    id: fanlowboard1
    frequency: 25000
    bit_depth: 10

#====== fan end

light:
  #LED LowBoard Led1
  - platform: fastled_spi
    chipset: APA102
    data_pin: GPIO23 #yellow
    clock_pin: GPIO22 #orange
    num_leds: 11
    rgb_order: BGR
    name: "LowBoard Led1"
    default_transition_length : 0s
    effects:
      - random:
      - strobe:
      - flicker:
      - addressable_rainbow:
      - addressable_color_wipe:
      - addressable_scan:
      - addressable_twinkle:
      - addressable_random_twinkle:
      - addressable_fireworks:
      - addressable_flicker:
  
  #LED LowBoard Led2
  - platform: fastled_spi
    chipset: APA102
    data_pin: GPIO32 #yellow
    clock_pin: GPIO33 #orange
    num_leds: 11
    rgb_order: BGR
    name: "LowBoard Led2"
    default_transition_length : 0s
    effects:
      - random:
      - strobe:
      - flicker:
      - addressable_rainbow:
      - addressable_color_wipe:
      - addressable_scan:
      - addressable_twinkle:
      - addressable_random_twinkle:
      - addressable_fireworks:
      - addressable_flicker:
  
  #LED LowBoard Led3
  - platform: fastled_spi
    chipset: APA102
    data_pin: GPIO25 #yellow
    clock_pin: GPIO26 #orange
    num_leds: 11
    rgb_order: BGR
    name: "LowBoard Led3"
    default_transition_length : 0s
    effects:
      - random:
      - strobe:
      - flicker:
      - addressable_rainbow:
      - addressable_color_wipe:
      - addressable_scan:
      - addressable_twinkle:
      - addressable_random_twinkle:
      - addressable_fireworks:
      - addressable_flicker:

  #LED LowBoard Led4
  - platform: fastled_spi
    chipset: APA102
    data_pin: GPIO27 #yellow
    clock_pin: GPIO14 #orange
    num_leds: 11
    rgb_order: BGR
    name: "LowBoard Led4"
    default_transition_length : 0s
    effects:
      - random:
      - strobe:
      - flicker:
      - addressable_rainbow:
      - addressable_color_wipe:
      - addressable_scan:
      - addressable_twinkle:
      - addressable_random_twinkle:
      - addressable_fireworks:
      - addressable_flicker:

```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

OttoWinter, 2019-04-24:
Logs please, without them this is impossible to debug.

celeroll, 2019-04-24:
> Logs please, without them this is impossible to debug.

Can I get logs without hooking up over USB?
By just starting it up?

OttoWinter, 2019-04-25:
> Can I get logs without hooking up over USB?

Depends on the issue, if the ESP is not connecting to the network then you have to use USB logs, yes.

celeroll, 2019-04-26:
So this is the log before it freezes.
Actually no info.
It was set to run rainbow effect on four LED strips. 
It freezes so after 2-3 min.

```
[20:58:45][VV][light.fast_led:041]: Writing RGB values to bus...
[20:58:45][VV][light.fast_led:041]: Writing RGB values to bus...
[20:58:45][VV][light.fast_led:041]: Writing RGB values to bus...
[20:58:45][VV][light.fast_led:041]: Writing RGB values to bus...
[20:58:45][VV][light.fast_led:041]: Writing RGB values to bus...
[20:58:45][VV][light.fast_led:041]: Writing RGB values to bus...
[20:58:45][VV][light.fast_led:041]: Writing RGB values to bus...
[20:58:45][VV][light.fast_led:041]: Writing RGB values to bus...
[20:58:45][VV][light.fast_led:041]: Writing RGB values to bus...
[20:58:45][VV][light.fast_led:041]: Writing RGB values to bus...
[20:58:45][VV][light.fast_led:041]: Writing RGB values to bus...
[20:58:45][VV][light.fast_led:041]: Writing RGB values to bus...
[20:58:45][VV][light.fast_led:041]: Writing RGB values to bus...
[20:58:45][VV][light.fast_led:041]: Writing RGB values to bus...
[20:58:45][VV][light.fast_led:041]: Writing RGB values to bus...
[20:58:45][VV][light.fast_led:041]: Writing RGB values to bus...
[20:58:45][VV][light.fast_led:041]: Writing RGB values to bus...
[20:58:45][VV][light.fast_led:041]: Writing RGB values to bus...
[20:58:45][VV][light.fast_led:041]: Writing RGB values to bus...
[20:58:45][VV][light.fast_led:041]: Writing RGB values to bus...
[20:58:45][VV][light.fast_led:041]: Writing RGB values to bus...
[20:58:45][VV][light.fast_led:041]: Writing RGB values to bus...
[20:58:45][VV][light.fast_led:041]: Writing RGB values to bus...
[20:58:45][VV][light.fast_led:041]: Writing RGB values to bus...
[20:58:45][VV][light.fast_led:041]: Writing RGB values to bus...
[20:58:45][VV][light.fast_led:041]: Writing RGB values to bus...
[20:58:45][VV][light.fast_led:041]: Writing RGB values to bus...
[20:58:45][VV][light.fast_led:041]: Writing RGB values to bus...
[20:58:45][VV][light.fast_led:041]: Writing RGB values to bus...
WARNING Disconnected from API.
INFO Connecting to 192.168.1.10:6053 (192.168.1.10)
WARNING Couldn't connect to API (Error connecting to 192.168.1.10: timed out). Trying to reconnect in 1 seconds
WARNING Disconnected from API.
INFO Connecting to 192.168.1.10:6053 (192.168.1.10)
WARNING Couldn't connect to API (Error connecting to 192.168.1.10: timed out). Trying to reconnect in 2 seconds
WARNING Disconnected from API.
INFO Connecting to 192.168.1.10:6053 (192.168.1.10)
WARNING Couldn't connect to API (Error connecting to 192.168.1.10: timed out). Trying to reconnect in 4 seconds
```

celeroll, 2019-04-26:
So I went even further with my testings.
When the ESP32 freezes I connect it additionally over USB and the "writing RGB values to bus" continues. If I leave it connected for couple of seconds and discconect the ESP32 will continue to work until next freeze. 
In the serial log I cant see anything else.
If I would leave ESP connected over USB also it will work without problems.

I have connected all components and ESP32 over same Ground and Line. On ESP it is the pin 5V pin.

celeroll, 2019-04-28:
OK, I found the solution.
I had to power my ESP32 through USB! instead through VIN 5V like the LED strip. 
I even tried with level shifter, but this has delayed the freeze time to about 1,5h but an the end same problem.
But don't forget to disconnect VIN if you connect ESP32 through USB!

closed 2019-04-28 by celeroll