esphome/issues #258: LEDC / Binary lights not working in newest dev update

davericher, 2019-04-24:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

PIP on current DEV

**ESP (ESP32/ESP8266, Board/Sonoff):**

NodeMcu-v2 esp32

**Affected component:**
ESP32 / LEDC / Binary GPIO

**Description of problem:**
Does not turn light on (does work in HA as far as showing the proper on-off status, just no effect)

**Problem-relevant YAML-configuration entries:**
```yaml
# Bootstrap
esphome:
  name: $devicename
  platform: $platform
  board: $board
  build_path: builds/${devicename}
  # esphome_core_version: dev
  includes:
    - common.h
web_server:
  port: $web_port
status_led:
  pin: $status_led
wifi:
  ssid: !secret primary_ap
  password: !secret primary_ap_psk
  manual_ip:
    static_ip: $static_ip
    gateway: $gateway
    subnet: $subnet
    dns1: $dns1
    dns2: $dns2
api:
  services:
    - service: display_message
      variables:
        msg: string
      then:
        - lambda:
            id(message).clear();
            id(message) = msg;
        - component.update: currentDisplay
        - delay: 5s
        - lambda: 'id(message).clear();'
        - component.update: currentDisplay

    - service: start_status_effect
      variables:
        brightness: int
        effect: string
      then:
        - light.turn_on:
            id: status_light
            brightness: !lambda 'return brightness;'
            effect: !lambda 'return effect;'
    - service: start_indicator_effect
      variables:
        brightness: int
        effect: string
      then:
        - light.turn_on:
            id: indicator_light
            brightness: !lambda 'return brightness;'
            effect: !lambda 'return effect;'

logger:
  level: $logger_level
ota:
  password: !secret ota_password
time:
  - platform: homeassistant
    id: currentTime

# Outputs
output:
  - platform:           ledc
    pin:                $led_pin
    id:                 indicator
    frequency:          250
  - platform:           ledc
    pin:                $status_pin
    id:                 statusLight

esp32_touch:
  # setup_mode: True

# Globals
globals:
  - id:                 screen_state
    type:               int
  - id:                 total_screens
    type:               int
    initial_value:      '1'
  - id:                 message
    type:               std::string
    initial_value:      ""

# Intervals
interval:
  - interval: $next_screen_interval
    then:
      - lambda:         |-
          if (id(screen_state) < id(total_screens)) {
            id(screen_state) += 1;
          } else {
            id(screen_state) = 1;
          }

# Text Sensors
text_sensor:
  - <<: !include ../common/versionTextSensor.yaml
  - <<: !include ../common/wifiInfoTextSensor.yaml
  - <<: !include ../common/octoprintPrintingPercentTextSensor.yaml

# Sensors
binary_sensor:
  - <<: !include ../common/statusBinarySensor.yaml
  - <<: !include ../common/octoprintPrintingBinarySensor.yaml
  ## Plus Button
  - platform: esp32_touch
    id: ${devicename}_touch_button_1
    name: ${upper_devicename} Touch Button 1
    pin: $t1_pin
    threshold: 930
    on_click:
      then:
        - homeassistant.service:
            service: switch.turn_on
            data:
              entity_id: "switch.office_bluray_volume_up"
  ## Minus Button
  - platform: esp32_touch
    id: ${devicename}_touch_button_2
    name: ${upper_devicename} Touch Button 2
    pin: $t2_pin
    threshold: 930
    on_click:
      then:
        - homeassistant.service:
            service: switch.turn_on
            data:
              entity_id: "switch.office_bluray_volume_down"

  ## Gear Button
  - platform: esp32_touch
    id: ${devicename}_touch_button_3
    name: ${upper_devicename} Touch Button 3
    pin: $t3_pin
    threshold: 930
    on_click:
      then:
        - homeassistant.service:
            service: automation.trigger
            data:
              entity_id: "automation.office_scene_switch_next"

  ## Star Button
  - platform: esp32_touch
    id: ${devicename}_touch_button_4
    name: ${upper_devicename} Touch Button 4
    pin: $t4_pin
    threshold: 930
    on_click:
      then:
        - homeassistant.service:
            service: automation.trigger
            data:
              entity_id: "automation.office_scene_switch_previous"

  ## "1" Button
  - platform: esp32_touch
    id: ${devicename}_touch_button_5
    name: ${upper_devicename} Touch Button 5
    pin: $t5_pin
    threshold: 1100
    on_click:
      then:
        - homeassistant.service:
            service: switch.turn_on
            data:
              entity_id: "switch.office_ac_power"

  ## "2" Button
  - platform: esp32_touch
    id: ${devicename}_touch_button_6
    name: ${upper_devicename} Touch Button 6
    pin: $t6_pin
    threshold: 1100
    on_click:
      then:
        - homeassistant.service:
            service: switch.turn_on
            data:
              entity_id: "switch.office_bluray_3d_sound"

# Sensors
sensor:
  - <<: !include ../common/wifiSignalSensor.yaml
  - <<: !include ../common/uptimeSensor.yaml
  - <<: !include ../common/octoprintCurrentBedTempSensor.yaml
  - <<: !include ../common/octoprintCurrentHotendTempSensor.yaml

# Individual switches
switch:
  - <<: !include ../common/restartSwitch.yaml
  - <<: !include ../common/shutdownSwitch.yaml

display:
  - platform:           lcd_gpio
    dimensions:         $lcd_dimensions
    data_pins:
      - GPIO21
      - GPIO19
      - GPIO18
      - GPIO17
      - GPIO16
      - GPIO25
      - GPIO26
      - GPIO5
    update_interval:    1s
    enable_pin:         $lcd_enable_pin
    rs_pin:             $lcd_rs_pin
    id:                 currentDisplay
    lambda:             |-
      const std::string empty = "";
      if(id(message).size() > 0){
        it.printf("%s", id(message).c_str());
      }
      else {
        switch(id(screen_state)) {
          case 0:
              it.print("Loading...");
              break;
          case 1:
              it.strftime(0, 1, "%I:%M:%S%p", id(currentTime).now());
              if (
                id(${upper_devicename}_octoprint_printing).state &&
                id(${devicename}_octoprint_current_bed_temp).has_state() &&
                id(${devicename}_octoprint_current_hotend_temp).has_state()
              ) {
                it.printf(
                  0,
                  0,
                  "Printing %.0f/%.0f",
                  id(${devicename}_octoprint_current_bed_temp).state,
                  id(${devicename}_octoprint_current_hotend_temp).state
                );
              }
              else {
                // Print Date/Time
                it.strftime(0, 0, "%a %b %d", id(currentTime).now());
              }
              if (
                id(${upper_devicename}_octoprint_printing).state &&
                id(${devicename}_octoprint_printing_percent).has_state()
              ) {
                std::string out = id(${devicename}_octoprint_printing_percent).state;
                out.insert(out.size(), 1, '%');
                padTo(out,  5, ' ', true);
                it.printf(11, 1, "%s", out.c_str());
              }
              break;
        }

      }

# Lights
light:
  - platform:           monochromatic
    id: indicator_light
    name:               ${upper_devicename} LCD Backlight
    output:             indicator
    effects:
      - strobe:
  - platform:           monochromatic
    id: status_light
    name:               ${upper_devicename} Status Light
    output:             statusLight
    effects:
      - strobe:

```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**
- tried with Binary light, no go
- hard wired to power, works, not a hardware issue

OttoWinter, 2019-04-24:
Yeah the dev branch is pretty experimental right now. I see the issue now, will be a quick fix.

closed 2019-04-24 by OttoWinter

davericher, 2019-04-24:
I figured :D I know the dangers of living on the edge. The work your doing seems fairly awesome in the housekeeping department. Keep it up

davericher, 2019-04-24:
Sadly I do not think it was as quick a fix as you imagined.