esphome/issues #260: Hi,  I’ve successfully put the MiFlora sensors into HA before through the core HA component. However, due to the position of the sensors and my RPi, I decided to use an ESP32 as a BLE hub for better connectivity.  I’ve followed the instructions and everything appears to be working. The sensor node is online and it is detecting the MAC address of the MiFlora sensor. It is not however showing up in Home Assistant.  This is my yaml:

eflemingtra, 2019-04-24:
**Operating environment/Installation (Hassbian):**

Hassbian installed on Raspberry Pi. Version 0.91 of HA installed.

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 on esp32doit-devkit-v1 board

**Affected component:**

Xiaomi MiFlora and/or BLE Tracker

**Description of problem:**
Hi,

I’ve successfully put the MiFlora sensors into HA before through the core HA component. However, due to the position of the sensors and my RPi, I decided to use an ESP32 as a BLE hub for better connectivity.

I’ve followed the instructions and everything appears to be working. The sensor node is online and it is detecting the MAC address of the MiFlora sensor. It is not however showing up in Home Assistant.

This is my yaml:

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: blehubtrack
  platform: ESP32
  board: esp32doit-devkit-v1

wifi:
  ssid: "***********"
  password: "***********"

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

esp32_ble_tracker:
  scan_interval: 300s

sensor:
  - platform: xiaomi_miflora
    mac_address: C4:7C:8D:67:C0:8B
    temperature:
      name: "Xiaomi MiFlora Temperature"
    moisture:
      name: "Xiaomi MiFlora Moisture"
    illuminance:
      name: "Xiaomi MiFlora Illuminance"
    conductivity:
      name: "Xiaomi MiFlora Soil Conductivity"
    battery_level:
      name: "Xiaomi MiFlora Battery Level" 

```
Log is as follows:

**Traceback (if applicable):**

```
[16:01:30][I][application:097]: esphome-core version 1.12.2 compiled on Apr 24 2019, 15:59:09
[16:01:30][C][wifi:371]: WiFi:
[16:01:30][C][wifi:253]:   SSID: [redacted]
[16:01:30][C][wifi:254]:   IP Address: 192.168.1.49
[16:01:30][C][wifi:256]:   BSSID: [redacted]
[16:01:30][C][wifi:257]:   Hostname: 'blehubtrack'
[16:01:30][C][wifi:261]:   Signal strength: -54 dB ▂▄▆█
[16:01:30][C][wifi:262]:   Channel: 6
[16:01:30][C][wifi:263]:   Subnet: 255.255.255.0
[16:01:30][C][wifi:264]:   Gateway: 192.168.1.1
[16:01:30][C][wifi:265]:   DNS1: 192.168.1.1
[16:01:30][C][wifi:266]:   DNS2: 0.0.0.0
[16:01:30][C][logger:142]: Logger:
[16:01:30][C][logger:143]:   Level: DEBUG
[16:01:30][C][logger:144]:   Log Baud Rate: 115200
[16:01:30][C][logger:145]:   Hardware UART: UART0
[16:01:30][C][esp32_ble_tracker:703]: BLE Tracker:
[16:01:30][C][esp32_ble_tracker:704]:   Scan Interval: 300 s
[16:01:30][C][esp32_ble_tracker:712]:   Xiaomi 0000C47C8D67C08B
[16:01:30][C][esp32_ble_tracker:713]:     Temperature  'Xiaomi MiFlora Temperature'
[16:01:30][C][esp32_ble_tracker:713]:       Unit of Measurement: '°C'
[16:01:30][C][esp32_ble_tracker:713]:       Accuracy Decimals: 1
[16:01:30][C][esp32_ble_tracker:715]:     Moisture  'Xiaomi MiFlora Moisture'
[16:01:30][C][esp32_ble_tracker:715]:       Unit of Measurement: '%'
[16:01:30][C][esp32_ble_tracker:715]:       Accuracy Decimals: 0
[16:01:30][C][esp32_ble_tracker:715]:       Icon: 'mdi:water-percent'
[16:01:30][C][esp32_ble_tracker:716]:     Illuminance  'Xiaomi MiFlora Illuminance'
[16:01:30][C][esp32_ble_tracker:716]:       Unit of Measurement: 'lx'
[16:01:30][C][esp32_ble_tracker:716]:       Accuracy Decimals: 0
[16:01:30][C][esp32_ble_tracker:716]:       Icon: 'mdi:brightness-5'
[16:01:30][C][esp32_ble_tracker:717]:     Conductivity  'Xiaomi MiFlora Soil Conductivity'
[16:01:30][C][esp32_ble_tracker:717]:       Unit of Measurement: 'µS/cm'
[16:01:30][C][esp32_ble_tracker:717]:       Accuracy Decimals: 0
[16:01:30][C][esp32_ble_tracker:717]:       Icon: 'mdi:flower'
[16:01:30][C][esp32_ble_tracker:718]:     Battery Level  'Xiaomi MiFlora Battery Level'
[16:01:30][C][esp32_ble_tracker:718]:       Unit of Measurement: '%'
[16:01:30][C][esp32_ble_tracker:718]:       Accuracy Decimals: 0
[16:01:30][C][esp32_ble_tracker:718]:       Icon: 'mdi:battery'
[16:01:30][C][api:101]: API Server:
[16:01:30][C][api:102]:   Address: blehubtrack.local:6053
[16:01:30][C][ota:127]: Over-The-Air Updates:
[16:01:30][C][ota:128]:   Address: blehubtrack.local:3232
[16:06:25][D][esp32_ble_tracker:431]: Found device C4:7C:8D:67:C0:8B RSSI=-44
[16:06:25][D][esp32_ble_tracker:452]:   Address Type: PUBLIC
```

**Additional information and things you've tried:**

Now, one thing I have noticed from the ESPHome guides. When discovering devices, it has never displayed this:

```
Xiaomi MiFlora 94:2B:FF:5C:91:61 Got temperature=23.4°C
```

Which as per the ESP guide for the MiFLora, it should. Is it a case that it can see the device, but isn’t recognising it as a MiFlora device? Running `sudo hcitool lescan` shows 

```
C4:7C:8D:67:C0:8B Flower care
```
 when run from my Pi.

OttoWinter, 2019-04-24:
> It is not however showing up in Home Assistant.

More detail please. Not showing up at all (no entity) or just not showing any data?
Please make sure you added the ESP in the integrations screen.

> When discovering devices, it has never displayed this:

That sounds like it's the issue. If you don't see that message, the MiFlora is not sending the required BLE frames. Please observe log output for some more time.

eflemingtra, 2019-04-24:
Hi Otto,

Found out the problem. I'd assumed that the firmware on the MiFlora was up to date. It wasn't. Flashed with the latest firmware and it's now broadcasting correctly. My apologies for wasting your time.

closed 2019-04-24 by eflemingtra

adrianmihalko, 2019-04-29:
I had same problem. Miflora was not recognized, but after upgrading from 2.7 to 3.1.9, it is working now. Maybe it is worth to add this info to the docs.

plutec, 2019-06-04:
I have now a MiFlora with version 3.2.1 and have the same issue, the MAC address does not appear. Any idea?

nhnam100, 2019-06-05:
My flora with ver. 3.2.1 now  is working well.

plutec, 2019-06-05:
Uhm, maybe I don't have ESPHome updated. I will try later. Thanks!