esphome/issues #261: Sonoff 4CH pro disables button after double click

lblokland, 2019-04-24:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hassio + ESPHome
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266, Sonoff 4CH Pro
**Affected component:**

https://esphome.io/devices/sonoff_4ch.html

**Description of problem:**
When I  set up on_double_click on the binary sensor I discovered after a succesful double click action the button isn't responding anymore.
Changing the on_double_click into on_multi_click doesn't help, in fact setting no click action at all will also break the button.
After 'some' time it looks like the button's function is restored, I cannot tell when or why the function is restored

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphome:
  name: sonoff_4pro_1
  platform: ESP8266
  board: esp01_1m
  
web_server:
  port: 80

wifi:
  ssid: "150"
  password: ""
  manual_ip:
    static_ip: 192.168.178.17
    gateway: 192.168.178.1
    subnet: 255.255.255.0
    dns1: 192.168.178.1
    
binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    name: "Sonoff 4pro1 Button 1"
    on_press:
      then:
        - switch.toggle: sonoff4pro1r1
  - platform: gpio
    pin:
      number: GPIO9
      mode: INPUT_PULLUP
      inverted: True
    name: "Sonoff 4pro1 Button 2"
    on_press:
      then:
        - switch.toggle: sonoff4pro1r2
  - platform: gpio
    pin:
      number: GPIO10
      mode: INPUT_PULLUP
      inverted: True
    name: "Sonoff 4pro1 Button 3"
  - platform: gpio
    pin:
      number: GPIO14
      mode: INPUT_PULLUP
      inverted: True
    name: "Sonoff 4pro1 Button 4"
    on_multi_click:
    - timing:
        - ON for at most 500ms
        - OFF for at least 5ms
        - ON for at most 500ms
        - OFF for at least 5ms
      then:
        - logger.log: "Double Clicked"
    - timing:
        - ON for 4s to 6s
        - OFF for at least 0.5s
      then:
        - logger.log: "Single Long Clicked"
        - switch.toggle: sonoff4pro1restart
    - timing:
        - ON for at most 1s
        - OFF for at least 0.5s
      then:
      - logger.log: "Single Short Clicked"
      - switch.toggle: sonoff4pro1r4
  - platform: status
    name: "Sonoff 4pro1 Status"

switch:
  - platform: gpio
    name: "Sonoff 4pro1 Relay 1"
    pin: GPIO12
    id: sonoff4pro1r1
    restore_mode: ALWAYS_OFF
    on_turn_on:
    - logger.log: "Switch sonoff4pro1r1 Turned On!"
    - delay: 2s
    - switch.turn_off: sonoff4pro1r1
  - platform: gpio
    name: "Sonoff 4pro1 Relay 2"
    pin: GPIO5
    id: sonoff4pro1r2
  - platform: gpio
    name: "Sonoff 4pro1 Relay 3"
    pin: GPIO4
    id: sonoff4pro1r3
  - platform: gpio
    name: "Sonoff 4pro1 Relay 4"
    pin: GPIO15
    id: sonoff4pro1r4
  - platform: restart
    name: "Sonoff basic1 Restart"
    id: sonoff4pro1restart

output:
  # Register the blue LED as a dimmable output ....
  - platform: esp8266_pwm
    id: blue_led
    pin: GPIO13
    inverted: True
    
sensor:
  - platform: uptime
    name: Sonoff 4pro1 Uptime Sensor
  - platform: wifi_signal
    name: "Sonoff 4pro1 WiFi Signal Sensor"
    update_interval: 60s

light:
  # ... and then make a light out of it.
  - platform: monochromatic
    name: "Sonoff 4pro1 Blue LED"
    output: blue_led

debug:
# Enable logging
logger:
  level: debug

# Enable Home Assistant API
api:
  password: ''

ota:
  password: ''
```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

OttoWinter, 2019-04-25:
Sounds like a hardware issue - don't think the triggers would be broken in that way.

Please attach logs showing the binary sensor events (with the timestamps if possible).

lblokland, 2019-04-25:
Hey Otto,

I found that I have the Sonoff 4ch pro R2.
That model also has 433MHz RF, and the double click initiates pairing to a 433MHz command to a button.
I guess I had to but a Sonoff 4ch R2 (without pro) to be able to use double click features of the code.

Unless you know another solution I guess I have to live with this…

Cheers,
Leon

Verzonden vanuit Mail<https://go.microsoft.com/fwlink/?LinkId=550986> voor Windows 10

OttoWinter, 2019-04-26:
Ok if there's another MCU in between then yes on_double_click won't work (and can't do so).

closed 2019-04-26 by OttoWinter