esphome/issues #263: ERROR : vPortTaskWrapper at /Users/ficeto/.../freertos/port.c:355 (discriminator 1)

velocyped, 2019-04-26:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

HassOS (Hass.io) on Raspberry Pi 3 B+

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 WROOM
https://fr.aliexpress.com/item/ESP-32S-ESP-WROOM-32-ESP32-ESP-32-Bluetooth-and-WIFI-Dual-Core-CPU-with-Low/32864722159.html

**Affected component:**

https://esphome.io/devices/nodemcu_esp32.html

**Description of problem:**
The ERROR happens since I've played with the ESP32 Board type when adding a new "node" in HASSIO EspHome Dashboard. Before that my ESP32 worked well with a BME280 connected to it via I2C, HASSIO gave me the values of the sensors. Now it does not work anymore.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
core-ssh:~# cat /config/esphome/klsesp1.yaml 
esphome:
  name: klsesp1
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: "MADOUFI"
  password: "onsenfout"

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

i2c:
  sda: 21
  scl: 22
  scan: True

sensor:
  - platform: bme280
    temperature:
      name: "BME280 Temperature"
      oversampling: 16x
    pressure:
      name: "BME280 Pressure"
    humidity:
      name: "BME280 Humidity"
    address: 0x76
    update_interval: 5s

```

**Traceback (if applicable):**

```
INFO Reading configuration...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/klsesp1
Processing klsesp1 (platform: espressif32@1.6.0; framework: arduino; board: nodemcu-32s)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/nodemcu-32s.html
PLATFORM: Espressif 32 > NodeMCU-32S
HARDWARE: ESP32 240MHz 320KB RAM (4MB Flash)
DEBUG: CURRENT(esp-prog) EXTERNAL(esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
[... 132 lines cut ...]
WARNING Decoded 0x400e5a91: nvs_set_blob at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/nvs_flash/src/nvs_api.cpp:521
WARNING Decoded 0x40130b02: Preferences::putBytes(char const*, void const*, unsigned int)
WARNING Decoded 0x400d40fc: esphome::ESPPreferenceObject::save_internal_()
WARNING Decoded 0x400d426f: esphome::ESPPreferenceObject::save_()
WARNING Decoded 0x400d54d5: bool esphome::ESPPreferenceObject::save<unsigned int>(unsigned int*)
WARNING Decoded 0x400d55f1: esphome::OTAComponent::start_safe_mode(unsigned char, unsigned int)
WARNING Decoded 0x400d2197: setup()
WARNING Decoded 0x400dcdf3: loopTask(void*)
WARNING Decoded 0x40091afd: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c:355 (discriminator 1)
[17:40:27]
[17:40:27]Rebooting...
```

**Additional information and things you've tried:**

I've tried :
Reflashing SD card of RPI with HassOS (complete reinstall)
Get another ESP32 (I've 2 of them) and read flash of it to copy it to the non-working one : 

`esptool.py -b 115200 read_flash 0x00000 0x400000 flash_4M.bin`

`esptool.py -b 115200  write_flash --flash_freq 80m 0x000000 flash_4M.bin`

And : 
`esptool.py erase_flash`

velocyped, 2019-04-26:
I have added debug with this option in YAML file : 
```
debug:
# Enable logging
logger:
  level: debug
```

And the output is with a different stack trace but ends on the same file error : "**freertos/port.c:355**"

```
INFO Reading configuration...
INFO Starting log output from /dev/ttyUSB0 with baud rate 115200
[22:27:32][I][logger:119]: Log initialized
[22:27:32][C][ota:484]: There have been 5 suspected unsuccessful boot attempts.
[22:27:32][I][application:055]: Running through setup()...
[22:27:32][C][sensor.bme280:101]: Setting up BME280...
[22:27:32]Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
[22:27:32]Core 1 register dump:
[22:27:32]PC      : 0x400d4fdf  PS      : 0x00060530  A0      : 0x800d5138  A1      : 0x3ffb1d30  
INFO Need to fetch platformio IDE-data, please stand by
INFO Running:  platformio run -d /config/esphome/kesp1 -t idedata
WARNING Decoded 0x400d4fdf: esphome::I2CComponent::raw_begin_transmission(unsigned char)
[22:27:41]A2      : 0x003ffb8c  A3      : 0x0000008c  A4      : 0x3ffb1de0  A5      : 0x3ffb1de0  
[22:27:41]A6      : 0x3ffb8a68  A7      : 0x00000000  A8      : 0x800d50e1  A9      : 0x3ffb1d30  
[22:27:41]A10     : 0x3ffc0c58  A11     : 0x0000004b  A12     : 0x00000001  A13     : 0x00000000  
[22:27:41]A14     : 0x3ffb8a68  A15     : 0x00000076  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
[22:27:41]EXCVADDR: 0x003ffba8  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  
[22:27:41]
[22:27:41]Backtrace: 0x400d4fdf:0x3ffb1d30 0x400d5135:0x3ffb1d50 0x400d5209:0x3ffb1d80 0x400d5257:0x3ffb1db0 0x400d6fe8:0x3ffb1dd0 0x400d3fa2:0x3ffb1e00 0x400d3529:0x3ffb1e50 0x400d22ef:0x3ffb1ea0 0x400dd41f:0x3ffb1fb0 0x40091afd:0x3ffb1fd0
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x400d4fdf: esphome::I2CComponent::raw_begin_transmission(unsigned char)
WARNING Decoded 0x400d5135: esphome::I2CComponent::write_bytes(unsigned char, unsigned char, unsigned char const*, unsigned char)
WARNING Decoded 0x400d5209: esphome::I2CComponent::write_byte(unsigned char, unsigned char, unsigned char)
WARNING Decoded 0x400d5257: esphome::I2CDevice::write_byte(unsigned char, unsigned char)
WARNING Decoded 0x400d6fe8: esphome::sensor::BME280Component::setup()
WARNING Decoded 0x400d3fa2: esphome::PollingComponent::call_setup()
WARNING Decoded 0x400d3529: esphome::Application::setup()
WARNING Decoded 0x400d22ef: setup()
WARNING Decoded 0x400dd41f: loopTask(void*)
WARNING Decoded 0x40091afd: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c:355 (discriminator 1)
[22:27:45]
[22:27:45]Rebooting...
```

OttoWinter, 2019-05-01:
> And the output is with a different stack trace but ends on the same file error : "freertos/port.c:355"

That's the top of the stack - that only says that the error is happening during the setup method. With these stack traces one always has to look at the top of the trace. Is this with the same config as before (plus the added logging)?

Anyway, that looks like a separate issue (although a very weird one). The initial stack trace points to code inside of the esp-idf SDK (so code by espressif)

velocyped, 2019-05-02:
Does that mean that you can find a solution for this bug ? Or must I open an issue in Expressif's bug tracker (they only seem to have a forum "report bugs") ?

velocyped, 2019-05-04:
I have added "VERY_VERBOSE" as log level and I also ran : 
`platformio platform update
`
 and here is the result.

INFO Reading configuration...
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Compiling app...
INFO Running:  platformio run -d config/klsesp2
Processing klsesp2 (platform: espressif32@1.6.0; framework: arduino; board: nodemcu-32s)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
LibraryManager: Installing id=1826 @ 1.0.3
Downloading...
Unpacking  [####################################]  100%
LibraryManager: Installing v1.12.2
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
LibraryManager: Installing id=3837 @ 5.13.3
Downloading...
Unpacking  [####################################]  100%
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/nodemcu-32s.html
PLATFORM: Espressif 32 > NodeMCU-32S
HARDWARE: ESP32 240MHz 320KB RAM (4MB Flash)
DEBUG: CURRENT(esp-prog) EXTERNAL(esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 27 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <AsyncTCP> 1.0.3
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <Preferences> 1.0
|-- <Update> 1.0
|-- <esphome-core> 1.12.2
|   |-- <ArduinoJson-esphomelib> 5.13.3
|   |-- <Update> 1.0
|   |-- <WiFi> 1.0
|   |-- <Preferences> 1.0
|   |-- <AsyncTCP> 1.0.3
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
Compiling .pioenvs/klsesp2/src/main.cpp.o
Generating partitions .pioenvs/klsesp2/partitions.bin
Compiling .pioenvs/klsesp2/lib252/AsyncTCP_ID1826/AsyncTCP.cpp.o
Compiling .pioenvs/klsesp2/libf5c/WiFi/ETH.cpp.o
Compiling .pioenvs/klsesp2/libf5c/WiFi/WiFi.cpp.o
Compiling .pioenvs/klsesp2/libf5c/WiFi/WiFiAP.cpp.o
Compiling .pioenvs/klsesp2/libf5c/WiFi/WiFiClient.cpp.o
Archiving .pioenvs/klsesp2/lib252/libAsyncTCP_ID1826.a
Indexing .pioenvs/klsesp2/lib252/libAsyncTCP_ID1826.a
Compiling .pioenvs/klsesp2/libf5c/WiFi/WiFiGeneric.cpp.o
Compiling .pioenvs/klsesp2/libf5c/WiFi/WiFiMulti.cpp.o
Compiling .pioenvs/klsesp2/libf5c/WiFi/WiFiSTA.cpp.o
Compiling .pioenvs/klsesp2/libf5c/WiFi/WiFiScan.cpp.o
Compiling .pioenvs/klsesp2/libf5c/WiFi/WiFiServer.cpp.o
Compiling .pioenvs/klsesp2/libf5c/WiFi/WiFiUdp.cpp.o
Compiling .pioenvs/klsesp2/liba13/ESPmDNS/ESPmDNS.cpp.o
Compiling .pioenvs/klsesp2/libcbf/Preferences/Preferences.cpp.o
Compiling .pioenvs/klsesp2/lib879/Update/Updater.cpp.o
Archiving .pioenvs/klsesp2/libf5c/libWiFi.a
Indexing .pioenvs/klsesp2/libf5c/libWiFi.a
Archiving .pioenvs/klsesp2/libc21/libArduinoJson-esphomelib_ID3837.a
Indexing .pioenvs/klsesp2/libc21/libArduinoJson-esphomelib_ID3837.a
Compiling .pioenvs/klsesp2/libf9f/esphome-core/esphome/api/api_message.cpp.o
Archiving .pioenvs/klsesp2/liba13/libESPmDNS.a
Indexing .pioenvs/klsesp2/liba13/libESPmDNS.a
Compiling .pioenvs/klsesp2/libf9f/esphome-core/esphome/api/api_server.cpp.o
[... 364 lines cut ...]
[17:24:01]EXCVADDR: 0x00000008  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  
[17:24:01]
[17:24:01]Backtrace: 0x400eafe3:0x3ffb17a0 0x400f1c46:0x3ffb1ab0 0x400f1c7e:0x3ffb1b40 0x400d477e:0x3ffb1b80 0x400d4619:0x3ffb1ba0 0x4008450a:0x3ffb1be0 0x4013799f:0x3ffb1c30 0x40132787:0x3ffb1c70 0x4010fa20:0x3ffb1c90 0x4010fbf6:0x3ffb1cc0 0x400e3bf9:0x3ffb1cf0 0x4012f566:0x3ffb1d10 0x400d7a80:0x3ffb1e10 0x400d6d21:0x3ffb1e40 0x400d3862:0x3ffb1e80 0x400d33b5:0x3ffb1ea0 0x400d22a5:0x3ffb1ef0 0x400dc1b7:0x3ffb1fb0 0x40091245:0x3ffb1fd0
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x400eafe3: _svfprintf_r at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vfprintf.c:1604
WARNING Decoded 0x400f1c46: _vsnprintf_r at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vsnprintf.c:72
WARNING Decoded 0x400f1c7e: vsnprintf at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vsnprintf.c:41
WARNING Decoded 0x400d477e: esphome::LogComponent::log_vprintf_(int, char const*, char const*, __va_list_tag)
WARNING Decoded 0x400d4619: esp_idf_log_vprintf_(char const*, __va_list_tag)
WARNING Decoded 0x4008450a: esp_log_write at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/log/log.c:215
WARNING Decoded 0x4013799f: pp_create_task
WARNING Decoded 0x40132787: ic_create_wifi_task
WARNING Decoded 0x4010fa20: wifi_init_in_caller_task
WARNING Decoded 0x4010fbf6: esp_wifi_init_internal
WARNING Decoded 0x400e3bf9: esp_wifi_init at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/wifi_init.c:98
WARNING Decoded 0x4012f566: WiFiGenericClass::mode(wifi_mode_t)
WARNING Decoded 0x400d7a80: esphome::WiFiComponent::wifi_mode_(esphome::optional<bool>, esphome::optional<bool>)
WARNING Decoded 0x400d6d21: esphome::WiFiComponent::setup()
WARNING Decoded 0x400d3862: esphome::Component::call_setup()
WARNING Decoded 0x400d33b5: esphome::Application::setup()
WARNING Decoded 0x400d22a5: setup()
WARNING Decoded 0x400dc1b7: loopTask(void*)
WARNING Decoded 0x40091245: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c:355 (discriminator 1)
[17:24:01]
[17:24:01]Rebooting...

velocyped, 2019-05-04:
```

Decoding stack results
0x400eafe3: _svfiprintf_r at ../../../.././newlib/libc/stdio/vfprintf.c line 1536
0x400f1c7e: pbuf_copy at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/pbuf.c line 981
0x4008450a: cache_sram_mmu_set at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/cache_sram_mmu.c line 73
0x4013799f: rijndaelEncrypt at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/wpa_supplicant/src/crypto/aes-internal-enc.c line 84
0x400e3bf9: _vfiprintf_r at ../../../.././newlib/libc/stdio/vfprintf.c line 1399
0x400d7a80: _Bfree at ../../../.././newlib/libc/stdlib/mprec.c line 138
0x400d6d21: tcpip_adapter_reset_ip_info at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/tcpip_adapter/tcpip_adapter_lwip.c line 1207
0x400d3862: uartRead at /home/k/.arduino15/packages/esp32/hardware/esp32/1.0.2/cores/esp32/esp32-hal-uart.c line 283
0x400d33b5: String::operator=(char const*) at /home/k/.arduino15/packages/esp32/hardware/esp32/1.0.2/cores/esp32/WString.cpp line 266
0x400d22a5: WiFiGenericClass::_eventCallback(void*, system_event_t*) at /home/k/.arduino15/packages/esp32/hardware/esp32/1.0.2/libraries/WiFi/src/WiFiGeneric.cpp line 356
0x400dc1b7: nvs::Storage::writeMultiPageBlob(unsigned char, char const*, void const*, unsigned int, nvs::VerOffset) at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/nvs_flash/src/nvs_storage.cpp line 150
0x40091245: rtc_clk_bbpll_enable at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/soc/esp32/rtc_clk.c line 473
```

velocyped, 2019-05-04:
I really don't know what to do with this, I hope someone will get an expert eye on this...

OttoWinter, 2019-07-11:
Closing due to inactivity - and the issue is probably something in a library like esp-idf or arduino-esp32, not esphome

closed 2019-07-11 by OttoWinter