esphome/issues #265: MPU-6050 suspicion of wrong value conversion

norges, 2019-04-27:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Fist tested with the Hass.io Addon, than with esphome in a Docker Container on a Raspberry (Ubuntu 18.04 LTS as Host and esphome/esphome-armhf as Image of the esphome container)  Pi3+ to have the possibility of an USB connection for debugging.

**ESP (ESP32/ESP8266, Board/Sonoff):**
 
ESP32 nodemcu, but tested same behavior with an esp 8266 nodemcu and a D1 mini

**Affected component:**

https://esphome.io/components/sensor/mpu6050.html

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: mpu6050 
  platform: ESP32
  board: nodemcu-32s

debug:

# Enable logging
logger:
  level: VERY_VERBOSE
  #level: VERBOSE
# Enable Home Assistant API

i2c:
  sda: 21
  scl: 22
  frequency: 100kHz
  scan: False
  
sensor:
  - platform: mpu6050
    address: 0x68
    accel_x:
      name: 'MPU6050 Accel X'
    accel_y:
      name: 'MPU6050 Accel Y'
    accel_z:
      name: 'MPU6050 Accel z'
    gyro_x:
      name: 'MPU6050 Gyro X'
    gyro_y:
      name: 'MPU6050 Gyro Y'
    gyro_z:
      name: 'MPU6050 Gyro z'
    temperature:
      name: 'MPU6050 Temperature'
    update_interval: 2s

```

**Traceback (if applicable):**

```
[09:08:03][V][sensor.mpu6050:114]:     Updating MPU6050...
[09:08:03][VV][i2c:063]: Beginning Transmission to 0x68:
[09:08:03][VV][i2c:100]:     Writing 0b00111011 (0x3B)
[09:08:03][VV][i2c:068]:     Transmission ended. Status code: 0x00
[09:08:03][VV][i2c:090]: Requesting 14 bytes from 0x68:
[09:08:03][VV][i2c:133]:     Received 0b0000000001010100 (0x0054)
[09:08:03][VV][i2c:133]:     Received 0b0000000001110000 (0x0070)
[09:08:03][VV][i2c:133]:     Received 0b0011111000110000 (0x3E30)
[09:08:03][VV][i2c:133]:     Received 0b1110110010100000 (0xECA0)
[09:08:03][VV][i2c:133]:     Received 0b1111111111000101 (0xFFC5)
[09:08:03][VV][i2c:133]:     Received 0b0000000000010101 (0x0015)
[09:08:03][VV][i2c:133]:     Received 0b0000000000000110 (0x0006)
[09:08:04][D][sensor.mpu6050:134]: Got accel={x=0.050 m/s², y=0.067 m/s², z=9.523 m/s²}, gyro={x=3992.460 °/s, y=1.280 °/s, z=0.366 °/s}, temp=214.695°C
[09:08:04][V][sensor.sensor:020]: 'MPU6050 Accel X': Received new state 0.050249
[09:08:04][V][sensor.sensor:020]: 'MPU6050 Accel Y': Received new state 0.066999
[09:08:04][V][sensor.sensor:020]: 'MPU6050 Accel z': Received new state 9.523434
[09:08:04][V][sensor.sensor:020]: 'MPU6050 Temperature': Received new state 214.694702
[09:08:04][V][sensor.sensor:020]: 'MPU6050 Gyro X': Received new state 3992.460205
[09:08:04][V][sensor.sensor:020]: 'MPU6050 Gyro Y': Received new state 1.280475
[09:08:04][V][sensor.sensor:020]: 'MPU6050 Gyro z': Received new state 0.365850
[09:08:05][VV][component:126]: Running interval 'update':0 with interval=2000 last_execution=1470 (now=3486)
[09:08:05][V][sensor.mpu6050:114]:     Updating MPU6050...
[09:08:05][VV][i2c:063]: Beginning Transmission to 0x68:
[09:08:05][VV][i2c:100]:     Writing 0b00111011 (0x3B)
[09:08:05][VV][i2c:068]:     Transmission ended. Status code: 0x00
[09:08:05][VV][i2c:090]: Requesting 14 bytes from 0x68:
[09:08:05][VV][i2c:133]:     Received 0b1111111111001100 (0xFFCC)
[09:08:05][VV][i2c:133]:     Received 0b1111111110111100 (0xFFBC)
[09:08:05][VV][i2c:133]:     Received 0b0011111001001100 (0x3E4C)
[09:08:05][VV][i2c:133]:     Received 0b1110110010100000 (0xECA0)
[09:08:05][VV][i2c:133]:     Received 0b1111111111000110 (0xFFC6)
[09:08:05][VV][i2c:133]:     Received 0b0000000000010001 (0x0011)
[09:08:05][VV][i2c:133]:     Received 0b0000000000000110 (0x0006)
[09:08:06][D][sensor.mpu6050:134]: Got accel={x=39.173 m/s², y=39.163 m/s², z=9.540 m/s²}, gyro={x=3992.521 °/s, y=1.037 °/s, z=0.366 °/s}, temp=214.695°C
[09:08:06][V][sensor.sensor:020]: 'MPU6050 Accel X': Received new state 39.172901
[09:08:06][V][sensor.sensor:020]: 'MPU6050 Accel Y': Received new state 39.163326
[09:08:06][V][sensor.sensor:020]: 'MPU6050 Accel z': Received new state 9.540184
[09:08:06][V][sensor.sensor:020]: 'MPU6050 Temperature': Received new state 214.694702
[09:08:06][V][sensor.sensor:020]: 'MPU6050 Gyro X': Received new state 3992.520996
[09:08:06][V][sensor.sensor:020]: 'MPU6050 Gyro Y': Received new state 1.036575
[09:08:06][V][sensor.sensor:020]: 'MPU6050 Gyro z': Received new state 0.365850
```

**Additional information and things you've tried:**
A full log from an reset on can be found on: https://gist.github.com/norges/75b4b807cef605b41b1d2ab4e31109b0

I looked up in a datasheet for the MPU-6050  how the values should be converted:
https://cdn-reichelt.de/documents/datenblatt/A300/MPU_6050_DB2.pdf
On the pages 29 -31 are descriptions for the output register for acceleration, temperature and gyroscopes.
The values for the temperature are  16 Bit signed values and the Values for acceleration and gyroscopes are 16 Bit 2's complement values.

I calculated for the raw values above the following tables for the two measurements:

  | raw value | 2' complement or signed short | normalized | as dec | normalized
-- | -- | -- | -- | -- | --
acc x | 0x0054 | -84 | -0,05 | 84 | 0,05
acc y | 0x0070 | -112 | -0,07 | 112 | 0,07
acc z | 0x3E30 | -15920 | -9,53 | 15920 | 9,53
temp | 0xECA0 | -4960 | 21,94 | 60576 | 214,69
gyro x | 0xFFC5 | 59 | 3,60 | 65477 | 3992,46
gyro y | 0x0015 | -21 | -1,28 | 21 | 1,28
gyro z | 0x0006 | -6 | -0,37 | 6 | 0,37

  | raw value | 2' complement or signed short | normalized | as dec | normalized
-- | -- | -- | -- | -- | --
acc   x | 0xFFCC | 52 | 0,03 | 65484 | 39,20
acc   y | 0xFFBC | 68 | 0,04 | 65468 | 39,19
acc   z | 0x3E4C | -15948 | -9,55 | 15948 | 9,55
temp | 0xECA0 | -4960 | 21,94 | 60576 | 214,69
gyro   x | 0xFFC6 | 58 | 3,54 | 65478 | 3992,52
gyro   y | 0x0011 | -17 | -1,04 | 17 | 1,04
gyro   z | 0x0006 | -6 | -0,37 | 6 | 0,37

In the first one at least the acceleration data are not obvious wrong, but in the second one e.g. acc x and y are bigger than the maximum of 2g (2*9,8) which would be more than the range 
of 2G wich is used per https://github.com/esphome/esphome-core/blob/acccea739754fc12689a880b1e5ef7ae8eed318a/src/esphome/sensor/mpu6050_component.cpp the normalization are done like in the mpu6050_component.cpp and the datasheet there I don't have found any differences.

The values that are treated as decimals are exactly the ones after normalization that are in the output of the log, so I assume that the problem is the treatment of the raw values as decimals instead of 2' complements or for temperature signed short values.

If there is any more information needed to reproduce the issue feel free to ask. As I'm not good at C I'm unfortunately not able to make an PR for this myself.

OttoWinter, 2019-05-01:
Thanks for this _very_ detailed bug report!

You're completely right. ESPHome does not properly read these values as signed 16-bit ints. I could have sworn that it worked when I initially developed it, but the code clearly says otherwise. Anyway, this will be an easy fix (have it here locally already).

Thanks again!

norges, 2019-05-09:
The temperature is still off and the datasheet is not very clear about the values, at least for me. I tried it with ramping up the temperature and if the temp values are also int16_t it was correct as I would expect it.