esphome/issues #270: No new code is updated

celeroll, 2019-04-28:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Docker

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP32S NodeMCU

**Affected component:**
Core
MQTT

**Description of problem:**
Since I added the MQTT component, the code was compiled and uploaded. 
But since then I tried changing the configuration multiple times, even commented out MQTT, but every time I hit "upload" it does not upload my new code.
This can be seen when I start the log and at the beginning just after a new compile  and upload is done I read this:

```
[14:03:39][I][application:097]: esphome-core version 1.12.2 compiled on Apr 28 2019, 12:16:41
```

Almost the compiled code is two hours old (when I first added mqtt component).

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

celeroll, 2019-04-28:
I have even removed every component and this are the results:

Configuration:
```
esphome:
  name: esp32_node1
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: !secret wifi_id
  password: !secret wifi_pass
  manual_ip:
    static_ip: 192.168.1.10
    gateway: 192.168.1.1
    subnet: 255.255.255.0

# Enable logging
logger:
  # level: VERY_VERBOSE

# Enable Home Assistant API
api:

ota:
```

Log after compiling:
```
Linking .pioenvs/esp32_node1/firmware.elf
Building .pioenvs/esp32_node1/firmware.bin
Retrieving maximum program size .pioenvs/esp32_node1/firmware.elf
Checking size .pioenvs/esp32_node1/firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [=         ]  12.3% (used 40176 bytes from 327680 bytes)
PROGRAM: [=====     ]  51.8% (used 848294 bytes from 1638400 bytes)
esptool.py v2.6
========================= [SUCCESS] Took 93.91 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.168.1.10
INFO Uploading /config/esp32_node1/.pioenvs/esp32_node1/firmware.bin (848416 bytes)
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 192.168.1.10 using esphome API
INFO Connecting to 192.168.1.10:6053 (192.168.1.10)
INFO Successfully connected to 192.168.1.10
[15:36:09][I][application:097]: esphome-core version 1.12.2 compiled on Apr 28 2019, 12:16:41
[15:36:09][C][wifi:371]: WiFi:
[15:36:09][C][wifi:253]:   SSID: [redacted]
[15:36:09][C][wifi:254]:   IP Address: 192.168.1.10
[15:36:09][C][wifi:256]:   BSSID: [redacted]
[15:36:09][C][wifi:257]:   Hostname: 'esp32_node1'
[15:36:09][C][wifi:261]:   Signal strength: -36 dB ?_?¦
[15:36:09][C][wifi:262]:   Channel: 10
[15:36:09][C][wifi:263]:   Subnet: 255.255.252.0
[15:36:09][C][wifi:264]:   Gateway: 192.168.1.1
[15:36:09][C][wifi:265]:   DNS1: 1.1.1.1
[15:36:09][C][wifi:266]:   DNS2: 1.0.0.1
[15:36:09][C][output.ledc:032]: LEDC Output:
[15:36:09][C][output.ledc:033]:   Pin: 21
[15:36:09][C][output.ledc:034]:   LEDC Channel: 1
[15:36:09][C][output.ledc:035]:   Frequency: 25000.0 Hz
[15:36:09][C][output.ledc:036]:   Bit Depth: 10
[15:36:09][C][light.fast_led:025]: FastLED light:
[15:36:09][C][light.fast_led:026]:   Num LEDs: 11
[15:36:09][C][light.fast_led:027]:   Max refresh rate: 0
[15:36:09][C][light.fast_led:025]: FastLED light:
[15:36:09][C][light.fast_led:026]:   Num LEDs: 11
[15:36:09][C][light.fast_led:027]:   Max refresh rate: 0
[15:36:09][C][light.fast_led:025]: FastLED light:
[15:36:09][C][light.fast_led:026]:   Num LEDs: 11
[15:36:09][C][light.fast_led:027]:   Max refresh rate: 0
[15:36:09][C][light.fast_led:025]: FastLED light:
[15:36:09][C][light.fast_led:026]:   Num LEDs: 11
[15:36:09][C][light.fast_led:027]:   Max refresh rate: 0
[15:36:09][C][fan.basic_fan:038]: Fan 'Lowboard Fan':
[15:36:09][C][fan.basic_fan:043]:   Mode: Speed
[15:36:09][C][fan.basic_fan:045]:   Speeds: Low=33% Medium=66% High=100%
[15:36:09][C][logger:142]: Logger:
[15:36:09][C][logger:143]:   Level: DEBUG
[15:36:09][C][logger:144]:   Log Baud Rate: 115200
[15:36:09][C][logger:145]:   Hardware UART: UART0
[15:36:09][C][light.state:225]: Light 'LowBoard Led1'
[15:36:09][C][light.state:227]:   Default Transition Length: 0 ms
[15:36:09][C][light.state:228]:   Gamma Correct: 2.80
[15:36:09][C][light.state:225]: Light 'LowBoard Led2'
[15:36:09][C][light.state:227]:   Default Transition Length: 0 ms
[15:36:09][C][light.state:228]:   Gamma Correct: 2.80
[15:36:09][C][light.state:225]: Light 'LowBoard Led3'
[15:36:09][C][light.state:227]:   Default Transition Length: 0 ms
[15:36:09][C][light.state:228]:   Gamma Correct: 2.80
[15:36:09][C][light.state:225]: Light 'LowBoard Led4'
[15:36:09][C][light.state:227]:   Default Transition Length: 0 ms
[15:36:09][C][light.state:228]:   Gamma Correct: 2.80
[15:36:09][C][api:101]: API Server:
[15:36:09][C][api:102]:   Address: 192.168.1.10:6053
[15:36:09][C][ota:127]: Over-The-Air Updates:
[15:36:09][C][ota:128]:   Address: 192.168.1.10:3232
[15:36:09][C][mqtt.client:062]: MQTT:
[15:36:09][C][mqtt.client:064]:   Server Address: 192.168.1.10:1883 (192.168.1.10)
[15:36:09][C][mqtt.client:065]:   Username: [redacted]
[15:36:09][C][mqtt.client:066]:   Client ID: [redacted]
[15:36:09][C][mqtt.client:068]:   Discovery prefix: 'homeassistant'
[15:36:09][C][mqtt.client:069]:   Discovery retain: YES
[15:36:09][C][mqtt.client:071]:   Topic Prefix: 'esp32_node1'
[15:36:09][C][mqtt.client:073]:   Log Topic: 'esp32_node1/debug'
[15:36:09][C][mqtt.client:076]:   Availability: 'esp32_node1/status'
[15:36:09][C][light.mqtt_json:054]: MQTT Light 'LowBoard Led1':
[15:36:09][C][light.mqtt_json:055]:   State Topic: 'esp32_node1/light/lowboard_led1/state'
[15:36:09][C][light.mqtt_json:055]:   Command Topic: 'esp32_node1/light/lowboard_led1/command'
[15:36:09][C][light.mqtt_json:054]: MQTT Light 'LowBoard Led2':
[15:36:09][C][light.mqtt_json:055]:   State Topic: 'esp32_node1/light/lowboard_led2/state'
[15:36:09][C][light.mqtt_json:055]:   Command Topic: 'esp32_node1/light/lowboard_led2/command'
[15:36:09][C][light.mqtt_json:054]: MQTT Light 'LowBoard Led3':
[15:36:09][C][light.mqtt_json:055]:   State Topic: 'esp32_node1/light/lowboard_led3/state'
[15:36:09][C][light.mqtt_json:055]:   Command Topic: 'esp32_node1/light/lowboard_led3/command'
[15:36:09][C][light.mqtt_json:054]: MQTT Light 'LowBoard Led4':
[15:36:09][C][light.mqtt_json:055]:   State Topic: 'esp32_node1/light/lowboard_led4/state'
[15:36:09][C][light.mqtt_json:055]:   Command Topic: 'esp32_node1/light/lowboard_led4/command'
```

glmnet, 2019-04-30:
My best bet is you have two nodes with the same ip.

celeroll, 2019-05-01:
I had to flash over USB and now I can upload again over OTA.
The strange thing here is, that the system is reporting no error over OTA.

OttoWinter, 2019-05-01:
This will need some USB logs during the OTA upload.

celeroll, 2019-05-01:
Currently I'm not facing the problem with OTA. Maybe  it was something temporally. 
Lets close this issue and if I encounter same behavior I'll open a new one.

celeroll, 2019-05-03:
I have again failures over OTA.
Over serial (termite) I have captured the log:
```
[1B][0;36m[D][ota:179]: Starting OTA Update from 192.168.1.2...[1B][0m
[1B][0;36m[D][ota:353]: OTA in progress: 0.1%[1B][0m
[1B][0;36m[D][ota:353]: OTA in progress: 7.3%[1B][0m
[1B][0;36m[D][ota:353]: OTA in progress: 13.8%[1B][0m
[1B][0;36m[D][ota:353]: OTA in progress: 20.0%[1B][0m
[1B][0;36m[D][ota:353]: OTA in progress: 26.3%[1B][0m
[1B][0;36m[D][ota:353]: OTA in progress: 32.5%[1B][0m
[1B][0;36m[D][ota:353]: OTA in progress: 38.6%[1B][0m
[1B][0;36m[D][ota:353]: OTA in progress: 45.0%[1B][0m
[1B][0;36m[D][ota:353]: OTA in progress: 50.9%[1B][0m
[1B][0;36m[D][ota:353]: OTA in progress: 56.9%[1B][0m
[1B][0;36m[D][ota:353]: OTA in progress: 63.0%[1B][0m
[1B][0;36m[D][ota:353]: OTA in progress: 68.8%[1B][0m
[1B][0;36m[D][ota:353]: OTA in progress: 75.2%[1B][0m
[1B][0;36m[D][ota:353]: OTA in progress: 81.1%[1B][0m
[1B][0;36m[D][ota:353]: OTA in progress: 87.8%[1B][0m
[1B][0;36m[D][ota:353]: OTA in progress: 93.3%[1B][0m
[1B][0;36m[D][ota:353]: OTA in progress: 99.7%[1B][0m
[1B][0;32m[I][ota:377]: OTA update finished![1B][0m
[1B][0;32m[I][helpers:266]: Rebooting safely... Reason: 'ota'[1B][0m
[1B][0;36m[D][api:246]: Error from client '192.168.1.2': -13[1B][0m
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:808
load:0x40078000,len:6084
load:0x40080000,len:6704
entry 0x400802e4
[1B][0;32m[I][logger:119]: Log initialized[1B][0m
[1B][0;35m[C][ota:484]: There have been 0 suspected unsuccessful boot attempts.[1B][0m
[1B][0;32m[I][application:055]: Running through setup()...[1B][0m
[1B][0;35m[C][light.fast_led:016]: Setting up FastLED light...[1B][0m
[1B][0;35m[C][light.fast_led:016]: Setting up FastLED light...[1B][0m
[1B][0;35m[C][light.fast_led:016]: Setting up FastLED light...[1B][0m
[1B][0;35m[C][light.fast_led:016]: Setting up FastLED light...[1B][0m
[1B][0;35m[C][light.state:138]: Setting up light 'LowBoard Led1'...[1B][0m
[1B][0;36m[D][light.state:418]: 'LowBoard Led1' Turning OFF[1B][0m
[1B][0;36m[D][light.state:425]:   Brightness: 52%[1B][0m
[1B][0;36m[D][light.state:449]:   Red=100%, Green=100%, Blue=57%[1B][0m
[1B][0;35m[C][light.state:138]: Setting up light 'LowBoard Led2'...[1B][0m
[1B][0;36m[D][light.state:418]: 'LowBoard Led2' Turning OFF[1B][0m
[1B][0;36m[D][light.state:425]:   Brightness: 44%[1B][0m
[1B][0;36m[D][light.state:449]:   Red=100%, Green=100%, Blue=57%[1B][0m
[1B][0;35m[C][light.state:138]: Setting up light 'LowBoard Led3'...[1B][0m
[1B][0;36m[D][light.state:418]: 'LowBoard Led3' Turning OFF[1B][0m
[1B][0;36m[D][light.state:425]:   Brightness: 40%[1B][0m
[1B][0;36m[D][light.state:449]:   Red=100%, Green=100%, Blue=71%[1B][0m
[1B][0;35m[C][light.state:138]: Setting up light 'LowBoard Led4'...[1B][0m
[1B][0;36m[D][light.state:418]: 'LowBoard Led4' Turning OFF[1B][0m
[1B][0;36m[D][light.state:425]:   Brightness: 31%[1B][0m
[1B][0;36m[D][light.state:449]:   Red=100%, Green=100%, Blue=57%[1B][0m
[1B][0;35m[C][wifi:027]: Setting up WiFi...[1B][0m
[1B][0;36m[D][wifi:271]: Starting scan...[1B][0m
[1B][0;36m[D][fan.basic_fan:066]: Setting speed: 0.00[1B][0m
[1B][0;36m[D][wifi:286]: Found networks:[1B][0m
[1B][0;32m[I][wifi:321]: - 'xxxxx' [1B][5m(CC:CE:1E:2D:9D:F5) [1B][6m[1B][0;32m▂▄▆█[1B][0m[1B][0m
[1B][0;36m[D][wifi:322]:     Channel: 10[1B][0m
[1B][0;36m[D][wifi:323]:     RSSI: -45 dB[1B][0m
[1B][0;32m[I][wifi:321]: - 'xxxxx' [1B][5m(CC:CE:1E:AF:43:A3) [1B][6m[1B][0;33m▂▄[1B][0;37m▆█[1B][0m[1B][0m
[1B][0;36m[D][wifi:322]:     Channel: 10[1B][0m
[1B][0;36m[D][wifi:323]:     RSSI: -78 dB[1B][0m
[1B][0;36m[D][wifi:325]: - [1B][5m''[1B][6m [1B][5m(B6:7C:9C:DD:13:C7) [1B][6m[1B][0;33m▂▄▆[1B][0;37m█[1B][0m[1B][0m
[1B][0;32m[I][wifi:167]: WiFi Connecting to 'xxxxx'...[1B][0m
[... 173 lines cut ...]
[1B][0;35m[C][mqtt.client:066]:   Client ID: [1B][5m'esp32_node1-3c71bf6a70d0'[1B][6m[1B][0m
[1B][0;35m[C][mqtt.client:071]:   Topic Prefix: 'esp32_LowBoard'[1B][0m
[1B][0;35m[C][mqtt.client:073]:   Log Topic: 'esp32_LowBoard/debug'[1B][0m
[1B][0;35m[C][mqtt.client:076]:   Availability: 'esp32_LowBoard/status'[1B][0m
[1B][0;35m[C][sensor.mqtt:025]: MQTT Sensor 'esp32_LowBoard Uptime Sensor':[1B][0m
[1B][0;35m[C][sensor.mqtt:027]:   Expire After: 300s[1B][0m
[1B][0;35m[C][sensor.mqtt:029]:   State Topic: 'esp32_LowBoard/sensor/esp32_lowboard_uptime_sensor/state'[1B][0m
[1B][0;35m[C][switch.mqtt:042]: MQTT Switch 'LowBoard ESP32 Restart ': [1B][0m
[1B][0;35m[C][switch.mqtt:043]:   State Topic: 'esp32_LowBoard/switch/lowboard_esp32_restart_/state'[1B][0m
[1B][0;35m[C][switch.mqtt:043]:   Command Topic: 'esp32_LowBoard/switch/lowboard_esp32_restart_/command'[1B][0m
[1B][0;35m[C][light.mqtt_json:054]: MQTT Light 'LowBoard Led1':[1B][0m
[1B][0;35m[C][light.mqtt_json:055]:   State Topic: 'esp32_LowBoard/state/led1'[1B][0m
[1B][0;35m[C][light.mqtt_json:055]:   Command Topic: 'esp32_LowBoard/command/led1'[1B][0m
[1B][0;35m[C][light.mqtt_json:054]: MQTT Light 'LowBoard Led2':[1B][0m
[1B][0;35m[C][light.mqtt_json:055]:   State Topic: 'esp32_LowBoard/state/led2'[1B][0m
[1B][0;35m[C][light.mqtt_json:055]:   Command Topic: 'esp32_LowBoard/command/led2'[1B][0m
[1B][0;35m[C][light.mqtt_json:054]: MQTT Light 'LowBoard Led3':[1B][0m
[1B][0;35m[C][light.mqtt_json:055]:   State Topic: 'esp32_LowBoard/state/led3'[1B][0m
[1B][0;35m[C][light.mqtt_json:055]:   Command Topic: 'esp32_LowBoard/command/led3'[1B][0m
[1B][0;35m[C][light.mqtt_json:054]: MQTT Light 'LowBoard Led4':[1B][0m
[1B][0;35m[C][light.mqtt_json:055]:   State Topic: 'esp32_LowBoard/state/led4'[1B][0m
[1B][0;35m[C][light.mqtt_json:055]:   Command Topic: 'esp32_LowBoard/command/led4'[1B][0m
[1B][0;36m[D][api:073]: Disconnecting ESPHome v1.12.2 (192.168.1.2)[1B][0m

```

OttoWinter, 2019-07-11:
Closing due to inactivity

closed 2019-07-11 by OttoWinter