esphome/issues #274: Program failed with code  Error 132

kaiba4786, 2019-04-29:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker on armbian chipset S905w ,  ESPHome 1.12.2
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266
**Affected component:**

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```
esphome:
  name: test
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: "hasst3"
  password: "phandung4786"

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

```

**Traceback (if applicable):**

```
INFO Reading configuration...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/test
Processing test (platform: espressif8266@1.8.0; framework: arduino; board: esp01_1m)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp01_1m.html
PLATFORM: Espressif 8266 > Espressif Generic ESP8266 ESP-01 1M
HARDWARE: ESP8266 80MHz 80KB RAM (1MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 27 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncTCP> 1.2.0
|   |-- <ESP8266WiFi> 1.0
|-- <Hash> 1.0
|-- <esphome-core> 1.12.2
|   |-- <ArduinoJson-esphomelib> 5.13.3
|   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266mDNS>
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESPAsyncTCP> 1.2.0
|   |   |-- <ESP8266WiFi> 1.0
Compiling /data/test/.pioenvs/test/src/main.cpp.o
Generating LD script /data/test/.pioenvs/test/ld/eagle.app.v6.common.ld
Compiling /data/test/.pioenvs/test/lib4d9/ESP8266WiFi/BearSSLHelpers.cpp.o
Compiling /data/test/.pioenvs/test/lib4d9/ESP8266WiFi/CertStoreBearSSL.cpp.o
Illegal instruction (core dumped)
*** [/data/test/.pioenvs/test/src/main.cpp.o] Error 132
Illegal instruction (core dumped)
*** [/data/test/.pioenvs/test/ld/eagle.app.v6.common.ld] Error 132
Illegal instruction (core dumped)
*** [/data/test/.pioenvs/test/lib4d9/ESP8266WiFi/BearSSLHelpers.cpp.o] Error 132
Illegal instruction (core dumped)
*** [/data/test/.pioenvs/test/lib4d9/ESP8266WiFi/CertStoreBearSSL.cpp.o] Error 132
========================== [ERROR] Took 9.70 seconds ==========================

```

**Additional information and things you've tried:**

OttoWinter, 2019-05-01:
What docker image are you using? More detail on the device running the dashboard please too.

kaiba4786, 2019-05-02:
> What docker image are you using? More detail on the device running the dashboard please too.

HI OttoWinter
Thank you for your interest in this issue

I using TV box TX3mini
image OS: Armbian_5.77_Aml-s905_Debian_stretch_default_5.0.2_20190401.img from this:  https://yadi.sk/d/pHxaRAs-tZiei/5.77/S905

then i runing  armbian-config to install docker and hassio.

then  i run command: 

sudo apt-get install docker-ce
sudo apt-get install bash
sudo apt-get install jq
sudo apt-get install curl
sudo apt-get install avahi-daemon
sudo apt-get install dbus
sudo apt-get install apparmor-utils
sudo apt-get install network-manager
sudo su
curl -sL https://raw.githubusercontent.com/home-assistant/hassio-installer/master/hassio_install.sh | bash -s -- -m raspberrypi3

final, i install esphome add-on in to hassio
When compiling build, getting the Error 132

Any help, makes me feel grateful !!!

OttoWinter, 2019-07-11:
Closing due to inactivity

closed 2019-07-11 by OttoWinter