esphome/issues #278: DHT12 component fails, seems to be initialized before I2C

AlexDanault, 2019-04-30:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Command line

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32

**Affected component:**

https://esphome.io/components/sensor/dht12.html?highlight=dht12

**Description of problem:**
In a barebones project, I can't get the ESP32 to read a DHT12 sensor. But trying with a BME280 works perfectly. The sensors is connected correctly, and is even detected during a scan, but the log seems to indicate that the component is loaded too early (before I2C). I tried to adjust the setup_priority but it didn't change anything.

**Problem-relevant YAML-configuration entries:**
```yaml
i2c:
  scan: True

sensor:
  - platform: dht12
    temperature:
      name: "TEST Temperature"
    humidity:
      name: "TEST Room Humidity"
    update_interval: 10s
```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

This is the log:
```
[19:20:50][I][logger:119]: Log initialized
[19:20:50][I][application:055]: Running through setup()...
[19:20:52][I][wifi:321]: - 'Alex' (B8:EC:A3:CE:CF:90) ÔûéÔûäÔûåÔûê
[19:20:52][I][wifi:167]: WiFi Connecting to 'Alex'...
[19:20:53][I][wifi:379]: WiFi connected!
[19:20:53][I][mqtt.client:173]: Connecting to MQTT...
[19:20:53][I][mqtt.client:213]: MQTT Connected!
[19:20:54][W][i2c:077]: Received NACK on transmit of address 0x5C
[19:20:54][W][sensor.dht12:068]: Updating DHT12 failed!
[19:20:54][E][component:151]: Component was marked as failed.
[19:20:54][I][application:089]: setup() finished successfully!
[19:20:54][I][application:097]: esphome-core version 1.12.2 compiled on Apr 30 2019, 19:19:28
[19:20:54][I][i2c:040]: Scanning i2c bus for active devices...
[19:20:54][I][i2c:047]: Found i2c device at address 0x5C
[19:20:54][E][sensor.dht12:060]: Communication with DHT12 failed!
[19:20:54][I][application:109]: Running through first loop()
[19:20:54][I][application:136]: First loop finished successfully!
[19:22:50][I][ota:152]: Boot seems successful, resetting boot loop counter.
```

AlexDanault, 2019-04-30:
After reading some current issues it seems this is *exactly* the same issue as #192 .

OttoWinter, 2019-05-01:
> but the log seems to indicate that the component is loaded too early (before I2C).

Nah, i2c is set up way before (actually one of the very first things). The i2c *scan* however is done in `dump_config`, after all setup() functions finished.

So this would be the same as #192, yes.

closed 2019-05-01 by OttoWinter